riscv-collab/riscv-openocd #358: how to use riscv-openocd(fFreedomStudio windows)

topchenhui, 2019-03-12:
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Error: libusb_open() failed with LIBUSB_ERROR_NOT_FOUND
Error: no device found
Error: unable to open ftdi device with vid 0403, pid 6010, description 'Dual RS232-HS', serial '*' at bus location '*'

timsifive, 2019-03-12:
This error indicates OpenOCD couldn't find the USB JTAG device, which is not RISC-V-specific, and more related to the operating system you're using. The following Google search will come up with a bunch of trouble shooting suggestions for people with the same problem: https://www.google.com/search?q=Error%3A+libusb_open()+failed+with+LIBUSB_ERROR_NOT_FOUND

timsifive, 2020-03-30:
Closing due to inactivity.

closed 2020-03-30 by timsifive