riscv-collab/riscv-openocd #359: riscv openocd tests don't work with unavailable harts

bluewww, 2019-03-18:
My (test)system has `hartid=0` unavailable, `hartid=1` first existing hart, `hartid=2,3,4...` nonexistent as allowed by the debug spec v0.13.1.

See p.10
> Systems with very large number of harts may permanently disable some during manufacturing, leaving holes in the otherwise continuous hart index space. In order to let the debugger discover all harts, they must show up as unavailable even if there is no chance of them ever becoming available.

This happens when I try to run the tests (with `-rtos`)
```
Debug: 222 439 core.c:1190 jtag_validate_ircapture(): IR capture validation scan
Debug: 223 481 core.c:1248 jtag_validate_ircapture(): riscv.cpu: IR capture 0x05
Debug: 224 481 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_dap init
Debug: 225 481 command.c:143 script_debug(): command - ocd_dap ocd_dap init
Debug: 227 481 arm_dap.c:105 dap_init_all(): Initializing all DAPs ...
Debug: 228 481 openocd.c:159 handle_init_command(): Examining targets...
Debug: 229 481 target.c:1591 target_call_event_callbacks(): target event 17 (examine-start) for core 0
Debug: 230 481 riscv.c:785 riscv_examine(): riscv_examine()
Debug: 231 524 riscv.c:227 dtmcontrol_scan(): DTMCONTROL: 0x0 -> 0x1071
Debug: 232 524 riscv.c:795 riscv_examine(): dtmcontrol=0x1071
Debug: 233 524 riscv.c:797 riscv_examine():   version=0x1
Debug: 234 524 riscv-013.c:1604 init_target(): init
Debug: 235 569 riscv-013.c:427 dtmcontrol_scan(): DTMCS: 0x0 -> 0x1071
Debug: 236 569 riscv-013.c:1370 examine(): dtmcontrol=0x1071
Debug: 237 569 riscv-013.c:1371 examine():   dmireset=0
Debug: 238 569 riscv-013.c:1372 examine():   idle=1
Debug: 239 569 riscv-013.c:1373 examine():   dmistat=0
Debug: 240 569 riscv-013.c:1374 examine():   abits=7
Debug: 241 569 riscv-013.c:1375 examine():   version=1
Debug: 242 614 riscv-013.c:379 scan(): 41b 0i w 00000000 @10 -> + 00000000 @00
Debug: 243 655 riscv-013.c:379 scan(): 41b 0i - 00000000 @10 -> + 00000000 @10
Debug: 244 698 riscv-013.c:379 scan(): 41b 0i w 00000001 @10 -> + 00000000 @10
Debug: 245 698 riscv-013.c:390 scan():  dmactive -> 
Debug: 246 739 riscv-013.c:379 scan(): 41b 0i - 00000000 @10 -> + 00000001 @10
Debug: 247 739 riscv-013.c:390 scan():  ->  dmactive
Debug: 248 782 riscv-013.c:379 scan(): 41b 0i w 03ffffc1 @10 -> + 00000000 @10
Debug: 249 782 riscv-013.c:390 scan():  hartselhi=1023 hartsello=1023 dmactive -> 
Debug: 250 823 riscv-013.c:379 scan(): 41b 0i - 00000000 @10 -> + 03ffffc1 @10
Debug: 251 823 riscv-013.c:390 scan():  ->  hartselhi=1023 hartsello=1023 dmactive
Debug: 252 866 riscv-013.c:379 scan(): 41b 0i r 00000000 @10 -> + 00000000 @10
Debug: 253 907 riscv-013.c:379 scan(): 41b 0i - 00000000 @10 -> b 00000000 @10
Debug: 254 907 riscv-013.c:438 increase_dmi_busy_delay(): dtmcs_idle=1, dmi_busy_delay=1, ac_busy_delay=0
Debug: 255 950 riscv-013.c:427 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x1c71
Debug: 256 993 riscv-013.c:379 scan(): 41b 1i - 00000000 @10 -> + 03ffffc1 @10
Debug: 257 993 riscv-013.c:390 scan():  ->  hartselhi=1023 hartsello=1023 dmactive
Debug: 258 1036 riscv-013.c:379 scan(): 41b 1i r 00000000 @11 -> + 00000000 @10
Debug: 259 1077 riscv-013.c:379 scan(): 41b 1i - 00000000 @11 -> b 00000000 @11
Debug: 260 1077 riscv-013.c:438 increase_dmi_busy_delay(): dtmcs_idle=1, dmi_busy_delay=2, ac_busy_delay=0
Debug: 261 1120 riscv-013.c:427 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x1071
Debug: 262 1163 riscv-013.c:379 scan(): 41b 2i - 00000000 @11 -> + 000ccc82 @11
Debug: 263 1163 riscv-013.c:390 scan():  ->  allhavereset anyhavereset allnonexistent anynonexistent allrunning anyrunning authenticated version=2
Debug: 264 1163 riscv-013.c:1413 examine(): dmstatus:  0x000ccc82
Debug: 265 1163 riscv-013.c:1429 examine(): hartsellen=20
Debug: 266 1206 riscv-013.c:379 scan(): 41b 2i r 00000000 @12 -> + 00000000 @11
Debug: 267 1247 riscv-013.c:379 scan(): 41b 2i - 00000000 @12 -> b 00000000 @12
Debug: 268 1247 riscv-013.c:438 increase_dmi_busy_delay(): dtmcs_idle=1, dmi_busy_delay=3, ac_busy_delay=0
Debug: 269 1290 riscv-013.c:427 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x1071
Debug: 270 1333 riscv-013.c:379 scan(): 41b 3i - 00000000 @12 -> + 00000000 @12
Debug: 271 1376 riscv-013.c:379 scan(): 41b 3i r 00000000 @38 -> + 00000000 @12
Debug: 272 1417 riscv-013.c:379 scan(): 41b 3i - 00000000 @38 -> b 00000000 @38
Debug: 273 1417 riscv-013.c:438 increase_dmi_busy_delay(): dtmcs_idle=1, dmi_busy_delay=4, ac_busy_delay=0
Debug: 274 1460 riscv-013.c:427 dtmcontrol_scan(): DTMCS: 0x10000 -> 0x1071
Debug: 275 1503 riscv-013.c:379 scan(): 41b 4i - 00000000 @38 -> + 20040404 @38
Debug: 276 1503 riscv-013.c:390 scan():  ->  sbaccess=2 sbasize=32 sbaccess32
Debug: 277 1546 riscv-013.c:379 scan(): 41b 4i r 00000000 @16 -> + 00000000 @38
Debug: 278 1588 riscv-013.c:379 scan(): 41b 4i - 00000000 @16 -> + 08000002 @16
Debug: 279 1588 riscv-013.c:390 scan():  ->  progbufsize=8 datacount=2
Info : 280 1588 riscv-013.c:1460 examine(): datacount=2 progbufsize=8
Info : 281 1588 riscv-013.c:1478 examine(): coreid=0 hart=0
Debug: 282 1631 riscv-013.c:379 scan(): 41b 4i r 00000000 @10 -> + 00000000 @16
Debug: 283 1673 riscv-013.c:379 scan(): 41b 4i - 00000000 @10 -> + 03ffffc1 @10
Debug: 284 1673 riscv-013.c:390 scan():  ->  hartselhi=1023 hartsello=1023 dmactive
Debug: 285 1716 riscv-013.c:379 scan(): 41b 4i w 00000001 @10 -> + 00000000 @10
[... 567 lines cut ...]
Debug: 695 9195 riscv-013.c:379 scan(): 41b 4i - 00000000 @11 -> + 00003082 @11
Debug: 696 9195 riscv-013.c:390 scan():  ->  allunavail anyunavail authenticated version=2
Error: 697 9195 riscv-013.c:2961 riscv013_is_halted(): Hart 0 is unavailable.
Debug: 698 9238 riscv-013.c:379 scan(): 41b 4i r 00000000 @11 -> + 00000000 @11
Debug: 699 9279 riscv-013.c:379 scan(): 41b 4i - 00000000 @11 -> + 00003082 @11
Debug: 700 9279 riscv-013.c:390 scan():  ->  allunavail anyunavail authenticated version=2
Error: 701 9279 riscv-013.c:2961 riscv013_is_halted(): Hart 0 is unavailable.
Debug: 702 9322 riscv-013.c:379 scan(): 41b 4i r 00000000 @11 -> + 00000000 @11
Debug: 703 9364 riscv-013.c:379 scan(): 41b 4i - 00000000 @11 -> + 00003082 @11
Debug: 704 9364 riscv-013.c:390 scan():  ->  allunavail anyunavail authenticated version=2
Error: 705 9364 riscv-013.c:2961 riscv013_is_halted(): Hart 0 is unavailable.
Debug: 706 9407 riscv-013.c:379 scan(): 41b 4i r 00000000 @11 -> + 00000000 @11
Debug: 707 9449 riscv-013.c:379 scan(): 41b 4i - 00000000 @11 -> + 00003082 @11
Debug: 708 9449 riscv-013.c:390 scan():  ->  allunavail anyunavail authenticated version=2
Error: 709 9449 riscv-013.c:2961 riscv013_is_halted(): Hart 0 is unavailable.
Debug: 710 9492 riscv-013.c:379 scan(): 41b 4i r 00000000 @11 -> + 00000000 @11
Debug: 711 9534 riscv-013.c:379 scan(): 41b 4i - 00000000 @11 -> + 00003082 @11
Debug: 712 9534 riscv-013.c:390 scan():  ->  allunavail anyunavail authenticated version=2
Error: 713 9534 riscv-013.c:2961 riscv013_is_halted(): Hart 0 is unavailable.
Debug: 714 9577 riscv-013.c:379 scan(): 41b 4i r 00000000 @11 -> + 00000000 @11
Debug: 715 9619 riscv-013.c:379 scan(): 41b 4i - 00000000 @11 -> + 00003082 @11
Debug: 716 9619 riscv-013.c:390 scan():  ->  allunavail anyunavail authenticated version=2
Error: 717 9619 riscv-013.c:2961 riscv013_is_halted(): Hart 0 is unavailable.
[...]
```

timsifive, 2019-03-18:
OpenOCD currently has no concept of debug index to hart id mapping. It assumes they're equal and, as you discovered, that they're continuous starting at 0. This is a problem and needs to be fixed. I have a bit of a backlog, however, and can't tell when I'll get to it.

If you want to take a look at this code, I would think the first step is to mess with riscv-013.c examine() so that it checks for the unavailable bits and doesn't terminate discovery. Then you have the problem that hart index 0 should be ignored altogether.

You may be able to work around this problem by explicitly specifying the hart id in the OpenOCD configuration file. See e.g.https://github.com/riscv/riscv-tests/blob/master/debug/targets/RISC-V/spike-2.cfg which configures core ID 0 and 1. I'm not sure if anything is going to break if the numbers aren't contiguous.

bluewww, 2019-03-18:
Don't you think we should establish a mapping between continuous "debug indices" to the existing harts in `examine()` and then go from there?

Another (ugly) way, is to establish a one to one correspondence between "debug indices" to hart id and then do the stuff of disabling harts. Infact currently I'm setting `RISCV_MAX_HARTS` to 1024 because I'm my setup (due to legacy reasons) the first available hart number might be as high as 992. This doesn't work for `-rtos` though.

timsifive, 2019-03-18:
Yes, there should be a mapping from indices to harts.

Some day, when `-rtos riscv` can go away and everybody will use `-rtos hwthread` instead, then RISCV_MAX_HARTS can be gotten rid of altogether.

timsifive, 2020-03-30:
This is a duplicate of #278.

closed 2020-03-30 by timsifive