riscv-collab/riscv-openocd #360: Unable to single step via gdb due to wrong PC?

lbmeng, 2019-03-21:
When debugging RV64 application in S-mode with SV39 virtual addressing, the GDB sees the PC is not the full PC address with 64-bit width, but a truncated one with only 40-bit. This makes it gdb unable to single step or disassemble the instructions at current PC.

Please check the following gdb output:

>>> p/x $pc
$19 = 0xe00003159c
>>> p/x $dpc
$20 = 0xe00003159c
>>> si
Warning:
Cannot insert breakpoint 0.
Cannot access memory at address 0xe00003159e

The PC can't be 0xe00003159c as it is not a canonical address. The actually PC is 0xffffffe00003159c.

Is this a known issue? From the debug spec, it says the dpc register is MXLEN width, but it does not mention MXLEN is which number.

aswaterman, 2019-03-21:
Which RISC-V machine are you using?

lbmeng, 2019-03-22:
It's FU540 core on HiFive Unleashed board.

aswaterman, 2019-03-22:
Yes, that's a known erratum in the silicon, related specifically to the handling of `dpc`.  It would be possible to make an openocd patch that corrects for this bug by unconditionally sign-extending from bit 39 of `dpc`, but it is not an openocd bug.

lbmeng, 2019-03-22:
Oops, hit a bug of the silicon. :)

Is this erratum publicly documented anywhere?

aswaterman, 2019-03-22:
I think not all errata are published yet, but this one is, see Sec. 25.4 of https://static.dev.sifive.com/FU540-C000-v1.0.pdf

lbmeng, 2019-03-22:
Great, I see the errata. Thanks for the quick response!

closed 2019-03-22 by lbmeng