riscv-collab/riscv-openocd #362: OpenOCD cannot connect to J-Link?

lbmeng, 2019-03-26:
HiFive Unleashed board, using OpenOCD to connect the board via the J-Link

> Open On-Chip Debugger 0.10.0+dev-00530-gaf3a034-dirty (2019-03-22-04:06)
> Licensed under GNU GPL v2
> For bug reports, read
> 	http://openocd.org/doc/doxygen/bugs.html
> adapter speed: 10000 kHz
> Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
> Info : J-Link V10 compiled Mar 21 2019 15:43:57
> Info : Hardware version: 10.10
> Info : VTarget = 0.000 V
> Info : clock speed 10000 kHz
> Error: JTAG scan chain interrogation failed: all ones
> Error: Check JTAG interface, timings, target power, etc.
> Error: Trying to use configured scan chain anyway...
> Error: riscv.cpu: IR capture error; saw 0x1f not 0x01
> Warn : Bypassing JTAG setup events due to errors
> Error: Unsupported DTM version: 15
> Info : Listening on port 3333 for gdb connections
> Error: Target not examined yet
> 
> Error: Unsupported DTM version: 15

Is this a known issue?

command-paul, 2019-03-26:
From a Jtag Perspective Looks like Your TDO is being held High ( your IR
capture also returns 0x1f and not 0x01).
Check your TDO/TCK connections , Ive noticed this When the the jtag target
is not powered up properly or has its reset asserted.

ᐧ

lbmeng, 2019-03-28:
> From a Jtag Perspective Looks like Your TDO is being held High ( your IR capture also returns 0x1f and not 0x01). Check your TDO/TCK connections , Ive noticed this When the the jtag target is not powered up properly or has its reset asserted. ᐧ

I am not sure what to do with this. This is the HiFive Unleashed board from SiFive. Is this a hardware issue? @aswaterman 

After googling, I see a similar question posted on SiFive's forum, but no answers.
https://forums.sifive.com/t/openocd-jtag-working-incorrectly-with-freedom-soc/1144

command-paul, 2019-03-28:
Just Noticed this , your VTarget is reporting 0.0 Volts , Is the Vcc/Ground
input to the Jlink Debugger floating ?
ᐧ

lbmeng, 2019-03-29:
> Just Noticed this , your VTarget is reporting 0.0 Volts , Is the Vcc/Ground input to the Jlink Debugger floating ? ᐧ

Thanks for the help!

It looks I inserted the 10-pin jtag adapter in the wrong direction. After I correct that, I now see VTarget is no longer zero, but still:

Open On-Chip Debugger 0.10.0+dev-00530-gaf3a034-dirty (2019-03-22-04:06)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : J-Link V10 compiled Mar 21 2019 15:43:57
Info : Hardware version: 10.10
Info : **VTarget = 1.216 V**
Info : clock speed 10000 kHz
Error: JTAG scan chain interrogation failed: all ones
Error: Check JTAG interface, timings, target power, etc.
Error: Trying to use configured scan chain anyway...
Error: riscv.cpu: IR capture error; saw 0x1f not 0x01
Warn : Bypassing JTAG setup events due to errors
Error: Unsupported DTM version: 15
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

Error: Unsupported DTM version: 15

shirlynan, 2019-11-10:
> > Just Noticed this , your VTarget is reporting 0.0 Volts , Is the Vcc/Ground input to the Jlink Debugger floating ? ᐧ
> 
> Thanks for the help!
> 
> I suspect I inserted the 10-pin jtag adapter in the wrong direction. After I changed that, I now see VTarget is no longer zero, but still:
> 
> Open On-Chip Debugger 0.10.0+dev-00530-gaf3a034-dirty (2019-03-22-04:06)
> Licensed under GNU GPL v2
> For bug reports, read
> http://openocd.org/doc/doxygen/bugs.html
> adapter speed: 10000 kHz
> Info : auto-selecting first available session transport "jtag". To override use 'transport select '.
> Info : J-Link V10 compiled Mar 21 2019 15:43:57
> Info : Hardware version: 10.10
> Info : **VTarget = 1.216 V**
> Info : clock speed 10000 kHz
> Error: JTAG scan chain interrogation failed: all ones
> Error: Check JTAG interface, timings, target power, etc.
> Error: Trying to use configured scan chain anyway...
> Error: riscv.cpu: IR capture error; saw 0x1f not 0x01
> Warn : Bypassing JTAG setup events due to errors
> Error: Unsupported DTM version: 15
> Info : Listening on port 3333 for gdb connections
> Error: Target not examined yet
> 
> Error: Unsupported DTM version: 15
> 
> Another attempt I saw the **VTarget is 1.742 V**. I don't know what's wrong :(

have u solved this problem?

iysheng, 2021-04-04:
I meet this mistake today , when i connect vref to 3.3 supply, it's ok, i can connect the riscv chip, maybe you can check your jtag wire connection.

en-sc, 2024-01-22:
@lbmeng, is the issue still relevant? If it is, please, provide OpenOCD verbose logs (via `-d3` flag or `debug_level 3` command, please, note that the log may be quite large, so it may be preferable to redirect it via `-l <filename` or `log_output <filename>`).
If the issue is not relevant anymore, I would like to close it.

closed 2024-01-26 by en-sc