riscv-collab/riscv-openocd #377: Support for multiple DMs

pdonahue-ventana, 2019-05-16:
Does the code support multiple DMs?  It appears that additional DMs can be added in get_dm() but:

- nothing appears to read the nextdm register to discover the base address of the next DM
- dm013_info_t doesn't appear to contain the base address of the DM
- calls to dmi_read() and dmi_write() appear to be to addresses like DMI_ABSTRACTCS rather than base+DMI_ABSTRACTCS.*

I'm not intimately familiar with the code so am I missing something?

Thanks,

-Paul

 *I also think it's weird that the #defines for DM registers use the DMI prefix rather than the DM prefix, but that's about code style rather than functionality.

timsifive, 2019-05-17:
The code does not support multiple DMs, but parts are written with that in mind.

zqb-all, 2023-05-05:
Are there any plans to support multiple DMs now? I haven't read the code yet, any suggestions if I want to support it?

zqb-all, 2023-05-05:
> @zqb-all: Do I understand correctly that you refer to case when there are multiple DMs on a single DMI bus? The first one located at DMI address 0, the other ones located on higher addresses specified by the `nextdm` DM register?

Yes, but I can't find anywhere in the code to use nextdm, so I guess there's no support for it right now

JanMatCodasip, 2023-05-05:
@zqb-all You're correct, this is still not supported as of now.

If there is need to implement that feature, this is roughly what it would take:

- The DMI address of the debug module would be passed as an extra argument to the `target create` command: 
  - `target create <NAME> riscv -chain-position <POS> -dmi-addr <DMI_BASE_ADDRESS> `
  - An example how to implement custom parameters to `target create` can be found here: https://github.com/riscv/riscv-openocd/blob/riscv/src/target/stm8.c#L1935
- There should be validity checks for the address.
  - That it is unique.
  - That such debug module really exists.
- The base address would the be used in `dmi_scan()` - added as a base to the offset.

zqb-all, 2023-05-05:
@JanMatCodasip  Thanks, I will try that.
I notice that target supports `-dbgbase` parameter, so maybe we could just use it  and don't need to add a `-dmi-addr `parameter.

zqb-all, 2023-05-06:
I try to add base in dmi_op_timeout
```c
diff --git a/src/target/riscv/riscv-013.c b/src/target/riscv/riscv-013.c
index 062fcd40d..5a84a2e2d 100644
--- a/src/target/riscv/riscv-013.c
+++ b/src/target/riscv/riscv-013.c
@@ -549,6 +549,9 @@ static int dmi_op_timeout(struct target *target, uint32_t *data_in,
                bool *dmi_busy_encountered, int dmi_op, uint32_t address,
                uint32_t data_out, int timeout_sec, bool exec, bool ensure_success)
 {
+       if (target->dbgbase_set)
+               address += target->dbgbase;
+
        select_dmi(target);

        dmi_status_t status;
```

config with
```
target create $_TARGETNAME_0 riscv -chain-position $_CHIPNAME.cpu  -dbgbase 0x0
target create $_TARGETNAME_2 riscv -chain-position $_CHIPNAME.cpu -dbgbase 0x400
```
then I can use 2 gdb to debug them, one gdb for one core

when i try to use smp
```
target create $_TARGETNAME_0 riscv -chain-position $_CHIPNAME.cpu -rtos hwthread -dbgbase 0
target create $_TARGETNAME_2 riscv -chain-position $_TARGETNAME -coreid 2 -dbgbase 0x400
target smp $_TARGETNAME_0 $_TARGETNAME_2
```
1 gdb can see 2 core, but continue always cause another core received signal SIGINT then Interrupt.
```
(gdb) i th
  Id   Target Id                                                      Frame
* 1    Thread 1 "riscv.cpu0" (Name: riscv.cpu0, state: debug-request) 0x000000004002386a in ?? ()
  2    Thread 3 "riscv.cpu2" (Name: riscv.cpu2, state: debug-request) 0x00000000400102b4 in ?? ()
(gdb) c
Continuing.

Thread 2 "riscv.cpu2" received signal SIGINT, Interrupt.
[Switching to Thread 3]
0x00000000400102b4 in ?? ()
(gdb) i th
  Id   Target Id                                                      Frame
  1    Thread 1 "riscv.cpu0" (Name: riscv.cpu0, state: debug-request) 0x000000004002386a in ?? ()
* 2    Thread 3 "riscv.cpu2" (Name: riscv.cpu2, state: debug-request) 0x00000000400102b4 in ?? ()
(gdb) c
Continuing.

Thread 1 "riscv.cpu0" received signal SIGINT, Interrupt.
[Switching to Thread 1]
0x000000004002386a in ?? ()
(gdb)
```

Can anyone give me some advice on using smp with multiple DMs?

JanMatCodasip, 2023-08-29:
Closing this thread since the feature has been implemented in: https://github.com/riscv/riscv-openocd/pull/875

closed 2023-08-29 by JanMatCodasip