riscv-collab/riscv-openocd #378: compilation error on macOS

filamoon, 2019-05-22:
I got the same error as #182, but the comments in it did not help.

discovery_tcp.c:305:9: error: implicit declaration of function 'select' is invalid in C99 [-Werror,-Wimplicit-function-declaration]
                ret = select(sock + 1, &rfds, NULL, NULL, &timeout);

Fresh checkout, branch riscv. Please help, thanks.

filamoon, 2019-05-23:
Ok, I solved it by adding one line to src/jtag/drivers/libjaylink/libjaylink/discovery_tcp.c

#include <sys/select.h>

TM1234, 2020-01-17:
Might be useful here too?

https://github.com/riscv/riscv-openocd/issues/423#issuecomment-575409664

agrobman, 2020-05-01:
got the same problem today. Solved this with the same include independently before seeing this ticket. 
@timsifive , could you, please,  just add this include in the file?
Not sure why it compiles on other platforms - the file uses external function without its prototype defined.

Mac uses apple version of  clang instead if GCC, which is more stringent than GCC, I guess.
My peer installed GCC on his Mac, and could compile the tool without this problem ..

timsifive, 2020-05-01:
@agrobman Can you submit a PR for this?

agrobman, 2020-05-01:
I can’t , I have other changes in my files

Alex

en-sc, 2024-01-22:
@filamoon, @agrobman, I would like to close this one. [The fix](https://gitlab.zapb.de/libjaylink/libjaylink/-/commit/28007268ac4f6035041d291e6349ae558c878ed6) is present in currently used version of `libjaylink`.

closed 2024-01-26 by en-sc