riscv-collab/riscv-openocd #385: bscan tunneling

olofk, 2019-06-28:
Hi,

I'm trying to use the bscan tunneling for RISC-V but I'm unable to get it going and at this point I have no clue about what's wrong. OpenOCD detects the FPGA just fine. I have pulled out tck, tms, tdi and tdo from bscane2 and expose them on pins and they look fine on the scope.

I also know that the debug module works fine since I can talk to it in simulation using jtag_vpi. 

I'm starting to wonder if there is some missing glue logic in the FPGA. Any clues?

Also curious if whoever implemented this looked at the bscan stuff we implemented for openrisc? In addition to bscan we also support altera's virtual jtag adapter

We have had something similar for

SiFiveGregS, 2019-07-02:
Hi, I implemented the software for driving the interface, but as far as the hardware details, I'll have to defer to some colleagues.  I'll ask if they can address your questions here.

olofk, 2019-07-02:
After gaining a better understanding of how it all fit together I decided to go with the lowRISC route instead and just instantiate three bscane2 primitives and use the `riscv set_ir` command instead. I think that's by far the most straight-forward method under the assumption that you don't need more than 4 instruction registers. So I'm fine with closing this for now. Eventually I would love to see a target-agnostic proxy-through-dr function in openocd but I will happily leave that to someone else

closed 2019-09-17 by timsifive