riscv-collab/riscv-openocd #391: RESUME_REQ and HALT_REQ simultanously

rajeshkrmeena, 2019-07-18:
I tried 'stepping' after implementation of system bus access (earlier its works fine). I found that Openocd is serving **resume** and **halt request** simultaneously.  

OPENOCD LOG :
Error: 34180 139097 riscv-013.c:3467 riscv013_step_or_resume_current_hart(): unable to resume hart 0
Debug: 34181 139097 riscv-013.c:379 scan(): 41b 10i r 00000000 @10 -> + 00000000 @11
Debug: 34182 139097 riscv-013.c:379 scan(): 41b 10i - 00000000 @10 -> + 40000001 @10
Debug: 34183 139097 riscv-013.c:390 scan():  ->  resumereq dmactive
Error: 34184 139097 riscv-013.c:3470 riscv013_step_or_resume_current_hart():   dmcontrol=0x40000001
Debug: 34185 139097 riscv-013.c:379 scan(): 41b 10i r 00000000 @11 -> + 00000000 @10
Debug: 34186 139097 riscv-013.c:379 scan(): 41b 10i - 00000000 @11 -> + 00430c82 @11
Debug: 34187 139097 riscv-013.c:390 scan():  ->  impebreak allresumeack anyresumeack allrunning anyrunning authenticated version=2
Error: 34188 139097 riscv-013.c:3473 riscv013_step_or_resume_current_hart():   dmstatus =0x00430c82
Error: 34189 139097 riscv-013.c:3476 riscv013_step_or_resume_current_hart():   was stepping, halting
Debug: 34190 139097 riscv-013.c:2890 riscv013_halt_current_hart(): halting hart 0
Debug: 34191 139097 riscv-013.c:379 scan(): 41b 10i r 00000000 @11 -> + 00000000 @11
Debug: 34192 139098 riscv-013.c:379 scan(): 41b 10i - 00000000 @11 -> + 00430c82 @11
Debug: 34193 139098 riscv-013.c:390 scan():  ->  impebreak allresumeack anyresumeack allrunning anyrunning authenticated version=2
Debug: 34194 139098 riscv-013.c:379 scan(): 41b 10i r 00000000 @10 -> + 00000000 @11
Debug: 34195 139098 riscv-013.c:379 scan(): 41b 10i - 00000000 @10 -> + 40000001 @10
Debug: 34196 139098 riscv-013.c:390 scan():  ->  resumereq ### dmactive
Debug: 34197 139098 riscv-013.c:379 scan(): 41b **10i w c0000001 @10** -> + 00000000 @10
Debug: 34198 139098 riscv-013.c:390 scan(): **haltreq resumereq dmactive** ->
Debug: 34199 139098 riscv-013.c:379 scan(): 41b 10i - 00000000 @10 -> + c0000001 @10
Debug: 34200 139098 riscv-013.c:390 scan():  -> haltreq resumereq dmactive
Debug: 34201 139098 riscv-013.c:379 scan(): 41b 10i r 00000000 @11 -> + 00000000 @10
Debug: 34202 139098 riscv-013.c:379 scan(): 41b 10i - 00000000 @11 -> + 00400c82 @11
Debug: 34203 139098 riscv-013.c:390 scan():  ->  impebreak allrunning anyrunning authenticated version=2

I did not get it. What is the sense of serving both simultaneously?

timsifive, 2019-07-18:
That looks like a bug in OpenOCD, although I suspect what's really causing you issues is the fact that hart 0 couldn't be resumed (first line in your excerpt).

This code has been fairly significantly rewritten not too long ago. Does this problem still occur if you build from the latest github source?

rajeshkrmeena, 2019-08-15:
Thanks @timsifive

closed 2019-08-15 by rajeshkrmeena