riscv-collab/riscv-openocd #395: Unaligned Address support

rajeshkrmeena, 2019-08-14:
As I understand, OPENOCD checked for Unaligned Address Support (UAS). If the design has not supported it(UAS) then OPENOCD breaks the data into small chunks with an aligned address. For example, if 0xcdac2019 (data) is write at 0x100b5 then OPENOCD breaks into 1-byte (at 0x100b5)+ 2byte (at 0x100b6) + 1byte (0x100b8) and write.

In my design, Unaligned Address Support is there, after that OPENOCD is sending data into small chunks (1-byte (at 0x100b5)+ 2byte (at 0x100b6) + 1byte (0x100b8) for the same). 

Why is OPENOCD breaking the data instead of sending full data ???  I looked into openocd logs but unable to get. (Note: In openocd code, there is checking mechanism for UAS. What is the use of that or UAS support if openocd always send  the data into small chunks.)

timsifive, 2019-08-14:
I suspect the behavior you're seeing comes from target.c, target_write_buffer_default().

rajeshkrmeena, 2019-08-16:
Dear Tim,
I looked into target.c. The target_write_buffer_default() is written in such a way that it always sends address aligned data i.e. breaking data into small chunks if an address is misaligned to 32 bits. It is not checking for Unaligned Address Support of Hardware but in 'riscv-013.c', riscv013_test_sba_config_reg() , Test 6 checking for support of misaligned address. If target_write_buffer_default() is essentially take-caring ( breaking data into small chunks if an address is misaligned to 32 bits ) then,

1. What is significant of TEST 6 of riscv013_test_sba_config_reg()  and sberror =3 (anyway misaligned address supported or not)?
2. What is the meaning of supporting Unaligned Address Access in the Hardware?
3. Is any future plan to modify the 'target_write_buffer_default()' to check Unaligned Support  (as per sberror=3)?

timsifive, 2019-08-16:
1. It looks like Test 6 is using a misaligned access to test that sberror gets set, even though the spec allows for unaligned accesses to be supported. That seems like a bug in the test.
2. If a bus allows unaligned accesses, then it makes sense to allow the debugger access to it. The debug spec is not limited to use cases that make sense with gdb/OpenOCD.
3. I have no such plans.

rajeshkrmeena, 2019-08-17:
Thanks for your valuable suggestions and comments.

closed 2019-08-17 by rajeshkrmeena