riscv-collab/riscv-openocd #398: flash write_image error on SiFive HiFive Unleashed

mcd500, 2019-09-02:
Hi, I encounter error of flash write_image with openocd on SiFive HiFive Unleashed board.

I was reading the instruction of installing coreboot on Unleashed.
https://doc.coreboot.org/mainboard/sifive/hifive-unleashed.html

Typing the command to write flash at openocd.
```
> flash write_image erase unlock hifive-unleashed-a00-1.0-2018-03-20.gpt 0x20000000
```
Typical messages, varies by trying many times.
```
auto erase enabled
auto unlock enabled
Found flash device 'issi is25wp256d' (ID 0x0019709d)
device needs paging or 4-byte addresses - not implemented
Algorithm returned error 1
error writing to flash at address 0x20000000 at offset 0x00000000
```
The gdb feature is working fine. Only the `flash write_image erase` is failing.
I would like to help resolving the error.

The following is the details of reproducing the errors.

* Download the OpenOCD script for Freedom Unleashed.
```
git clone https://github.com/riscv/riscv-openocd.git
cd riscv-openocd
./bootstrap
./configure
make install
wget https://raw.github.com/sifive/freedom-u-sdk/master/bsp/env/freedom-u500-unleashed/openocd.cfg
```
* Connect Unleashed and PC with USB-micro cable.
* Start OpenOCD
```
$ openocd -f openocd.cfg
```
log
```
Open On-Chip Debugger 0.10.0+dev-00626-g7eaf60f1b (2019-08-14-17:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 10000 kHz
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
Info : ftdi: if you experience problems at higher adapter clocks, try the command "ftdi_tdo_sample_edge falling"
Info : clock speed 10000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x20000913 (mfg: 0x489 (SiFive, Inc.), part: 0x0000, ver: 0x2)
Info : datacount=2 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Examined RISC-V core; found 5 harts
Info :  hart 0: XLEN=64, misa=0x8000000000101105
Info :  hart 1: XLEN=64, misa=0x800000000014112d
Info :  hart 2: XLEN=64, misa=0x800000000014112d
Info :  hart 3: XLEN=64, misa=0x800000000014112d
Info :  hart 4: XLEN=64, misa=0x800000000014112d
Info : Listening on port 3333 for gdb connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```
* Connect to OpenOCD’s command interface (via telnet) and enter the line marked with > :
```
$ telnet localhost 4444
Trying 127.0.0.1...
Connected to localhost.
Escape character is '^]'.
Open On-Chip Debugger
```
* Type the command to write flash.
```
> flash write_image erase unlock hifive-unleashed-a00-1.0-2018-03-20.gpt 0x20000000
```
* Then it fails with error messages.
Typical messages, varies by trying many times.
-- error message 1 --
```
auto erase enabled
auto unlock enabled
Found flash device 'issi is25wp256d' (ID 0x0019709d)
device needs paging or 4-byte addresses - not implemented
Algorithm returned error 1
error writing to flash at address 0x20000000 at offset 0x00000000
```
-- error message 2 --
```
auto erase enabled
auto unlock enabled
Algorithm returned error 268500992
error writing to flash at address 0x20000000 at offset 0x00000000
```
-- error message 3 --
```
auto erase enabled
auto unlock enabled
Algorithm returned error 4
error writing to flash at address 0x20000000 at offset 0x00000000
```
-- error message 4 --
```
Disabling abstract command writes to CSRs.
Algorithm timed out after 10000 ms.
  now   = 0xb6d20667
  start = 0xb6d1df56
zero = 0x0
ra = 0xffffffe000c147e0
sp = 0xffffffe1b7ad3f80
gp = 0xffffffe00104e980
tp = 0xffffffe1b7aaec00
t0 = 0x0
t1 = 0x6000
t2 = 0x2
s0 = 0xffffffe1b7ad3f90
s1 = 0xffffffe00104e9f8
a0 = 0x10040000
a1 = 0x100
a2 = 0x80000320
a3 = 0x0
a4 = 0x23f0
a5 = 0x0
a6 = 0xffffffe001026c68
a7 = 0xffffffe001049460
s2 = 0x10
s3 = 0xffffffe00104ebbc
s4 = 0xffffffe0010494e2
s5 = 0xffffffe000f98690
s6 = 0x1
s7 = 0x0
s8 = 0x0
s9 = 0x0
s10 = 0x0
s11 = 0x0
t3 = 0x40
t4 = 0x344
t5 = 0x4
t6 = 0x40
pc = 0x80000004
Failed to execute algorithm at 0x80000000: -302
error writing to flash at address 0x20000000 at offset 0x00000000
```
I also tried to use old commit in openocd if it was working in the old openocd.
```
$ git checkout 91faf1a573d6931cc2f563751421330944243b68
Author: Tim Newsome <tim@sifive.com>
Date:   Fri Jun 21 09:59:28 2019 -0700

    Reduce abstract command execution by one scan. (#383)
```
-- error message 5 --
```
auto erase enabled
auto unlock enabled
PC ended up at 0x80004b8 instead of 0x80000004
Failed to execute algorithm at 0x80000000: -4
error writing to flash at address 0x20000000 at offset 0x00000000
```
Tried the different commit.
```
$ git checkout 6983eda0e9d83c49af1366f4363efb570f5b79bd
Author: Tim Newsome <tim@sifive.com>
Date:   Mon Jul 15 10:32:28 2019 -0700

    Make resume order configurable. (#388)
```

```
> flash write_image erase unlock hifive-unleashed-a00-1.0-2018-03-20.gpt 0x20000000
```

-- error message 6 --
```
auto erase enabled
auto unlock enabled
timeout
error writing to flash at address 0x20000000 at offset 0x00000000
```

Also we've tried ARM-USB-TINY-H adaptor connected to the on-board JTAG 10pin connector and a slightly modified openocd.cfd for ARM-USB-TINY-H. The result looks similar with same error messages.

timsifive, 2019-09-03:
There is some issue with flashing HiFive Unleashed using the latest OpenOCD. It's my top priority right now.

timsifive, 2019-09-09:
Fixed by #402, which just merged.

mcd500, 2019-09-10:
Thanks, I will try it, preferably tomorrow. It is nice to see `-march=rv32e` is starting to play already for who is happy to see RISC-V getting better. :)

mcd500, 2019-09-12:
Hi @timsifive, we confirmed that the https://github.com/riscv/riscv-openocd/pull/402 fixing the `flash write_image` command on openocd for writing spi flash on HiFive Unleashed board. Great! This gives everyone to develop and customize the bootrom code (zsbl) of HiFive Unleashed with using only openocd. Thank you again.

closed 2019-09-12 by mcd500