riscv-collab/riscv-openocd #401: Openocd with Xilinx Platform Cable

rajeshkrmeena, 2019-09-09:
I tried to run Openocd with Xilinx Platform Cable. I used  'usb-jtag.cfg' for configuring Platform  . But failed to run Openocd. 
The following error message is shown with RISC-V Openocd and normal Openocd 

**RISC-V Openocd**
....
Debug: 207 5 command.c:364 register_command_handler(): registering 'ocd_reset_nag'...
Debug: 208 5 command.c:364 register_command_handler(): registering 'ocd_ps'...
Debug: 209 5 command.c:364 register_command_handler(): registering 'ocd_test_mem_access'...
Error: 210 5 usb_blaster.c:861 ublast_init(): **No lowlevel driver available**
Debug: 211 5 command.c:651 run_command(): Command 'init' failed with error code -107
User : 212 5 command.c:715 command_run_line(): 
Debug: 213 5 riscv.c:460 riscv_deinit_target(): riscv_deinit_target()
Debug: 214 5 riscv-013.c:1463 deinit_target(): riscv_deinit_target()
Debug: 215 5 target.c:1959 target_free_all_working_areas_restore(): freeing all working areas

**Openocd (Normal)**
....
Debug: 83 5 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_init
Debug: 84 5 command.c:143 script_debug(): command - init ocd_init
Debug: 86 5 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_target init
Debug: 87 5 command.c:143 script_debug(): command - ocd_target ocd_target init
Debug: 89 5 command.c:143 script_debug(): command - ocd_command ocd_command type ocd_target names
Debug: 90 5 command.c:143 script_debug(): command - ocd_target ocd_target names
Debug: 91 5 target.c:1324 handle_target_init_command(): Initializing targets...
Info : 92 5 ublast_access_ftdi.c:85 ublast_ftdi_init(): usb blaster interface using libftdi
Error: 93 11 ublast_access_ftdi.c:91 ublast_ftdi_init(): **unable to open ftdi device: device not found**
Debug: 94 11 command.c:626 run_command(): Command failed with error code -100
User : 95 11 command.c:687 command_run_line():

timsifive, 2019-09-09:
This looks like OpenOCD can't find the USB device. If this is on Linux, I'd look at permissions on the USB devices. You might need to update the udev rules. I don't know what might cause this error on Windows/Mac.

In any case this is not a RISC-V-specific problem. You might have better luck posting on general OpenOCD support forums, or searching for the error message you encountered.

closed 2019-09-09 by rajeshkrmeena

march-791, 2025-02-25:
> Thanks

Did it work?