riscv-collab/riscv-openocd #407: write_memory_progbuf() sends data while target is busy

niwis, 2019-09-13:
Hi,

while loading into slow memory (using the gdb command `load`) we ran into a problem. While the openocd debug log shows a success, some of the memory regions are not loaded properly. Apparently, openocd attempts to write to `data0` before `abstractcs.busy` is cleared. This is problematic, since `data0` might still be accessed from the program buffer.
Should `abstractcs` be polled inside the write loop of `write_memory_progbuf()` or what's the proper way to handle this?

Best,
Nils

timsifive, 2019-09-13:
Per the spec definition of data0:

> Accessing these registers while an abstract command is executing causes cmderr to be set to 1
(busy) if it is 0.
> Attempts to write them while busy is set does not change their value.

So if OpenOCD writes data0 while an abstract command is executed, the debug module should set cmderr and ignore the value written.

timsifive, 2020-03-30:
Closed due to inactivity.

closed 2020-03-30 by timsifive