riscv-collab/riscv-openocd #412: Riscv compliance tests fails - NDMRESET should move DPC to reset value.

ctt-wis, 2019-10-02:
Hello,
My risc-v implementation has only one hart (Hart ID = 0) and boot address is 0x0.
When running riscv test_compliance, I get FAILED at test 284:
"Executing test 284 (riscv-013.c:4540): NDMRESET should move DPC to reset value."
Looking at the code (riscv-013.c:4539):
   `COMPLIANCE_MUST_PASS(register_read_direct(target, &value, GDB_REGNO_DPC));
    COMPLIANCE_TEST(bogus_dpc != value, "NDMRESET should move DPC to reset value.");`
the failure is due to dpc being same value as bogus_dpc whis is equal to 0x0. 
Why is bogus_dpc set to 0x0 in previous step (riscv-013.c:4480)?:
```
if (hartsel == 0)
			bogus_dpc = dpc; /* For a later test step `*/`
```
Thanks for your help,
CTt

timsifive, 2019-10-03:
I don't know, but the core it was written for didn't use 0 as the reset vector. In general the compliance test code is incomplete and not really supported. I'll add a note to that effect.

timsifive, 2021-01-19:
Compliance testing is removed as of #567.

closed 2021-01-19 by timsifive