riscv-collab/riscv-openocd #415: Test if hart has support for `fence.i`

MarekPikula, 2019-10-10:
As raised in lowRISC/ibex#365 hart that doesn't support _Zifencei_ extension may raise an illegal instruction exception when `fence.i` instruction is executed by OpenOCD. This can lead to unexpected behaviour on a given platform, like resetting some registers (in this case `mstatus`), because of entering exception handler.

It would be nice if OpenOCD could determine if `fence.i` is supported for given hart. I am aware that there is no flag in `misa` for that, but it could be verified by checking if exception was raised by processor after executing `fence.i`.

In case of [Ibex](https://github.com/lowRISC/ibex) processor with [pulp-platform/riscv-dbg](https://github.com/pulp-platform/riscv-dbg) as a debug core, after encountering illegal instruction the core enters [_exception](https://github.com/pulp-platform/riscv-dbg/blob/master/debug_rom/debug_rom.S#L62) routine and `cmderror` flag is risen in DM.

timsifive, 2019-10-10:
That would be a good improvement, but I don't have time for it right now.

Note that fence.i shouldn't have any side effects when executed in debug mode. The spec says in the Debug Mode section: "Exceptions don’t update any registers. That includes cause, epc, tval, dpc, and mstatus.
They do end execution of the Program Buffer."

aap-sc, 2024-06-05:
> It would be nice if OpenOCD could determine if fence.i is supported for given hart. I am aware that there is no flag in misa for that, but it could be verified by checking if exception was raised by processor after executing fence.i.

This is what OpenOCD does after https://github.com/riscv-collab/riscv-openocd/issues/933 .

However, I believe we should just introduce a configuration parameter to allow user to specify if execution of ifence is not desired. I don't believe auto-detection of this feature is reasonable, since it may be way too intrusive to execute ifence . Maybe in future the situation could be addressed by "Uniform Discovery" specification