riscv-collab/riscv-openocd #423: How to cross compile a Windows version openocd

xfdzjz, 2019-11-19:
When I was trying to generate a Windows version openocd.exe using cross compiling, but the terminal show me I need a file named windows.h, I can not find the file. And one more question is that I want to generate a win32 version Windows openocd on PC, a normal version is fine, I am not sure how to write the HOST, could you give me some tips? Thanks for your help

timsifive, 2019-11-19:
I've never built a Windows binary of OpenOCD. I suggest you ask at one of the resources mentioned at http://openocd.org/discussion/

TM1234, 2020-01-17:
Bit late to the party but Liviu's xPack Project docker based build scripts for OpenOCD (and other stuff such as Arm and RISC-V embedded/bare metal GCC toolchains) are very useful in this respect and give predictable and reliable results in my experience. 

https://xpack.github.io/openocd/
https://xpack.github.io/openocd/#build-details

By default the scripts build from his repos/snapshots but with a little hacking you can get them to build from alternative sources - e.g. the SiFive OpenOCD fork repo. If you need any guidance post back here.

Hzv587, 2020-02-21:
> Bit late to the party but Liviu's xPack Project docker based build scripts for OpenOCD (and other stuff such as Arm and RISC-V embedded/bare metal GCC toolchains) are very useful in this respect and [...]

Hi, can you please give me more guidance?

TM1234, 2020-02-21:
> Hi, can you please give me more guidance?

What do you mean?
Liviu already covers his build process in detail in the links that I posted.

TM1234, 2020-02-21:
Oh sorry - I presume you mean about fetching the sources from somewhere other than Liviu's repos? Well you can uncomment these lines to build from the master repos:

https://github.com/xpack-dev-tools/openocd-xpack/blob/xpack/scripts/defs-source.sh#L43

timsifive, 2020-03-30:
Closed due to inactivity.

closed 2020-03-30 by timsifive