riscv-collab/riscv-openocd #438: Authorisation problem when trying to run programs

joeski65, 2020-01-06:
Whenever I try and run a program that has been loaded into memory using gdb attached to openocd (using monitor reset [halt|run]), I get an authorisation error:

`Error: Debugger is not authenticated to target Debug Module. (dmstatus=0x1). Use `riscv authdata_read` and `riscv authdata_write` commands to authenticate.`

Although `riscv authdata_read` returns 0, my dmstatus is 0x382, whereby the authentication bit is set. Any attempts to `riscv authdata_write 1` still result in the value being 0.

It may be of note that test_compliance tests pass until the test HALTSUM1.

Trying to step through the program does nothing as the program counter never moves.

What could be the issue here? Thanks!

timsifive, 2020-01-06:
This sounds like your reset also resets the Debug Module, which isn't supposed to happen.

If you run OpenOCD with `-d` you'll get output containing every scan that's done, and you should be able to look at what leads up to OpenOCD getting 0x1 for dmstatus.

joeski65, 2020-01-07:
I've trawled through the log from when the reset was initiated, and I don't see anything immediately out of place (though, I wouldn't know).

From what I understand, calling `reset halt` invokes the following:

1. jtag arp_init-reset
2. arp_examine
3. arp_reset assert 1
4. arp_reset deassert 1

From the looks of it, it's the 'arp_reset deassert 1' that's resetting the Debug Module. Is both the arp_reset assert and deassert supposed to happen?

Perhaps of importance: `Caller passed out-of-range JEP106 ID!` This looks fine? just results in `JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)`

I've added a few newlines in the log to hopefully make it a little more intuitive:
```
Debug: 5829 33305 command.c:143 script_debug(): command - reset reset halt
Debug: 5830 33305 riscv.c:1926 riscv_openocd_poll(): polling all harts
Debug: 5831 33305 riscv.c:2864 riscv_set_current_hartid(): setting hartid to 0, was 0
Debug: 5832 33305 riscv.c:1880 riscv_poll_hart(): polling hart 0, target->state=2
Debug: 5833 33305 riscv-013.c:397 scan(): 41b 0i r 00000000 @11 -> + 00000382 @00
Debug: 5834 33305 riscv-013.c:397 scan(): 41b 0i - 00000000 @11 -> + 00000382 @00
Debug: 5836 33305 gdb_server.c:401 gdb_put_packet_inner(): sending packet '$O#4f'
Debug: 5837 33305 target.c:1634 target_call_reset_callbacks(): target reset 2 (halt)
Debug: 5838 33305 command.c:143 script_debug(): command - target target names
Debug: 5839 33305 command.c:143 script_debug(): command - riscv.cpu riscv.cpu invoke-event reset-start
Debug: 5840 33305 command.c:143 script_debug(): command - transport transport select

Debug: 5841 33305 command.c:143 script_debug(): command - jtag jtag arp_init-reset
Debug: 5842 33306 core.c:1508 jtag_init_reset(): Initializing with hard TRST+SRST reset
Debug: 5843 33306 core.c:742 jtag_add_reset(): JTAG reset with TLR instead of TRST
Debug: 5844 33306 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 5845 33306 core.c:1394 jtag_init_inner(): Init JTAG chain
Debug: 5846 33306 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 5847 33306 core.c:1060 jtag_examine_chain(): DR scan interrogation for IDCODE/BYPASS
Debug: 5848 33306 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 5849 33306 jep106.c:33 jep106_manufacturer(): BUG: Caller passed out-of-range JEP106 ID!
Info : 5850 33306 core.c:959 jtag_examine_chain_display(): JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Debug: 5851 33306 gdb_server.c:401 gdb_put_packet_inner(): sending packet '$O4a544147207461703a2072697363762e637075207461702f64657669636520666f756e643a203078303030303030303120286d66673a20307830303020283c696e76616c69643e292c20706172743a203078303030302c207665723a20307830290a#3e'
Debug: 5852 33306 core.c:1190 jtag_validate_ircapture(): IR capture validation scan
Debug: 5853 33307 core.c:1248 jtag_validate_ircapture(): riscv.cpu: IR capture 0x01

Debug: 5854 33307 command.c:143 script_debug(): command - transport transport select
Debug: 5855 33307 command.c:143 script_debug(): command - riscv.cpu riscv.cpu cget -chain-position
Debug: 5856 33307 command.c:143 script_debug(): command - jtag jtag tapisenabled riscv.cpu
Debug: 5857 33307 command.c:143 script_debug(): command - riscv.cpu riscv.cpu invoke-event examine-start
Debug: 5858 33307 command.c:143 script_debug(): command - riscv.cpu riscv.cpu arp_examine allow-defer
Debug: 5859 33307 riscv.c:1006 riscv_examine(): riscv_examine()
Debug: 5860 33307 riscv.c:1008 riscv_examine(): Target was already examined.

Debug: 5861 33307 command.c:143 script_debug(): command - riscv.cpu riscv.cpu invoke-event examine-end
Debug: 5862 33307 command.c:143 script_debug(): command - riscv.cpu riscv.cpu invoke-event reset-assert-pre
Debug: 5863 33307 command.c:143 script_debug(): command - transport transport select
Debug: 5864 33307 command.c:143 script_debug(): command - riscv.cpu riscv.cpu cget -chain-position
Debug: 5865 33307 command.c:143 script_debug(): command - jtag jtag tapisenabled riscv.cpu
Debug: 5866 33307 command.c:143 script_debug(): command - riscv.cpu riscv.cpu arp_reset assert 1
Debug: 5867 33307 target.c:1966 target_free_all_working_areas_restore(): freeing all working areas
Debug: 5868 33307 riscv.c:1168 riscv_assert_reset(): [0]
Debug: 5869 33307 riscv.c:2880 riscv_invalidate_register_cache(): [0]
Debug: 5870 33307 riscv-013.c:397 scan(): 41b 0i w 80000003 @10 -> + 00000382 @00
Debug: 5871 33307 riscv-013.c:408 scan(): haltreq ndmreset dmactive -> 
Debug: 5872 33308 riscv-013.c:397 scan(): 41b 0i - 00000000 @10 -> + 00000001 @00

Debug: 5873 33308 command.c:143 script_debug(): command - riscv.cpu riscv.cpu invoke-event reset-assert-post
Debug: 5874 33308 command.c:143 script_debug(): command - riscv.cpu riscv.cpu invoke-event reset-deassert-pre
Debug: 5875 33308 command.c:143 script_debug(): command - transport transport select
Debug: 5876 33308 command.c:143 script_debug(): command - riscv.cpu riscv.cpu cget -chain-position
Debug: 5877 33308 command.c:143 script_debug(): command - jtag jtag tapisenabled riscv.cpu
Debug: 5878 33308 command.c:143 script_debug(): command - riscv.cpu riscv.cpu arp_reset deassert 1
Debug: 5879 33308 target.c:1966 target_free_all_working_areas_restore(): freeing all working areas
Debug: 5880 33308 riscv.c:1176 riscv_deassert_reset(): [0]
Debug: 5881 33308 riscv-013.c:397 scan(): 41b 0i w 80000001 @10 -> + 00000001 @00
[... cut ...] 1757468656e7469636174652e0a#ad'
Debug: 5889 33309 command.c:630 run_command(): Command 'reset' failed with error code -4
User : 5890 33310 command.c:695 command_run_line(): 
Debug: 5891 33310 gdb_server.c:401 gdb_put_packet_inner(): sending packet '$O0a#e0'
Debug: 5892 33310 riscv.c:1926 riscv_openocd_poll(): polling all harts
Debug: 5893 33310 riscv.c:2864 riscv_set_current_hartid(): setting hartid to 0, was 0
Debug: 5894 33310 riscv.c:1880 riscv_poll_hart(): polling hart 0, target->state=3
```

timsifive, 2020-01-07:
On Tue, Jan 7, 2020 at 9:12 AM joeski65 <notifications@github.com> wrote:

>
> Debug: 5833 33305 riscv-013.c:397 scan(): 41b 0i r 00000000 @11 -> + 00000382 @00
> Debug: 5834 33305 riscv-013.c:397 scan(): 41b 0i - 00000000 @11 -> + 00000382 @00
>
> Here we read a reasonable value of dmstatus.

...

> Debug: 5884 33308 riscv-013.c:1898 deassert_reset(): Waiting for hart 0 to halt out of reset.
> Debug: 5885 33308 riscv-013.c:397 scan(): 41b 0i r 00000000 @11 -> + 00000001 @00
> Debug: 5886 33308 riscv-013.c:397 scan(): 41b 0i - 00000000 @11 -> + 00000001 @00
>
> And here we read 0x1 for dmstatus, which can't be right. Bits 3:0 at least
must be 2 for the version.
The only time that might not be the case is when dmactive is cleared, but
that didn't happen.

It still looks like writing ndmreset as a side effect did something that
messed up the next value read from dmstatus.

>

closed 2020-03-18 by joeski65

makermuyi, 2020-12-28:
I have the same problem and don't know how to solve it. Here is the log generated after starting OpenOCD in Eclipse.
xPack OpenOCD, x86_64 Open On-Chip Debugger 0.10.0+dev (2020-10-13-17:29)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : J-Link ARM V8 compiled Nov 28 2014 13:44:46
Info : Hardware version: 8.00
Info : VTarget = 3.306 V
Info : clock speed 3000 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Info : datacount=1 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=32, misa=0x40101124
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Started by GNU MCU Eclipse
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : accepting 'gdb' connection on tcp/3333
Info : Disabling abstract command reads from FPRs.
Warn : Prefer GDB command "target extended-remote 3333" instead of "target remote 3333"
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Error: Debugger is not authenticated to target Debug Module. (dmstatus=0x0). Use `riscv authdata_read` and `riscv authdata_write` commands to authenticate.

Error: Debugger is not authenticated to target Debug Module. (dmstatus=0x0). Use `riscv authdata_read` and `riscv authdata_write` commands to authenticate.
Error: Debugger is not authenticated to target Debug Module. (dmstatus=0x0). Use `riscv authdata_read` and `riscv authdata_write` commands to authenticate.
Error: Debugger is not authenticated to target Debug Module. (dmstatus=0x0). Use `riscv authdata_read` and `riscv authdata_write` commands to authenticate.
Error: Debugger is not authenticated to target Debug Module. (dmstatus=0x0). Use `riscv authdata_read` and `riscv authdata_write` commands to authenticate.
Error: Debugger is not authenticated to target Debug Module. (dmstatus=0x0). Use `riscv authdata_read` and `riscv authdata_write` commands to authenticate.
.........................
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
Error: failed read at 0x11, status=1
...........

timsifive, 2020-12-28:
> dmstatus=0x0

That is not a valid value for dmstatus. At a minimum the version field should contain 1, 2, or 3. This looks like a hardware problem, or possibly a problem with your JTAG connection to the target.

makermuyi, 2021-05-09:
Because this is a newly developed chip-based on RISC-V instruction architecture, the reason for this problem is caused by hardware.