riscv-collab/riscv-openocd #440: Single hart reset uses ndmreset instead of hartreset

jbdubois-safran, 2020-01-08:
Current riscv reset implementation uses NDMRESET bit of DMCONTROL to reset a hart, which result is a full system reset (except DM). This could be an intended behavior in SMP configurations but is not in non-SMP.
In non-SMP system, **HARTRESET bit shall be used**.

Also, when performing a _reset_ and not a _reset halt_, HALTREQ bit should be cleared before releasing HARTRESET. Current code de-assert HALTREQ and expect an ALLRUNNING status, which never happens is hart was halted before reset.

mwachs5, 2020-01-08:
even in a non-SMP system, users may want to reset the entire system not just the hart.

jbdubois-safran, 2020-01-08:
Is there a specific command to reset only selected harts?
Maybe system-wide reset should only be performed at init.

timsifive, 2020-01-08:
The OpenOCD reset command resets "[a]ll defined targets" (http://openocd.org/doc/html/General-Commands.html). Fixing that would require making target-independent changes, which ideally come down from the OpenOCD project. Having said that, if there is a real need/code I'm fine accepting such a change into this fork, but the long-term goal is to merge everything up so we don't need a separate RISC-V fork of OpenOCD.

Once OpenOCD has support for resetting individual targets, adding RISC-V support to use hartreset should be relatively straightforward.

jbdubois-safran, 2020-01-09:
Thanks for the heads up. We may end up using external hardware reset signals for individual cores if the changes in OpenOCD are too heavy, or change the behavior of reset command for our internal needs.

What about the second issue I have: 
> Also, when performing a reset and not a reset halt, HALTREQ bit should be cleared before releasing HARTRESET. Current code de-assert HALTREQ and expect an ALLRUNNING status, which never happens is hart was halted before reset.

allexoll, 2021-11-25:
@timsifive could this issue be resolved by a specific `riscv hart_reset 0/1/n` subcommand?

timsifive, 2021-11-30:
Yeah, a RISC-V-specific subcommand would work. I'd accept a change that implements that.