riscv-collab/riscv-openocd #443: warning sysctl.h stop compilation

damiencarrier, 2020-01-27:
Hello,
I have a warning and cc1 says: all warnings being treated as errors
I don't find how to disable it in the makefile.
What can i do?
Thank you :)

[image]

TM1234, 2020-01-27:
Remove -Werror?
I don't get that warning/compilation failure when compiling the RISC-V openocd fork.
Have you modified the code to cause this or something?

rtwfroody, 2020-01-27:
Specifically, when running configure you can add `--disable-werror` to the command line to ignore warnings.

towoe, 2020-02-07:
The problem is glibc 2.30 deprecating `sys/sysctl.h`.
@damiencarrier probably uses a distribution which already uses the new glibc version.

Upstream fix http://openocd.zylin.com/#/c/5317/

damiencarrier, 2020-02-07:
Hello,
Thank you, i re-download openocd and find the -Werror to remove it in the makefile.

towoe, 2020-02-10:
Yes, removing `-Werrror` will help as it disables the handling of warnings as errors. But this should not be the final solution.
Not sure about the process of importing changes from upstream, but it would make sense to incorporate the upstream fix.

closed 2021-07-19 by timsifive