riscv-collab/riscv-openocd #445: What is general way to access CPU internal RAMs?

agrobman, 2020-02-05:
If a CPU complex includes internal RAMs,  which may be configurable in size and base addresses, how does openOCD supposed to access these?

There are two ways to access memories - System Bus commands and abstract memory commands,
How does openOCD select which one to use?

Related topic : access size. Some memories ( internal or external) may be limited to specific access size,
for ex only accessed with 4 bytes (word),  how does openOCD "know" that? 
Is there way for openOCD "discover" these limitations?

If a CPU doesn't provide "program buffer",  how will openOCD set SW breakpoint if the CPU has instruction cache?

 Is there defined common way for debugger to invalidate caches, without running instructions or have specific knowledge about cache control registers?

How does openOCD discovers what is implemented in processor DM module?
Is the discovery done while CPU is in reset/halted/running?

Is there any configuration file, which can be read by the openOCD at startup to provide processor implementation details, including privet CSRs internal memories, features implemented, etc ?

timsifive, 2020-02-05:
> If a CPU complex includes internal RAMs, which may be configurable in size and base addresses, how does openOCD supposed to access these?

The same way all memory is accessed.

> There are two ways to access memories - System Bus commands and abstract memory commands,
> How does openOCD select which one to use?

Defaults to program buffer, uses System Bus if that doesn't work or if prefer_sba is set, and if all that fails then uses abstract.

> Related topic : access size. Some memories ( internal or external) may be limited to specific access size,
> for ex only accessed with 4 bytes (word), how does openOCD "know" that?
> Is there way for openOCD "discover" these limitations?

Not AFAIK.

> If a CPU doesn't provide "program buffer", how will openOCD set SW breakpoint if the CPU has instruction cache?

It will write the memory and not do anything else. The Debug Spec currently provides no other mechanism. If you need that, bring it up on https://lists.riscv.org/g/tech-debug/topics

> Is there defined common way for debugger to invalidate caches, without running instructions or have specific knowledge about cache control registers?

No.

> How does openOCD discovers what is implemented in processor DM module?

See examine() in src/target/riscv/riscv-013.c.

> Is the discovery done while CPU is in reset/halted/running?

OpenOCD always briefly halts every hart when first connecting.

> Is there any configuration file, which can be read by the openOCD at startup to provide processor implementation details, including privet CSRs internal memories, features implemented, etc ?

You can configure additional CSRs in the OpenOCD config file. `make pdf` will create a document that includes all riscv-specific commands.

agrobman, 2020-02-05:
Hi Tim,

Thanks a lot for clarifications.

make pdf gives me:
make: *** No rule to make target `pdf'.  Stop.

is there any example of the OpenOCD config file?

Alex

rtwfroody, 2020-02-07:
https://hackaday.io/page/4991-compiling-openocd-from-source-on-ubuntu-1604 has some instructions on how to build OpenOCD. You should be able to run `make pdf` whereever `make` would also succeed.

https://github.com/riscv/riscv-tests/tree/master/debug/targets contains some OpenOCD .cfg files. There are also some SiFive targets included in https://github.com/riscv/riscv-openocd/tree/riscv/tcl/board.

timsifive, 2021-01-19:
Inactive.

closed 2021-01-19 by timsifive

zhangdujiao, 2024-04-24:
> Defaults to program buffer, uses System Bus if that doesn't work or if prefer_sba is set, and if all that fails then uses abstract.
> 
> >

Hi Tim,
For the external debug, before debugging starts, GDB needs to load the program into memory or flash. At this time, is it also done through the three methods you mentioned: PB, SBA, AC. However, PB needs to work together with AC to access memory, and AC can only be used when the core enters the debug mode (during program loading, the core should not have entered the debug mode, right?). 
So, in this scenario, can the program only be burned through SBA?

TommyMurphyTM1234, 2024-04-24:
> Hi Tim

I'm not sure that you're going to get a reply from @rtwfroody at this stage. However others may be able to assist if he can't reply.

> AC can only be used when the core enters the debug mode (during program loading, the core should not have entered the debug mode, right?). 

Doesn't that really depend on your OpenOCD debug script? 
And wouldn't be common for a debug script to put the target into debug halt before loading a program?

TommyMurphyTM1234, 2024-04-24:
Oh, I just saw the duplicate post and @aap-sc's reply here now: 

* https://github.com/riscv-collab/riscv-openocd/issues/1057#issuecomment-2074047642

zhangdujiao, 2024-04-24:
> I'm not sure that you're going to get a reply from @rtwfroody at this stage. However others may be able to assist if he can't reply.
>
>
> Doesn't that really depend on your OpenOCD debug script? And [...]

Many thanks,
Does this mean that, in general, after configuring GDB to connect to OpenOCD and loading the OpenOCD configuration file, the core automatically enters debug mode? which is configured by the script?

aap-sc, 2024-04-24:
```
Does this mean that, in general, after configuring GDB to connect to OpenOCD and loading the OpenOCD configuration file, the core automatically enters debug mode? which is configured by the script?
```

The exact sequence of actions performed by OpenOCD is defined by configuration script. Namely, `gdb-attach` event handler. See the OpenOCD documentation for details. If user does not provide such handler, the default handler is used. The default handler halts the hart:

```
foreach t $targets {                                                                                                  
        set_default_target_event $t gdb-flash-erase-start "reset init"                                                      
        set_default_target_event $t gdb-flash-write-end "reset halt"                                                        
        set_default_target_event $t gdb-attach "halt 1000"                                                                  
} 
```

zhangdujiao, 2024-04-24:
Thanks aap, Thanks Tommy
My question is perfectly solved. 
really appreciated