riscv-collab/riscv-openocd #446: Is there way to use openOCD with verilog CPU simulations?

agrobman, 2020-02-06:
Does openOCD have a "verilog" target/driver for simulations?

TM1234, 2020-02-06:
You mean to connect openocd to a Verilog simulator and debug that?
I'm pretty sure that it doesn't.

mwachs5, 2020-02-06:
Yes you can, using either jtag_vpi or the remote-bitbang interface. There
are instructions and useful files in github.com/chipsalliance/rocket-chip.

This assumes the thing you’re simulating has a real JTAG interface.

TM1234, 2020-02-06:
Apologies - I wasn't aware of those interfaces being relevant here.

agrobman, 2020-02-06:
Megan, thanks a lot for the  pointers,
Could you, please, point to  verilog part of the bitbang interface and some documentations for it?
Thanks again
Alex

agrobman, 2020-03-17:
How do I compile/build the openOCD with the bitbang interface? what are the build options?

TM1234, 2020-03-17:
You mean remote bitbang?

`configure` .... --enable-remote-bitbang

https://github.com/riscv/riscv-openocd/blob/riscv/configure.ac#L356

mwachs5, 2020-03-17:
Hi Alex, check out this readme section of the rocket-chip repo:
https://github.com/chipsalliance/rocket-chip/blob/master/README.md#1-generating-the-remote-bit-bang-rbb-emulator

Sources:

https://github.com/chipsalliance/rocket-chip/blob/master/src/main/resources/vsrc/SimJTAG.v
and
https://github.com/chipsalliance/rocket-chip/blob/master/src/main/resources/csrc/SimJTAG.cc
,
https://github.com/chipsalliance/rocket-chip/blob/master/src/main/resources/csrc/remote_bitbang.cc

agrobman, 2020-03-17:
I need to co-simulate openOCD with CPU verilog model. Is this setup suitable for this?

mwachs5, 2020-03-17:
Yes, this will let you connect with GDB + OpenOCD to a running verilog
simulation, if you compile the C and V srcs into your simulation.

agrobman, 2020-03-17:
Thanks a lot guys, I'm more HW/verilog person - know nothing about linux internals, program building and inter-processes communications. Hopefully the provided info will be enough for me to build something working :)

agrobman, 2020-03-17:
got this:

...
checking for LIBUSB1... no
configure: WARNING: libusb-1.x not found, trying legacy libusb-0.1 as a fallback; consider installing libusb-1.x instead
checking for LIBUSB0... yes
checking for HIDAPI... no
checking for HIDAPI... no
checking for HIDAPI... no
checking for LIBFTDI... no
checking for LIBFTDI... no
checking for LIBJAYLINK... no
checking for environ in unistd.h and stdlib.h... yes
configure: creating ./config.status
config.status: error: cannot find input file: `Makefile.in'

Is this because of libusb1 ? Why do I need USB driver if I build bitbang?

TM1234, 2020-03-17:
As ever I'll suggest Liviu Ionescu's binary builds or his docker based build scripts as arguably the easiest and most predictable way to (cross) compile openocd for Windows (32 or 64 bit), Linux (32 or 64 bit) and MacOS.

* https://xpack.github.io/openocd/
* https://github.com/xpack-dev-tools/openocd-xpack/blob/xpack/README-BUILD.md
* https://github.com/xpack-dev-tools/openocd-xpack/blob/xpack/scripts/defs-source.sh#L43

agrobman, 2020-03-20:
Hi,

How should I run the openocd with bitbang driver/CPU RTL? what are the command option/initialization scripts, etc.  We just made connection, but looks like the openocd tries to detect  availiable TAP controllers reads JTAG ID from our RTL and terminates with:

warn: There are no enabled taps.  AUTO PROBING MIGHT NOT WORK!!
 ...

olofk, 2020-08-20:
@agrobman You can try jtag_vpi instead. There are instructions in the [SweRVolf repo](https://github.com/chipsalliance/Cores-SweRVolf) on how to run it against simulated models of SweRV EH1

agrobman, 2020-08-20:
@olofk , we are already done with this task, thanks for advice

closed 2020-08-20 by agrobman

hyperion009, 2023-07-25:
@agrobman I plan to do the same task as you. Could you provide some instructions , doc, or linker to some useful guides.

TommyMurphyTM1234, 2023-07-25:
> @agrobman I plan to do the same task as you. Could you provide some instructions , doc, or linker to some useful guides.

What exactly have you tried from the links posted earlier and what issues did you encounter?

JanMatCodasip, 2023-07-25:
@hyperion009, there are at least two protocols supported in OpenOCD that allow to transmit the JTAG transactions over a TCP connection, e.g. towards the simulation:

- `remote_bitbang`
- `jtag_vpi`

For `jtag_vpi`, you can find a reusable example of the server-side side code (which should be integrated into the simulation) here: https://github.com/fjullien/jtag_vpi