riscv-collab/riscv-openocd #452: Incorrect physical address translation with `set_enable_virt2phys`

KireinaHoro, 2020-02-24:
I'm running into problems with the features from #425 when trying to debug Linux (after `setup_vm()`), the following translation happens:

```
Debug: 3246 5880 riscv.c:1487 riscv_address_translate(): 0xffffffe0004a7d16 -> 0xff000000406a7d16
```

... which results in gdb still reading out all zeroes despite the translation succeeded.  Manually zeroing out the top 2 bytes of the translated address will make the read succeed with intended data.

Is there something wrong with the translation logic here?
https://github.com/riscv/riscv-openocd/blob/1449af5bdb99052c2f3bf92747f8438544cd8f3f/src/target/riscv/riscv.c#L1482-L1489

timsifive, 2020-02-24:
There could be a bug there. Based on your description, it might be the case that somewhere an address is sign-extended where it should be zero-extended.

I wasn't able to easily reproduce this. Did you manage to find a fix? I might have more time to look at this next week.

KireinaHoro, 2020-02-27:
A fix would be very naive to implement, but I don't feel that my method was elegant or even correct:

https://github.com/KireinaHoro/riscv-openocd/commit/03db7213c31ec89c41d240c5bb285ab4cba73cf1

I'd also like to mention that I'm using a modified version of the `use_bscan_tunnel` option on a Zynq UltraScale+ device running Rocket Chip.  Other debug functions (stepping, breakpoints, memory load, etc.) seem to be working just fine, as well as the virtual translation function with the above commit, but I'm not sure if there would be any regressions.  The related OpenOCD commits are as follows:

https://github.com/KireinaHoro/riscv-openocd/commit/cb7f6be16f56a62fc1bdafe0030862e42446e6b2

https://github.com/KireinaHoro/riscv-openocd/commit/11238e1eb4ecb883ee36fcf24668187c782ade5a

timsifive, 2020-02-29:
I've got a testcase that lets me reproduce this. I'll have a fix soon, or next week.

PR #453 (2020-03-05): Fix address translation when high bits are set.
Fixes #452.
Also check that the high bits match the MSB of the virtual address.

Change-Id: Ib1d3d04db9ad9327ef71ea3736d5cf5d3b65b9c4
commit 1ae21b38 (2020-03-05): Fix address translation when high bits are set. (#453)

Fixes #452.
Also check that the high bits match the MSB of the virtual address.

Change-Id: Ib1d3d04db9ad9327ef71ea3736d5cf5d3b65b9c4
closed 2020-03-05 by timsifive