riscv-collab/riscv-openocd #454: Building fails on CentOS Linux 7

abdullahyildiz, 2020-03-11:
I get the following error when I try to build riscv-openocd (I'm trying to build it on CentOS Linux release 7.7.1908 (Core)):

```
$ ./bootstrap 
+ aclocal
+ libtoolize --automake --copy
+ autoconf
+ autoheader
+ automake --gnu --add-missing --copy
configure.ac:28: installing './compile'
configure.ac:37: installing './config.guess'
configure.ac:37: installing './config.sub'
configure.ac:16: installing './install-sh'
configure.ac:16: installing './missing'
src/Makefile.am:4: error: bad characters in variable name '%C%_openocd_SOURCES'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:7: error: bad characters in variable name '%C%_libopenocd_la_SOURCES'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:11: error: bad characters in variable name '%C%_openocd_LDADD'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:13: error: bad characters in variable name '%C%_openocd_LDADD'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:16: error: bad characters in variable name '%C%_openocd_LDADD'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:18: error: bad characters in variable name '%C%_openocd_LDADD'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:21: error: bad characters in variable name '%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:26: error: bad characters in variable name '%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:27: error: bad characters in variable name '%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:29: error: bad characters in variable name '%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:30: error: bad characters in variable name '%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:31: error: bad characters in variable name '%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:35: error: bad characters in variable name '%C%_libopenocd_la_CPPFLAGS'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:38: error: bad characters in variable name '%C%_libopenocd_la_LDFLAGS'
Makefile.am:143:   'src/Makefile.am' included from here
src/Makefile.am:46: error: bad characters in variable name '%C%_libopenocd_la_LIBADD'
Makefile.am:143:   'src/Makefile.am' included from here
doc/Makefile.am:2: error: bad characters in variable name '%C%_openocd_TEXINFOS'
Makefile.am:144:   'doc/Makefile.am' included from here
Makefile.am:46: warning: wildcard $(srcdir: non-POSIX variable name
Makefile.am:46: (probably a GNU make extension)
Makefile.am: installing './INSTALL'
Makefile.am: installing './depcomp'
automake: error: cannot open < ./%D%/openocd.texi: No such file or directory
```
To fix this error, I've compiled and installed automake-1.14 from source and then tried to build riscv-openocd again:

This time, I get this error:

```
$ ./bootstrap                                                                                           
+ aclocal
+ libtoolize --automake --copy
+ autoconf
configure.ac:12: error: possibly undefined macro: AC_MSG_WARN
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:36: error: possibly undefined macro: AC_DISABLE_SHARED
configure.ac:37: error: possibly undefined macro: AC_PROG_LIBTOOL
configure.ac:200: error: possibly undefined macro: AC_DEFINE
configure.ac:598: error: possibly undefined macro: AC_MSG_NOTICE
```

Is there anyone who encountered the same issue?

TM1234, 2020-03-11:
Why not compile it using Liviu Ionescu's (@ilg-ul) xPack Project docker based scripts? 
Much more reliable in my experience and opinion.

https://xpack.github.io/openocd/
https://github.com/xpack-dev-tools/openocd-xpack/blob/xpack/README-BUILD.md

Hope this helps.
Cheers
Tommy

abdullahyildiz, 2020-03-12:
@TM1234 Does it support RISC-V?

TM1234, 2020-03-12:
Yes. And you can change this to build from the sifive RISC-V openocd repo (or the upstream openocd repo etc) if you want (by default it builds from @ilg-ul 's own source snapshots for predictable results).

https://github.com/xpack-dev-tools/openocd-xpack/blob/xpack/scripts/defs-source.sh#L43

abdullahyildiz, 2020-03-15:
As far as I understand, I should build the OpenOCD binary (targeting CentOS) on an Ubuntu machine:

https://github.com/xpack-dev-tools/openocd-xpack/blob/xpack/README-BUILD.md#build-the-gnulinux-and-windows-binaries

Am I right?

PS I followed the instructions and tried to install OpenOCD via docker based build scripts on CentOS machine, however, I couldn't find a folder named **deploy** after the following script finishes running:

`sudo bash ~/Downloads/openocd-xpack.git/scripts/build.sh --linux64`

ilg-ul, 2020-03-15:
did you check in `~/Work/openocd`?

abdullahyildiz, 2020-03-15:
@ilg-ul, The following shows the contents of `~/Work`:

```
├── cache
└── openocd-0.10.0-14
    ├── 20200315-1813
    ├── build.git
    │   ├── CHANGELOG.md
    │   ├── LICENSE
    │   ├── package.json
    │   ├── patches
    │   │   ├── libftdi1-1.2-cmake-FindUSB1.patch
    │   │   ├── libftdi1-1.4-cmake-FindUSB1.patch
    │   │   ├── libusb-win32-1.2.6.0-mingw-w64.patch
    │   │   └── README.md
    │   ├── pkgconfig
    │   │   ├── hidapi-0.7.0-linux.pc
    │   │   ├── hidapi-0.7.0-mac.pc
    │   │   ├── hidapi-0.7.0-windows.pc
    │   │   ├── hidapi-0.8.0-rc1-windows.pc
    │   │   ├── libusb-win32-1.2.6.0.pc
    │   │   └── README.md
    │   ├── README-BUILD.md
    │   ├── README-DEVELOP.md
    │   ├── README.md
    │   ├── README-PUBLISH.md
    │   └── scripts
    │       ├── build-native.sh
    │       ├── build.sh
    │       ├── common-apps-functions-source.sh
    │       ├── common-functions-source.sh
    │       ├── common-libs-functions-source.sh
    │       ├── common-versions-source.sh
    │       ├── container-build.sh
    │       ├── defs-source.sh
    │       ├── git-clone.sh
    │       ├── helper
    │       ├── host-defs-source.sh
    │       ├── README-0.10.0-13.md
    │       ├── README-0.10.0-14.md
    │       └── VERSION
    └── sources

8 directories, 31 files
```

ilg-ul, 2020-03-15:
Then the build failed on the way.

abdullahyildiz, 2020-03-16:
@ilg-ul, this time I've applied [the instructions you suggested](https://github.com/riscv/riscv-openocd/issues/454#issuecomment-597714712) on an Ubuntu machine (Ubuntu 18.04.3 LTS) and now I can run RISC-V OpenOCD on CentOS:

```
$ make
echo run.spike64.CheckMisa
run.spike64.CheckMisa
./gdbserver.py \
	./targets/RISC-V/spike64.py \
	CheckMisa \
	--isolate \
	--print-failures \
	--sim_cmd /home/edaman/riscv/bin/spike \
	--server_cmd /home/edaman/riscv/xpack-openocd-0.10.0-14/bin/openocd
Using $misa from hart definition: 0x8000000000141125
[CheckMisa] Starting > logs/20200316-093933-spike64-CheckMisa.log
[CheckMisa] pass in 2.80s
::::::::::::::::::::::::::::[ ran 1 tests in 3s ]:::::::::::::::::::::::::::::
1 tests returned pass
echo run.spike64.CustomRegisterTest
run.spike64.CustomRegisterTest
./gdbserver.py \
	./targets/RISC-V/spike64.py \
	CustomRegisterTest \
	--isolate \
	--print-failures \
	--sim_cmd /home/edaman/riscv/bin/spike \
	--server_cmd /home/edaman/riscv/xpack-openocd-0.10.0-14/bin/openocd
Using $misa from hart definition: 0x8000000000141125
[CustomRegisterTest] Starting > logs/20200316-093936-spike64-CustomRegisterTest.log
[CustomRegisterTest] pass in 12.34s
```

Thanks for your help. :+1:

closed 2020-03-16 by abdullahyildiz

ilg-ul, 2020-03-16:
What is the reason for not using the provided binaries and deciding to compile it yourself?

abdullahyildiz, 2020-03-16:
Did you mean that:

https://github.com/xpack-dev-tools/openocd-xpack/releases/

ilg-ul, 2020-03-16:
yes

abdullahyildiz, 2020-03-16:
Sorry, I've just noticed that.

KingFrige, 2020-07-30:
maybe you can look here:  [autoreconf -i fails #141](https://github.com/riscv/riscv-openocd/issues/141)

abdullahyildiz, 2020-08-04:
@KingFrige,

aclocal path is already set to `/usr/local/share/aclocal` in my machine:

```
# aclocal --print-ac-dir

/usr/local/share/aclocal
```

By the way, there is no .m4 file in that directory:

```
# ls /usr/local/share/aclocal
README
```

TommyMurphyTM1234, 2020-08-04:
> @KingFrige,
> 
> aclocal path is already set to `/usr/local/share/aclocal` in my machine:
> 
> ```
> # aclocal --print-ac-dir
> 
> /usr/local/share/aclocal
> ```
> 
> By the way, there is no .m4 file in that directory:
> 
> ```
> # ls /usr/local/share/aclocal
> README
> ```

I thought you said here that it was already compiling OK for you? 

https://github.com/riscv/riscv-openocd/issues/454#issuecomment-599370744

abdullahyildiz, 2020-08-04:
@TommyMurphyTM1234,

> I thought you said here that it was already compiling OK for you?
> 
> [#454 (comment)](https://github.com/riscv/riscv-openocd/issues/454#issuecomment-599370744)

Yes, I can run xPack OpenOCD on CentOS.

I tried to use @KingFrige's answer to build [riscv-openocd](https://github.com/riscv/riscv-openocd) and wanted to give an update on this.

KingFrige, 2020-08-08:
yes, there is nothing in `/usr/local/share/aclocal`, `*.m4` maybe in `/usr/share/aclocal`

so, you need execute the command :
```
sudo cp -rf /usr/share/aclocal/* /usr/local/share/aclocal
```

abdullahyildiz, 2020-08-10:
@KingFrige,

Thanks. The problem is now fixed. Sorry for the misunderstanding.