riscv-collab/riscv-openocd #457: Broken pipe error while connecting to simulation

abdullahyildiz, 2020-03-19:
I get the following error when I try to connect RTL simulation and OpenOCD:

```
Info : only one transport option; autoselect 'jtag'
Info : Initializing remote_bitbang driver
Info : Connecting to localhost:38000
Info : remote_bitbang driver initialized
Info : This adapter doesn't support configurable speed
Error: fflush: Broken pipe
Error: read: count=-1, error=Broken pipe
Error: Trying to use configured scan chain anyway...
Error: fflush: Broken pipe
Error: fflush: Broken pipe
Warn : Bypassing JTAG setup events due to errors
Error: fflush: Broken pipe
Error: fflush: Broken pipe
Error: failed jtag scan: -4
Error: Unsupported DTM version: 12
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet

Error: Unsupported DTM version: 12
Error: fflush: Broken pipe
Error: failed: -4
```

What can be the reason to get such an error?

TM1234, 2020-03-19:
Your simulation is not listening on localhost:38000?

abdullahyildiz, 2020-03-19:
@TM1234, this is how I run the RTL simulation first and then OpenOCD:

At the first terminal window I run

`./simulator-chipyard-jtagRocketConfig +jtag_rbb_enable=1 --rbb-port=38000 /home/edaman/riscv/setup/riscv-tests/isa/rv64ui-p-simple`

and then at the second terminal window I run

`openocd -f riscv_openocd.cfg`

The following is the content of OpenOCD config file:

```
interface remote_bitbang
remote_bitbang_host localhost
remote_bitbang_port 38000

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME

gdb_report_data_abort enable

init
halt
```

timsifive, 2020-03-19:
The unsupported DTM message implies that OpenOCD did connect and manage to
scan some bits, and got back a result that doesn't make sense. You should
start debugging this. You can see in detail what OpenOCD did by adding `-d`
to the OpenOCD command line.
Broken pipe is probably due to the simulator closing the connection, which
may be related to the first problem.

abdullahyildiz, 2020-03-20:
@timsifive,

Thanks for your comment. I'll try to find why RTL simulation couldn't send a proper response to OpenOCD.

When I take a look at in detail what OpenOCD did, the first error I get is the following:

```
Debug: 189 39 core.c:1128 jtag_examine_chain(): DR scan interrogation for IDCODE/BYPASS
Debug: 190 39 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 191 741 server.c:609 sig_handler(): Terminating on Signal 13
Error: 192 741 remote_bitbang.c:106 remote_bitbang_quit(): fflush: Broken pipe
```

I've attached the detailed log file.

[openocd.log](https://github.com/riscv/riscv-openocd/files/4358966/openocd.log)

timsifive, 2020-03-20:
This bit in the log is most concerning to me:
Debug: 189 39 core.c:1128 jtag_examine_chain(): DR scan interrogation for
IDCODE/BYPASS
Debug: 190 39 core.c:327 jtag_call_event_callbacks(): jtag event: TAP reset
Debug: 191 741 server.c:609 sig_handler(): Terminating on Signal 13

Signal 13 is SIGPIPE. I don't understand why some RISC-V code got to run
afterwards, but I assume that something went wrong between OpenOCD and the
simulator, and everything after that point is irrelevant.

abdullahyildiz, 2020-03-23:
Can it be because of the locale settings which the machine uses?

The following shows the locales defined on my machine:

```
$ locale
LANG=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=
```

By the way, which locales are defined on your machine?

timsifive, 2020-03-23:
I would be really surprised if locale settings have anything to do with it.

tnewsome@compy-linux:~$ locale
LANG=en_US.UTF-8
LANGUAGE=en_US
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=

timsifive, 2020-03-23:
I should add that I've never run OpenOCD this way, so the settings on my
machine are probably irrelevant.

TM1234, 2020-03-23:
SIGPIPE signal 13 means that there's no reader for the pipe so I would guess that the simulation is prematurely exiting or otherwise failing to act as a reader of the pipe so that when openocd starts trying to communicate with it via the remote bitbang port it's no longer there or responding? You probably need to diagnose things at the simulator end.

abdullahyildiz, 2020-03-24:
@timsifive,

> I should add that I've never run OpenOCD this way, so the settings on my machine are probably irrelevant.

No problem. I just wondered if the locale settings are somewhat related to my issue.

***

@TM1234,

I've applied the instructions at [Chipyard’s documentation (8.2.2.1. Creating a DTM+JTAG Config)](https://chipyard.readthedocs.io/en/latest/Advanced-Concepts/Chip-Communication.html#creating-a-dtm-jtag-config).

This is how I run the simulator:

`./simulator-chipyard-jtagRocketConfig +verbose +jtag_rbb_enable=1 --rbb-port=38000 /home/edaman/riscv/setup/riscv-tests/isa/rv64ui-p-simple`

And the following is the corresponding simulation log file:

[verilator_sim.log](https://github.com/riscv/riscv-openocd/files/4374014/verilator_sim.log)

abdullahyildiz, 2020-03-26:
@timsifive, @TM1234,

When I used another [RocketChip repository](https://github.com/chipsalliance/rocket-chip) to build a Remote Bit-Bang (RBB) Emulator, OpenOCD connected to simulation.

However, CPU didn't halted and continued to run. It may be due to a wrong configuration in RocketChip's JTAG interface or debug transport module (DTM).

```
xPack OpenOCD, 64-bit Open On-Chip Debugger 0.10.0+dev (2019-07-17-11:25)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'jtag'
Info : Initializing remote_bitbang driver
Info : Connecting to localhost:38000
Info : remote_bitbang driver initialized
Info : This adapter doesn't support configurable speed
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (<invalid>), part: 0x0000, ver: 0x0)
Info : datacount=2 progbufsize=16
Info : Disabling abstract command reads from CSRs.
Info : Examined RISC-V core; found 1 harts
Info :  hart 0: XLEN=64, misa=0x800000000094112d
Info : Listening on port 3333 for gdb connections
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
```
I've attached the detailed log file.

[openocd.log](https://github.com/riscv/riscv-openocd/files/4387322/openocd.log)

TM1234, 2020-03-26:
Don't you need to call openocd's reset init or reset halt or halt to stop it on connect?
These may be specified in the openocd script or via gdb's monitor command - e.g.

`monitor reset init`

abdullahyildiz, 2020-03-26:
> Don't you need to call openocd's reset init or reset halt or halt to stop it on connect?
> These may be specified in the openocd script or via gdb's monitor command - e.g.
> 
> `monitor reset init`

This is the content of OpenOCD config file that I use:

```
interface remote_bitbang
remote_bitbang_host localhost
remote_bitbang_port 38000

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5

set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME riscv -chain-position $_TARGETNAME

gdb_report_data_abort enable

init
halt
```

TM1234, 2020-03-26:
OK - the log seems to be saying that the halts are being halted so I don't know why there's a problem.

```
Debug: 440 26787 target.c:3048 handle_halt_command(): -
Debug: 441 26787 riscv.c:1321 riscv_openocd_halt(): [0] halting all harts
Debug: 442 26787 riscv.c:2039 riscv_halt_one_hart(): halting hart 0
```

I don't know if the many fflush failure and keep_alive messages are also relevant here?

```
Debug: 1177 86458 riscv.c:1193 riscv_openocd_poll(): polling all harts
Debug: 1178 86458 riscv.c:2152 riscv_set_current_hartid(): setting hartid to 0, was 0
Debug: 1179 86459 riscv.c:1148 riscv_poll_hart(): polling hart 0, target->state=2
Error: 1180 86459 remote_bitbang.c:106 remote_bitbang_quit(): fflush: Broken pipe
Error: 1181 86459 remote_bitbang.c:155 remote_bitbang_rread(): read: count=0, error=Broken pipe
Error: 1182 86459 riscv-013.c:490 dmi_scan(): dmi_scan failed jtag scan
Error: 1183 86459 riscv-013.c:548 dmi_op_timeout(): failed read at 0x11, status=2
Debug: 1184 86459 riscv.c:1158 riscv_poll_hart():   triggered running
...
Debug: 478 30080 log.c:409 keep_alive(): keep_alive() was not invoked in the 1000ms timelimit (4058). This may cause trouble with GDB connections.

```

abdullahyildiz, 2020-03-27:
@TM1234,

> I don't know if the many fflush failure and keep_alive messages are also relevant here?
>
> ```
> Debug: 1177 86458 riscv.c:1193 riscv_openocd_poll(): polling all harts
> Debug: 1178 86458 riscv.c:2152 [...]

I also have no idea yet.

abdullahyildiz, 2020-03-28:
I think the problem I have occurs due to the OS configuration I use.

I built riscv-openocd on a fresh CentOS 8 image and copied it to my CentOS machine.

Now I can use OpenOCD to access RTL simulation from GDB.

@timsifive, @TM1234,

Thanks for your help.

closed 2020-03-28 by abdullahyildiz