riscv-collab/riscv-openocd #460: Could/should FP reg accesses be predicated on misa?

TM1234, 2020-03-23:
At the moment openocd (maybe driven by gdb?) will speculatively try to read FP registers and issue a warning if the target hart does not support FP. Could/should such accesses be predicated on misa indicating that FP is actually supported and skipped otherwise? Maybe it's not that simple if some of this functionality is driven from gdb and, perhaps, a "superset" approach to the registers potentially supported by a target?

aswaterman, 2020-03-23:
Don’t forget that implementations are allowed to wire the entire misa
register to 0. So the optimization you propose is valid, but only if misa
is nonzero. (In other words, misa.F is trustworthy if and only if misa is
nonzero overall.)

timsifive, 2020-03-23:
The way I read the code, OpenOCD will not read FPRs if F or D is not present in misa.

I had no idea that misa is allowed to be 0. That's really annoying. How much hardware is really saved by not implementing misa, compared to the cost of the FPU?

aswaterman, 2020-03-23:
I think it's pretty annoying, too, and I think it's legitimate for OpenOCD to refuse to work on such cores.  (FWIW, I don't know of any off the top of my head.)

timsifive, 2020-03-23:
As it is, OpenOCD will work on those cores, but assume it doesn't support anything beyond the base ISA. That's probably reasonable behavior given the kind of core where somebody would opt to not support misa=0. If somebody has a real core where this is a problem, I'll consider adding a mechanism for the user to provide the misa value.

TM1234, 2020-03-23:
Thanks for the info Tim/Andrew.
When I connect to Sifive U540 based targets (1 x e51 = rv64imac + 4 x u54 = rv64gc) using gdb/openocd it seems to try to read FPU regs for *all* cores/harts including the e51 and then issues a warning when it fails for the e51. Any idea why that is so? Thanks.

timsifive, 2020-03-23:
Ah, yes. When you do that OpenOCD exposes the different harts to gdb as threads. Them being threads, gdb assumes they are all on the same hardware, and specifically it assumes they all have the same sets of registers. In this case OpenOCD tells gdb the superset of all the registers available on any hart, so gdb thinks FPRs exist on all "threads." OpenOCD looks at misa when building the register list, but not when actually being asked to read a register. I'm not sure if there'd even be a benefit to that, since either case gdb will ask to read the FPRs, and OpenOCD's only option is to return an error message when that's done for hart 0.

TM1234, 2020-03-23:
Thanks Tim for the explanation. 
That explains things for me.
Since gdb will ask for these regs regardless I'll close the issue.
Thanks again.

closed 2020-03-23 by TM1234