riscv-collab/riscv-openocd #462: OpenOCD does not recognize 2nd active hart of the CPU

agrobman, 2020-03-24:
Hi, 

We are running openocd with RTL simulation environment to check if the CPU somewhat compliant with RISCV debug spec and can work with openocd application. The CPU has to active harts, but OpenOCD reports:

Info : 316 128288 riscv-013.c:1771 examine():  hart 0: XLEN=32, misa=0x40001105
Info : 317 128289 riscv-013.c:1773 examine():  hart 1: currently disabled

while we have both harts enabled.

I was poking the code and seems these lines are related to the active harts detection:
in riscv-013.c:

```
	/* Don't call any riscv_* functions until after we've counted the number of
	 * cores and initialized registers. */
	for (int i = 0; i < dm->hart_count; ++i) {
		if (!riscv_rtos_enabled(target) && i != target->coreid)
			continue;

```

What are "rtos_enabled" and "coreid",  how are these things  set?
Are these configuration settings from *.cfg files?

timsifive, 2020-03-24:
This output is expected if you're using `-rtos hwthread`, or if you have
only one hart configured. The problem is likely in your OpenOCD cfg file.
Here's an example of one for two harts:
https://github.com/riscv/riscv-tests/blob/master/debug/targets/RISC-V/spike-2-hwthread.cfg

agrobman, 2020-03-25:
Hi Tim,

thanks for the tip,
I got following  log:
Info : only one transport option; autoselect 'jtag'
Info : Hardware thread awareness created
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Initializing remote_bitbang driver
Info : Connecting to uls-ep-hpcbe041.wdc.com:46069
Info : remote_bitbang driver initialized
Info : This adapter doesn't support configurable speed
help
Info : JTAG tap: ehx2.cpu tap/device found: 0x1000008b (mfg: 0x045 (Western Digital Technologies Inc), part: 0x0000, ver: 0x1)
help
Info : datacount=2 progbufsize=0
Warn : We won't be able to execute fence instructions on this target. Memory may not always appear consistent. (progbufsize=0, impebreak=0)
halt
break

Info : Core 0 could not be made part of halt group 1.
Info : Examined RISC-V core; found 2 harts
Info :  hart 0: XLEN=32, misa=0x40001105
Info :  hart 1: currently disabled
Info : datacount=2 progbufsize=0
Warn : We won't be able to execute fence instructions on this target. Memory may not always appear consistent. (progbufsize=0, impebreak=0)

help
Info : Core 1 could not be made part of halt group 1.
Info : Examined RISC-V core; found 2 harts
Info :  hart 0: currently disabled
Info :  hart 1: XLEN=32, misa=0x40001105
Info : Listening on port 3333 for gdb connections

for my script:
interface remote_bitbang
remote_bitbang_port 46069
remote_bitbang_host uls-ep-hpcbe041.wdc.com

set CHIP ehx2
set ENDIAN little

jtag newtap $CHIP cpu -irlen 5 -expected-id 0x1000008b
set TARGET $CHIP.cpu0
set TARGET1 $CHIP.cpu1
target create $TARGET riscv -endian $ENDIAN -chain-position $CHIP.cpu -rtos hwthread
target create $TARGET1 riscv -endian $ENDIAN -chain-position $CHIP.cpu  -coreid 1
target smp $TARGET $TARGET1

Looks like the tool thinks that we have 2 CPUs with 2 threads and only one hart is enabled on each.

Is there any manual for CFG scripts writing/syntax ?

timsifive, 2020-03-25:
> Looks like the tool thinks that we have 2 CPUs with 2 threads and only
one hart is enabled on each.

It looks that way, but this is actually exactly the output you expect to
see given your configuration. It's confusing, I agree. Once I get rid of
`-rtos riscv` (at the end of this year) this will be cleaned up. Is there
any problem beyond simply the content of the Info messages?

agrobman, 2020-03-25:
We don’t know yet, we’re just learning how to use openocd for our simulation/verification needs.
It was very confusing for us that the tool could not find two enabled harts – we didn’t know if this is was RTL or openocd fault .
We don’t know yet if our implementation/debug spec understanding is correct – we had to change RTL recently because openocd couldn’t understand some of our CPU features.

I would think the openocd shouldn’t use application level abstractions, such as  rtos/linux etc and should be more close to HW terminology at least on low level such as examine task …

So far I see that the tool doesn’t support :

1)      Internal memories access via abstract commands. We have instruction and data memories and memory mapped I/O.

These mems have also certain restrictions – can be access as words only;

The internal memories have base address and size configurable by RTL parameters, which are selected by customers. We need a way to provide these parameters to the tool, to select correct method to access memory (  system bus or abstract )

These restrictions have effect, for ex, how to setup SW breakpoints ….

2)      Fence.i emulation – we have a privet CSR, write to it will cause Instruction Cache invalidation .

3)      We have bunch of privet CSRs which need to be accessed by debugger – is there way to define these without changing the application code, by config script for ex?

How do we control our harts ? What are the commands?
How do we stop certain hart? Both harts?
How do we resume run for a hart? Both harts?
How do we single step harts, including both harts simultaneously?

Somehow, the openocd doesn’t stop CPU run after connecting which seems to me quite a strange …

BTW, is there any program reference to user command syntax for config scripts and user commands?

Regards
Alex

TM1234, 2020-03-26:
> Is there any manual for CFG scripts writing/syntax ?

The openocd documentation and example scripts cover these in detail.

timsifive, 2020-03-26:
One thing to understand is that OpenOCD is made to work with gdb, and gdb
is made to debug threads on a native UNIX system. This means neither are
particularly suited for working on embedded multi-core hardware, but it all
hangs together enough to be useful.

On Wed, Mar 25, 2020 at 1:37 PM agrobman <notifications@github.com> wrote:

> So far I see that the tool doesn’t support :
>
> 1) Internal memories access via abstract commands. We have instruction and
> data memories and memory mapped I/O.

What are you trying to do? As it is OpenOCD will use abstract memory access
if neither program buffer nor system bus access are available. I don't
think there's any provision in the abstract command in the debug spec to
treat instruction and data memory separately.

The internal memories have base address and size configurable by RTL
> parameters, which are selected by customers. We need a way to provide these
> parameters to the tool, to select correct method to access memory ( system
> bus or abstract )

That doesn't exist, but I'd be happy to review code for that improvement.

2) Fence.i emulation – we have a privet CSR, write to it will cause
> Instruction Cache invalidation .

That's a real weakness in the RISC-V specifications.

3) We have bunch of privet CSRs which need to be accessed by debugger – is
> there way to define these without changing the application code, by config
> script for ex?

Yes. Per the documentation (which gets built along with the source, into
doc/openocd.pdf):
riscv expose_custom n0[-m0][,n1[-m1]]...
[Command]
The RISC-V Debug Specification allows targets to expose custom registers
through
abstract commands. (See Section 3.5.1.1 in that document.) This command
config-
ures a list of inclusive ranges of those registers to expose. Number 0
indicates the first
custom register, whose abstract command number is 0xc000. This command must
be
executed before ‘init‘.

How do we control our harts ? What are the commands?
> How do we stop certain hart? Both harts?
> How do we resume run for a hart? Both harts?

This is all addressed in the normal OpenOCD documentation.

> How do we single step harts, including both harts simultaneously?

If harts are combined into a single "rtos" as they were in the config you
mentioned in this discussion, OpenOCD will attempt to step both
simultaneously. This requires support for hart arrays.

> BTW, is there any program reference to user command syntax for config
> scripts and user commands?

The documentation is built along with the source, into doc/openocd.pdf.

>

agrobman, 2020-03-26:
Hi Tom,

Thanks a lot for clarifications, but some things are still not clear for me:

> So far I see that the tool doesn’t support :
>
> 1) Internal memories access via abstract commands. We have instruction and
> data memories and memory mapped I/O.

What are you trying to do? As it is OpenOCD will use abstract memory access
if neither program buffer nor system bus access are available. I don't
think there's any provision in the abstract command in the debug spec to
treat instruction and data memory separately.

The story is that the CPU had implemented both abstract and system bus mechanisms to access  memory. We need a way to select one or another, depending on the memory address and configuration parameters.
There was attempt to solve it with “fixed” addresses, for one particular project, but going ahead with open sourcing, we need  agreed way to let the tool know what way to access the memory.

Since you know internals of the tool way more than anybody else we need your advice how to implement this selection/ provide a configuration/ command to define what address ranges are external, accessed with SB and what are internals, accessible with abstracts.

Another thing is access size for these internals – for ex, internal I/O can be accessed only with aligned words, while memory could be accessed with smaller sizes … We need a common way to define these limitations ..

The internal memories have base address and size configurable by RTL
> parameters, which are selected by customers. We need a way to provide these
> parameters to the tool, to select correct method to access memory ( system
> bus or abstract )

That doesn't exist, but I'd be happy to review code for that improvement.

2) Fence.i emulation – we have a privet CSR, write to it will cause
> Instruction Cache invalidation .

That's a real weakness in the RISC-V specifications.

I agree – the spec should define standard way to invalidate/flush caches or better one cache line, where debugger need to place ebreak …
Or read data from ( for DCache.)

3) We have bunch of privet CSRs which need to be accessed by debugger – is
> there way to define these without changing the application code, by config
> script for ex?

Yes. Per the documentation (which gets built along with the source, into
doc/openocd.pdf):
riscv expose_custom n0[-m0][,n1[-m1]]...
[Command]
The RISC-V Debug Specification allows targets to expose custom registers
through
abstract commands. (See Section 3.5.1.1 in that document.) This command
config-
ures a list of inclusive ranges of those registers to expose. Number 0
indicates the first
custom register, whose abstract command number is 0xc000. This command must
be
executed before ‘init‘.

Is there way to use/define names for these privet CSRs?

> How do we single step harts, including both harts simultaneously?

If harts are combined into a single "rtos" as they were in the config you
mentioned in this discussion, OpenOCD will attempt to step both
simultaneously. This requires support for hart arrays.

What did you mean by “hart arrays”?  - Is this implemented or need to be implemented?
For now the tool finds two CPUs with 2 harts each, and one hart enabled in each …
Should I add –rtos argument to both our target commands?

timsifive, 2020-03-26:
> The story is that the CPU had implemented both abstract and system bus
> mechanisms to access memory. We need a way to select one or another,
> depending on the memory address and configuration [...]

The existing `riscv set_prefer_sba` command does something closely related,
but it should really be replaced with something more general. You can
certainly find all the relevant code by searching for that in the source.
Ideally it would be replaced with something like `riscv set_memory_access
order` where order is a sequence of sba, progbuf, and abstract in whatever
the preferred order is. If you need to extend that with memory addresses
afterwards that is fine.

Another thing is access size for these internals – for ex, internal I/O can
> be accessed only with aligned words, while memory could be accessed with
> smaller sizes … We need a common way to define these limitations ..

That could possibly be rolled into the same command. As long as it's
possible to not specify access size/address for the common case where all
memory should be treated the same, that is reasonable.

> Is there way to use/define names for these privet CSRs?

Not directly, although I'm sure OpenOCD's TCL lets you create commands that
are close enough.

> What did you mean by “hart arrays”?

Hart arrays are a feature of the hardware defined in the Debug Spec. See
the section about Selecting Multiple Harts. OpenOCD supports this.

> Should I add –rtos argument to both our target commands?

No. You only have one "rtos" that contains 2 "threads."

>

agrobman, 2020-03-30:
Hi Tim,

I tried to run 'compliance' test and got following response:

Info : accepting 'telnet' connection on tcp/4444
Info : Basic compliance test against RISC-V Debug Spec v0.13
Info : This test is not complete, and not well supported.
Info : Your core might pass this test without being compliant.
Info : Your core might fail this test while being compliant.
Info : Use your judgment, and please contribute improvements.
Error: Please run with -rtos riscv to run compliance test.

Error: Command does not take any parameters.
    ehx2.cpu0 riscv test_compliance riscv test_compliance
    ehx2.cpu1 riscv test_compliance riscv test_compliance
  riscv test_compliance riscv test_compliance

what exactly should I run with -rtos option? considering the *cfg script I'm running the openocd with:

```
interface remote_bitbang
remote_bitbang_port 46069
remote_bitbang_host uls-ep-hpcbe041.wdc.com

set CHIP ehx2
set ENDIAN little

jtag newtap $CHIP cpu -irlen 5 -expected-id 0x1000008b
set TARGET $CHIP.cpu0
set TARGET1 $CHIP.cpu1
target create $TARGET riscv -endian $ENDIAN -chain-position $CHIP.cpu -rtos hwthread
target create $TARGET1 riscv -endian $ENDIAN -chain-position $CHIP.cpu  -coreid 1
target smp $TARGET $TARGET1

```

timsifive, 2020-03-30:
`-rtos riscv` is deprecated, so I guess that means the compliance test is
also deprecated. It was already incomplete and unsupported.

closed 2021-07-19 by timsifive

jimaandro, 2025-02-06:
Hello.
I have a similar issue. My target is an Openpiton System with 4 CVA6 cores.
I used the configuration file below, but my problem is that I get
`Core x could not be made part of halt group 1.` as you can see in screenshot below

```
adapter speed 1000

adapter driver ftdi
ftdi vid_pid 0x0403 0x6010
ftdi channel 0
transport select jtag
ftdi tdo_sample_edge falling
ftdi layout_init 0x0008 0x000b
reset_config none

set _CHIPNAME riscv
jtag newtap $_CHIPNAME cpu -irlen 5

set _TARGETNAME $_CHIPNAME.cpu

set _TARGETNAME_0 $_CHIPNAME.cpu0
set _TARGETNAME_1 $_CHIPNAME.cpu1
set _TARGETNAME_2 $_CHIPNAME.cpu2
set _TARGETNAME_3 $_CHIPNAME.cpu3

# Create targets for each hart
target create $_TARGETNAME_0 riscv -chain-position $_TARGETNAME -coreid 0 -rtos hwthread
target create $_TARGETNAME_1 riscv -chain-position $_TARGETNAME -coreid 1
target create $_TARGETNAME_2 riscv -chain-position $_TARGETNAME -coreid 2
target create $_TARGETNAME_3 riscv -chain-position $_TARGETNAME -coreid 3

# Configure work area for debugging
$_TARGETNAME_0 configure -work-area-phys 0x80000000 -work-area-size 1000 -work-area-backup 1
$_TARGETNAME_1 configure -work-area-phys 0x80001000 -work-area-size 1000 -work-area-backup 1
$_TARGETNAME_2 configure -work-area-phys 0x80002000 -work-area-size 1000 -work-area-backup 1
$_TARGETNAME_3 configure -work-area-phys 0x80003000 -work-area-size 1000 -work-area-backup 1

target smp $_TARGETNAME_0 $_TARGETNAME_1 $_TARGETNAME_2 $_TARGETNAME_3

gdb_report_data_abort enable
gdb_report_register_access_error enable

riscv set_reset_timeout_sec 120
riscv set_command_timeout_sec 120

# Try enabling address translation (only works for newer versions)
if { [catch {riscv set_enable_virtual on} ] } {
    echo "Warning: This version of OpenOCD does not support address translation. To debug on virtual addresses, please update to the latest version." }

init
halt
echo "Ready for Remote Connections"

```

[image]

My problem is that using this configuration, only the last hart/target that is created is halted (in this case is hart 3).
As you can see, only hart 3 is set to ZSBL address (0xfff1010000), and only the registers of this hart are 0.

[image]

How can I fix this?

TommyMurphyTM1234, 2025-02-06:
A lot has changed with OpenOCD in the five years since this issue was originally created so your issue may have nothing to do with the original one. 

You should probably open a new issue and include all relevant details such as: 

- what version of OpenOCD are you using and how did you configure and build it? 
- a verbose `openocd -d3` log
- your OpenOCD configuration file
- any other relevant details

aap-sc, 2025-02-06:
@jimaandro in addition to comments by @TommyMurphyTM1234 , please file  a new issue.