riscv-collab/riscv-openocd #468: hifive1 rev B

githubquestions, 2020-04-26:
Hello, try to use [hifive1 rev B cfg](https://github.com/riscv/riscv-openocd/blob/riscv/tcl/board/sifive-hifive1-revb.cfg) with Sifive prebuilt binary, but get `Error: invalid subcommand "set_enable_virt2phys off" in procedure 'script'`. Then try to use same config with newer version of this repo built from instruction and get `Error: No J-Link device found`. Any ideas?

TM1234, 2020-04-26:
> Hello, try to use hifive1 rev B cfg with Sifive prebuilt binary, but get Error: invalid subcommand "set_enable_virt2phys off" in procedure 'script'. 

Most likely that build of openocd predates this commit?

https://github.com/riscv/riscv-openocd/commit/9886f7737403d2ea378dd3349de4b70b1b62ce54

> Then try to use same config with newer version of this repo built from instruction and get Error: No J-Link device found. 

Not being smart but ... do you have a J-Link connected and configured correctly?

I thought that most or all SiFive boards were also accessible via an on board OpenOCD FTDI interface (similar to an Olimex etc.) but I can't seem to find a board config file that usese that sort of interface instead of J-Link.

Maybe check the SiFive support forums as sometimes stuff is posted there that may not be integrated into the OpenOCD fork.

Hope this helps.

Cheers
Tommy

TM1234, 2020-04-26:
Is this of any use?

https://forums.sifive.com/t/cant-get-hifive1-rev-b-to-work-on-arch-linux-at-all/2443/14

githubquestions, 2020-04-26:
Not really since hoping not to use platform io (minimal language support). Also no see cfg file for alternate interface. 

Perhaps possible J-Link not connected or configured correctly. Working with platform io, but not sure if this means much. Just followed J-Link instructions, not particularly familiar with their software. Are there any good heuristics to determine if something obvious is wrong there?

TM1234, 2020-04-26:
Use the J-Link software (Commander?) to ensure that the J-Link is working and can connect to the target. If that is OK then the openocd route through J-Link should also work as far as I know. All I can suggest is that you read all the relevant docs, the SiFive forum and maybe contact them directly for assistance.

Hope this helps.

githubquestions, 2020-04-26:
Okay, then J-Link is working.

githubquestions, 2020-04-27:
here is jlink output if it is helpful
```
Connecting to target via JTAG
ConfigTargetSettings() start
ConfigTargetSettings() end
TotalIRLen = 5, IRPrint = 0x01
JTAG chain detection found 1 devices:
 #0 Id: 0x20000913, IRLen: 05, Unknown device
Debug architecture:
  RISC-V debug: 0.13
  AddrBits: 7
  DataBits: 32
  IdleClks: 5
Memory access:
  Via system bus: No
  Via ProgBuf: Yes (16 ProgBuf entries)
DataBuf: 1 entries
  autoexec[0] implemented: Yes
Detected: RV32 core
CSR access via abs. commands: No
Temp. halted CPU for NumHWBP detection
HW instruction/data BPs: 8
Support set/clr BPs while running: No
HW data BPs trigger before execution of inst
RISC-V identified.
J-Link>
```

githubquestions, 2020-04-29:
Resolved -- thank you for suggesting posting in SiFive forums, I likely would not have done so otherwise!

closed 2020-04-29 by githubquestions