riscv-collab/riscv-openocd #469: Hifive1 Rev B support

KendiMedAmine, 2020-04-28:
I was trying to connect openocd to the hifive1 Rev B board using the modifications made by #456  and I noticed that the PID in my board is **1061** and not **1051** (using **lsusb**). OpenOCD did not detect the board until I added the **idProduct** attribute **1061**.
Is it an error or did I just miss something ?

alistair23, 2020-05-12:
I have the same problem.

timsifive, 2021-01-19:
I assume this can be fixed by adding a line to contrib/60-openocd.rules, but I don't have a board to test this myself.

alistair23, 2021-01-19:
Upstream OpenOCD has HiFive1 rev B support: https://repo.or.cz/openocd.git/commit/7e78c04f1c275176bdee8f9d2553300218164577

timsifive, 2021-01-19:
Great! That change is also present in this branch: https://github.com/riscv/riscv-openocd/commit/7e78c04f1c275176bdee8f9d2553300218164577

closed 2021-01-19 by timsifive