riscv-collab/riscv-openocd #474: Don't use MMU in M mode (and other situations)

TM1234, 2020-05-10:
See the related discussion on this commit.

https://github.com/riscv/riscv-openocd/commit/60eccb29677560e58a3133ec43c7185892024316

It was concluded there that this issue is not an issue with that commit per se but with the way that openocd currently tries to read (and write?) memory using the MMU if detected.

The issue was previously masked because many (e.g. SiFive based) RISC-V targets used the first hart to read memory and this hart did not use an MMU (e.g. the hart 0 e51 of the FU540 SoC). The purpose of this commit is to use the corret hart/"thread" to read memory and so now the MMU can come into play (e.g. on harts 1-4 u54_1-4 of the FU540 SoC).

It was also concluded there that the current openocd mechanism for using the MMU should be refined so that it does NOT get used if any of the the following apply:

(priv == M)
((priv == M) && (MPRV.MPP == 0))
((priv == M) && (MPRV.MPP == M))

where priv is the current privilege mode of the hart (as per table 1.1 of the privileged spec) and M is machine mode (2'b11 == decimal 3).

I'm not 100% sure if there might be other scenarios in which an MMU, although present and configured, should be avoided?

I wonder if the fix should go in this or some other closely related code - e.g. to add further checks before deciding whether or not to return enabled = 0 or 1?

https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv.c#L1374

aswaterman, 2020-05-10:
I believe the logic should be something along the lines of “ignore satp unless prv < M or (mstatus.MPRV=1 and mstatus.MPP < M)“.

timsifive, 2020-05-11:
Agreed that this is broken, and should be fixed in riscv_mmu(). I think Tommy's and Andrew's logic come down to the same thing.

TM1234, 2020-05-11:
Yes - I think that myseld and @aswaterman are on the same track with the logic - Andrew was simply more succinct! :-)

I just wonder if additional checks will significantly impact memory accesses? 
And I'm not sure that any state (priv or mprv.mpp) can be cached as it probably needs to be checked afresh each time?
On the other hand if it's needed then it's needed I guess.

Note also that a workaround for the moment is for the debugger (e.g. openocd debug script or similar) to explicitly set the satp to 0 although this is only OK if you're NOT doing a connect to running target debug session.

This can also be done from gdb (or gdbinit or whatever) as follows for the FU540

`thread set 2-5 set $satp=0`

TM1234, 2020-05-20:
Does this (code below) look OK?
If it does then I will create a PR for it.
I still have a few remaining concerns...

1. Am I covering all situations in which enabled should immediately be set to 1?
2. Am I using the appropriate "get register" calls as there seem to be a variety of different variants in the riscv target code?
3. Will the additional code will significantly impact performance of riscv_mmu()?
4. Is my approach to handling failure of riscv_get_register() is OK?

```
static int riscv_mmu(struct target *target, int *enabled)
{
	if (!riscv_enable_virt2phys) {
		*enabled = 0;
		return ERROR_OK;
	}

	if (riscv_rtos_enabled(target))
		riscv_set_current_hartid(target, target->rtos->current_thread - 1);

	/* If in Machine mode don't check SATP or use MMU */
	riscv_reg_t priv;
	riscv_reg_t mstatus;
	if ((riscv_get_register(target, &priv, GDB_REGNO_PRIV) == ERROR_OK) &&
		(riscv_get_register(target, &mstatus, GDB_REGNO_MSTATUS) == ERROR_OK) &&
		((priv == 3) || ((get_field(mstatus, MSTATUS_MPRV) == 1) && (get_field(mstatus, MSTATUS_MPP) == 3)))) {
		LOG_DEBUG("SATP/MMU ignored in M mode.");
		*enabled = 0;
		return ERROR_OK;
	}

	riscv_reg_t satp;
	if (riscv_get_register(target, &satp, GDB_REGNO_SATP) != ERROR_OK) {
		LOG_DEBUG("Couldn't read SATP.");
		/* If we can't read SATP, then there must not be an MMU. */
		*enabled = 0;
		return ERROR_OK;
	}

	if (get_field(satp, RISCV_SATP_MODE(riscv_xlen(target))) == SATP_MODE_OFF) {
		LOG_DEBUG("MMU is disabled.");
		*enabled = 0;
	} else {
		LOG_DEBUG("MMU is enabled.");
		*enabled = 1;
	}

	return ERROR_OK;
}
```

aswaterman, 2020-05-20:
I went back and looked at the Spike source code, and the condition we use there is "ignore `satp` if `(MPRV ? MPP : priv) == M`".  This is consistent with the priv spec.  I think this differs from your construction in that yours ignores `satp` unconditionally when priv=M, possibly because of my previous incorrect advice.

TM1234, 2020-05-20:
Thanks @aswaterman .
I'll tweak my code so and submit a PR for review if that's ok.

TM1234, 2020-05-26:
Thanks @timsifive.
Closed.

closed 2020-05-26 by TM1234