riscv-collab/riscv-openocd #480: Multiple definitions of bitbang_swd originating from jtag/drivers/bitbang.h

firephyz, 2020-05-25:
Bitbang.h seems to be missing an extern declaration on a global struct. I'm not sure if the original author intended the struct to be defined globally in the header file but it is defined with actual values in bitbang.c.

I added 'extern' at the start of the "declaration" in the header file and the source seems to build just fine.

hdante, 2020-08-20:
I can confirm both the bug and the fix.

matsbror, 2020-11-27:
I was just hit by the same issue, it seems not be fixed in the source yet.

timsifive, 2020-11-30:
Can you describe exactly what change you needed to make to get this to compile? bitbang_swd is declared extern in my copy of things.

firephyz, 2020-12-01:
@matsbror I believe it is fixed if I'm not mistaken: #[https://github.com/riscv/riscv-openocd/commit/c60252ac2b636c4d99b766a574b9df0966151696#diff-d51890a68f780fb7abba58c970e0635ea5484df813d66cff86779dbec1f4ec8d](url)

It's apparently only a problem on recent GCC compilers? (starting from GCC 10). The commit above mentions a change to defaulting to -fno-common.

@timsifive Are you having problems compiling recent after recent commits? I think all that needs to be changed is to add the 'extern' keyword in front of the const swd_driver struct in bitbang.h.

ryankosta, 2021-01-30:
I also just had this same issue with
cc (GCC) 10.2.1 20201125 (Red Hat 10.2.1-9)

timsifive, 2021-02-01:
> @timsifive Are you having problems compiling recent after recent commits? I think all that needs to be changed is to add the 'extern' keyword in front of the const swd_driver struct in bitbang.h.

I have no problems building, but I'm still on GCC 9.

bitbang_swd is already defined const in https://github.com/riscv/riscv-openocd/blob/riscv/src/jtag/drivers/bitbang.h#L70

If there is a simple fix that addresses this, I'm happy to merge it.

en-sc, 2024-01-22:
As mentioned above, the issue is fixed.

closed 2024-01-22 by en-sc