riscv-collab/riscv-openocd #484: ./configure option

kingstone1927, 2020-06-12:
how should I configure the make file for RISCV?

Thanks

TM1234, 2020-06-12:
I think you need to clarify/expand on your query.
What exactly do you mean, what are you trying to do or what problem are you having?
To build openocd you use the usual [bootstrap], configure and make approach.
It's even explained in the README.

This also seems like a straightforward question better asked on the RISC-V debug mailing list than an issue/bug to be logged here...

kingstone1927, 2020-06-12:
I'm sorry for not being clear with my question. I'm new to ubuntu and riscv. My question is when we type ./configure [option].
What options can I put there if I want to configure for a Riscv system. I know I can do ./configure  --help, but there are too many options and I am noob, so I don't know what i should do at that stage. I hope this clear up some abstractions.

TM1234, 2020-06-12:
Do you need to build it yourself?
Could you use the xPack Project binary builds from @ilg-ul ?
If you really do need to build it yourself then you could use @ilg-ul's docker based build scripts.

https://xpack.github.io/openocd/

Hope this helps.

timsifive, 2020-06-12:
RISC-V support is always included. You don't need to pass any special options to configure to include it.

kingstone1927, 2020-06-14:
> Do you need to build it yourself?
> Could you use the xPack Project binary builds from @ilg-ul?
> If you really do need to build it yourself then you could use @ilg-ul's docker based build scripts.
> I find that they simplify the process of building openocd and you can even point them at another openocd repo, branch and commit (e.g. sifive forked repo) if you don't want to build from @ilg-ul's snapshot sources.
> 
> https://xpack.github.io/openocd/
> 
> But if you need to do something custom then you'll simply have to bite the bullet and learn about the openocd configure script and the options it presents.
> 
> Hope this helps.

I think  I will need to educate myself more about openOCD and RISCV. Thank you so much.

kingstone1927, 2020-06-14:
> RISC-V support is always included. You don't need to pass any special options to configure to include it.

Thanks! that was exactly what I did. I just typed ./configure but had no idea what would happen

ShraddhaDevaiya, 2020-11-06:
Hi, I am facing the same issue. While extracting the binaries from https://github.com/xpack-dev-tools/openocd-xpack/releases/download/v0.10.0-15/xpack-openocd-0.10.0-15-linux-x64.tar.gz. this, I am not able to get **riscv** target.
And for ./configure option, it is giving an following error:
```
checking for makeinfo... no
configure: WARNING: Info documentation will not be built.
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables... 
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking for gcc option to accept ISO C99... none needed
checking for ranlib... ranlib
./configure: line 4513: syntax error near unexpected token `0.23'
./configure: line 4513: `PKG_PROG_PKG_CONFIG(0.23)'
```
How can I resolve this?
Can anyone please help?

TommyMurphyTM1234, 2020-11-06:
> Hi, I am facing the same issue. While extracting the binaries from https://github.com/xpack-dev-tools/openocd-xpack/releases/download/v0.10.0-15/xpack-openocd-0.10.0-15-linux-x64.tar.gz. this, I am not able to get **riscv** target.
> And for ./configure option, it is giving an following error:
> 

If you want to build openocd using the xPack approach then you need to follow the relevant build instructions:

https://xpack.github.io/openocd/#build-details
https://github.com/xpack-dev-tools/openocd-xpack/blob/xpack/README-BUILD.md

ShraddhaDevaiya, 2020-11-09:
Hi @TommyMurphyTM1234 , I tried this steps but still  I am not able to get target riscv, if possible can you suggest to solve that error of ```./configure``` ?

TommyMurphyTM1234, 2020-11-09:
If you get an error then you are not following @ilg-ul's instructions for running his docker based build scripts correctly. Please review them again and if you still get problems describe the *exact* steps that you are taking.

en-sc, 2024-01-22:
@kingstone1927, @ShraddhaDevaiya, is the issue still relevant? I would like to close it.

closed 2024-01-26 by en-sc