riscv-collab/riscv-openocd #485: the version check for the version in  dtmcs and in dmstatus

gxflying, 2020-06-18:
Now the openOCD get version check info from dtmcs, but i think the version in dmstatus is what actully matrers and imply the actual feature（in spec） the DM supporting.
So what do we need to do to fully use both of the versions(in  dtmcs and in dmstatus) ?

timsifive, 2020-06-19:
OpenOCD checks dmstatus version in dmstatus_read_timeout().

I don't understand what problem you are having.

gxflying, 2020-06-20:
The version in dtmcs  is  used to cllaim the dtm protocal which consists of "idcode" "dmi" and so on
The version in dmstatus is used to claim all the features a DM supports.  Am I right?  

then comes the question that  how should I use
OpenOCD when my DM complies with the 0.14 spec?
In this situation ,the dmstatus should be set to 2 ,right?
But the 0.14 spec says the dmstatus version should be 3 to be compliant.
actually this issue is not about  a problem but about a sugestion for version control  implementation in openocd.  the two versions should have two layers in code for each one.
You can think it will be very difficult to maintain if the dtmcs & dmstatus version both going on with diffrent number 0.15 0.16 0.17 ....

timsifive, 2020-06-23:
You're right. OpenOCD mostly implements 0.14, so it should accept both 2 and 3 in the dmversion field.

PR #489 (2020-06-25): Accept dmstatus.version==3 (0.14)
Fixes #485.

Change-Id: I60b3d68827ca726558bc28035c0b74c5cf0d9754
commit 0b1b9e20 (2020-06-25): Accept dmstatus.version==3 (0.14) (#489)

Fixes #485.

Change-Id: I60b3d68827ca726558bc28035c0b74c5cf0d9754
closed 2020-06-25 by timsifive