riscv-collab/riscv-openocd #488: dmactive check in openOCD

gxflying, 2020-06-23:
hi, sir , I find that the Debug Module spec says we should **poll and check** dmactive until it's high when we write dmactive =1  to confirm that the dm has worked normally.
but in the openOCD::riscv-013.c :: examine() ，I don't the the poll operation, actually it will return **ERROR_FAIL** if it readbcak dmactive=0 ?
I think it may NOT always work on the right way when DM takes a long time to finish the "reset" triggered by dmactive.
how  do you think about ti ?
https://github.com/riscv/riscv-openocd/blob/03f943ae239f59727ef47e46fffc31dd405a642d/src/target/riscv/riscv-013.c#L1604

[image]

TommyMurphyTM1234, 2020-06-23:
Isn't the issue here that you are citing the draft 0.14 debug spec but cross referencing it with the current 0.13 openocd implementation? The 0.13 spec says nothing about polling dmcontrol:dmactive:

[image]

I don't think that there's any openocd available yet that implements/conforms to the draft 0.14 spec and changes that it introduces since 0.13? I don't see any branch at https://github.com/riscv/riscv-openocd that fits the bill anyway.

gxflying, 2020-06-23:
yeahhh, the it's from version  0.14, 
and you are right when 013 is taken in consideration
but the openocd' 013 target suports not only spec 0.13 
think about the dmcs2 register, it's also implemented in 013 target
anyway the openocd is in progess and i'll close this issue,

TommyMurphyTM1234, 2020-06-23:
> but the openocd' 013 target suports not only spec 0.13
> think about the dmcs2 register, it's also implemented in 013 target

You're right - it looks like at least some draft 0.14 features are supported by riscv-013.c although in the case of dmcs2 it's optional and falls back gracefully to "base" 0.13 support if dmcs2 (e.g. halt groups) is not supported. So I guess that riscv-013.c is actually some union of 0.13 and (some) 0.14 features?

Unfortunately I'm not clear on the precise extent to which 0.14 features are supported in riscv-013.c though... :-(

timsifive, 2020-06-23:
riscv-013 should also support 0.14, which is very close to 0.13.

As far as which optional features are supported, it's generally the ones that SiFive has a need for.

Going back to the original filed issue, I agree that OpenOCD should wait for dmactive to go high after writing it high, before returning an error. This doesn't affect compatibility with 0.13.