riscv-collab/riscv-openocd #491: Trigger enumeration should check tselect *and* tdata1.type != 0

TommyMurphyTM1234, 2020-06-26:
See here for the issue which I mistakenly logged against the debug spec in the first instance.

https://github.com/riscv/riscv-debug-spec/issues/529

The current trigger enumeration only checks that an index written to tselect reads back as the same value to infer that the trigger exists.
It should also check that tdata1.type != 0 to be sure that a functional trigger exists at that index.

timsifive, 2020-06-26:
@TommyMurphyTM1234, are you able to test this fix: https://github.com/riscv/riscv-openocd/tree/enumerate_triggers

I don't have any targets easily available that let me reproduce this problem.

TommyMurphyTM1234, 2020-06-26:
Yes I will give it a go over the weekend.
Thanks.

TommyMurphyTM1234, 2020-07-01:
Hi @timsifive - sorry about the delay in getting back to this.
I tried that mod and it seems to work fine.
Our target hardware has 10 triggers and with your mod I get this with the openocd telnet interface and this all looks correct:

```
Open On-Chip Debugger                                     
> reset init                                              
Disabling abstract command writes to CSRs.                
Disabling abstract command writes to CSRs.                
Disabling abstract command writes to CSRs.                
Disabling abstract command writes to CSRs.                
Disabling abstract command writes to CSRs.                
> break                                                   
                                                          
> bp                                                      
> help bp                                                 
bp [<address> [<asid>] <length> ['hw'|'hw_ctx']]          
      list or set hardware or software breakpoint         
rbp address                                               
      remove breakpoint                                   
> bp 0x08000000 4 hw                                      
[0] Found 10 triggers                                     
[1] Found 10 triggers                                     
[2] Found 10 triggers                                     
[3] Found 10 triggers                                     
[4] Found 10 triggers                                     
breakpoint set at 0x08000000                              
                                                          
> bp 0x08000004 4 hw                                      
breakpoint set at 0x08000004                              
                                                          
> bp 0x08000008 4 hw                                      
breakpoint set at 0x08000008                              
                                                          
> bp 0x0800000c 4 hw                                      
breakpoint set at 0x0800000c                              
                                                          
> bp 0x08000010 4 hw                                      
breakpoint set at 0x08000010                              
                                                          
> bp 0x08000014 4 hw                                      
breakpoint set at 0x08000014                              
                                                          
> bp 0x08000018 4 hw                                      
breakpoint set at 0x08000018                              
                                                          
> bp 0x0800001c 4 hw                                      
breakpoint set at 0x0800001c                              
                                                          
> bp 0x08000020 4 hw                                      
breakpoint set at 0x08000020                              
                                                          
> bp 0x08000024 4 hw                                      
breakpoint set at 0x08000024                              
                                                          
> bp 0x08000028 4 hw                                      
Couldn't find an available hardware trigger.              
can't add breakpoint: resource not available              
                                                          
>                                                         
```

Thanks a lot
Cheers
Tommy

TommyMurphyTM1234, 2020-07-01:
BTW - do you know what "IVA" means?
(Doesn't seem to be RISC-V specific).
Instruction Virtual Address perhaps?

timsifive, 2020-07-01:
Thanks for testing this.

I do not know what IVA means. Your guess sounds good to me.

PR #496 (2020-07-07): Triggers with type=0 aren't real.
Fixes #491.

@TommyMurphyTM1234 confirmed this works.

Change-Id: Id01adcc68d8c7d95f7e86d49d5d2b0c97c9fb1b0
commit 8b929c64 (2020-07-07): Triggers with type=0 aren't real. (#496)

Fixes #491.

Change-Id: Id01adcc68d8c7d95f7e86d49d5d2b0c97c9fb1b0
closed 2020-07-07 by timsifive