riscv-collab/riscv-openocd #502: tcl server only returns 0x1a when i send "reg" + 0x1a

quantrpeter, 2020-08-08:
Hi
   tcl server only returns 0x1a when i send "reg" + 0x1a, is it a bug?
Thanks
Peter

TommyMurphyTM1234, 2020-08-08:
What is the exact command/code that you're issuing to the openocd tcl interface?
What is your debug target?

quantrpeter, 2020-08-08:
i found out "ocd_reg pc" can read out the register pc, but "ocd_reg" returns nothing again. I am using java to send out the bytes.

```
	static void write(OutputStream os, String str) throws IOException {
		os.write(str.getBytes());
		os.write(0x1a);
		os.flush();
	}

	static String read(InputStream is) throws IOException {
		int b;
		b = is.read();
		String str = "";
		while (b != -1 && b != 0x1a) {
			str += (char) b;
			b = is.read();
		}
		return str.trim();
	}
```

quantrpeter, 2020-08-08:
sorry, it works now. i forget to read some byte from the error stream, so it blocked everything, thanks

closed 2020-08-08 by quantrpeter