riscv-collab/riscv-openocd #503: how to escape the double quote of the command we send to capture command

quantrpeter, 2020-08-08:
hi
   we can send **capture "command"** to tcl port. If the command contains double quote, how to escape it? I tried changing " to \" , not working
Thanks
Peter

timsifive, 2020-08-11:
What is the capture command? What commands have a " in them? Do you have a specific example with more details of what you're doing?

quantrpeter, 2020-08-12:
i am sending the command via tcl port, not telnet. Some command need to use "capture", such as

capture "ocd_reg"

If pass the tcl statment to capture, such as echo "peter", then i need to escape the double quote. But i don't know how to

Thanks
Peter

timsifive, 2020-08-12:
I see. I haven't used the TCL port for anything. You might try asking about this on the regular openocd user mailing list.

JanMatCodasip, 2020-08-13:
This seems like a generic TCL string escaping issue (not related just to the "TCL port" of OpenOCD).

@quantrpeter - The following suggestion has worked for me - escape curly braces `{` and `}`, then wrap your string to a pair of curly braces:
https://stackoverflow.com/questions/5302120/general-string-quoting-for-tcl/5302213#5302213

If the problems persist, please try asking on openocd-user mail list: http://openocd.org/discussion/mailing-lists/

en-sc, 2024-01-22:
@quantrpeter, are you OK with the provided resolution? I would like to close the issue.

closed 2024-01-26 by en-sc