riscv-collab/riscv-openocd #507: how to make "reg" command to read all values from registers instead of cache

quantrpeter, 2020-08-11:
Hi
   how to make "reg" command to read all values from registers instead of cache?
Thanks
Peter

TommyMurphyTM1234, 2020-08-11:
Use the "force" option as documented in the openocd user guide.

quantrpeter, 2020-08-11:
Thanks, but i can only force one register such as "reg pc force", i can't force all registers. We are building a cpu-recorder, to record every values changes of all registers, if invoke the command per each register, we found it would be too slow, need over 15 seconds to read all 50+ registers.

TommyMurphyTM1234, 2020-08-11:
Ah - good point. Force only works with individual registers.
I don't see any obvious generic openocd or specific riscv way to do the same for all registers but maybe I'm overlooking something...

timsifive, 2020-08-11:
This seems like a target-independent issue with OpenOCD. If you get the problem addressed in mainline, I'll make sure to pull down whatever solution you come up with into riscv-openocd quickly.

quantrpeter, 2020-08-12:
Thanks all

closed 2020-08-12 by quantrpeter