riscv-collab/riscv-openocd #524: Will OpenOCD work for Risc-v MPU.

ghalib210, 2020-09-08:
Hello,

            I just want to debug my Applications on RISC-V MPU with this Openocd, can it work's. If anyone have info, can you please share with me.
 
Thanks.

timsifive, 2020-09-08:
I don't know what RISC-V MPU is. In general, this branch of OpenOCD will be able to debug RISC-V hardware, but there are probably some exceptions where that's not true.

ghalib210, 2020-09-09:
RISCV MPU i Mean to say processor, i  was using RISCV Hifive U540 Board.  I what to debug using openocd.

timsifive, 2020-09-09:
This version of OpenOCD definitely works with HiFive Unleashed. Here's a config file that you can use: https://github.com/riscv/riscv-tests/blob/master/debug/targets/SiFive/HiFiveUnleashed.cfg

ghalib210, 2020-12-30:
Hello Tim,

            i have lunched openocd as follow's with your configuration file :- 
https://github.com/riscv/riscv-tests/blob/master/debug/targets/SiFive/HiFiveUnleashed.cfg  
but am getting an error as follows:-
Error: The specified debug interface was not found (ftdi)
The following debug adapters are available:
1: jlink
2: remote_bitbang

Here i have an doubt like we need to connect any External debugger like J-tag or j-link to U540 target ?

can you please help me to know more info on openocd to u540.

TommyMurphyTM1234, 2020-12-30:
You need to use or configure/build an openocd with support for the debug interface(s) that you need.

If you don't want to build it for yourself then you could maybe try Liviu Ionescu's xPack builds which are built from upstream openocd sources plus sifive RISC-V fork mods.

https://xpack.github.io/openocd/
https://xpack.github.io/openocd/#releases

https://github.com/riscv/riscv-tests/blob/master/debug/targets/SiFive/HiFiveUnleashed.cfg

ghalib210, 2020-12-30:
> You need to use or configure/build an openocd with support for the debug interface(s) that you need. If you don't want to build it for yourself then you could maybe try Liviu Ionescu's xPack builds [...]

I have build openocd binary for risc-v and it's working fine, i want know info like, if am using openocd is it mandatory that openocd has to use with J-tag (or)  j-link hardware's.

if it is mandatory then which model of j-tag (or) j-link has to buy with respect to Hifive U540 board.

Thanks .

TommyMurphyTM1234, 2020-12-30:
You probably need to ask sifive about what's needed to debug their board but as far as I know it uses an FTDI interface for debugging. But whatever it uses you'll need an openocd build that supports the relevant debug interface.

________________________________ From: ghalibkhan notifications@github.com<mailto:notifications@github.com> Sent: Wednesday, December 30, 2020 10:05:08
AM To: riscv/riscv-openocd riscv-openocd@noreply.github.com<mailto:riscv-openocd@noreply.github.com> Cc: Subscribed subscribed@noreply.github.com<mailto:

timsifive, 2020-12-30:
> I have build openocd binary for risc-v and it's working fine, i want know info like, if am using openocd is it mandatory that openocd has to use with J-tag (or) j-link hardware's.

HiFive Unleashed boards have a USB-JTAG interface on-board. You don't need any hardware besides a USB cable.

ghalib210, 2020-12-31:
Hi tim,

Can help me how to use USB-JTAG on HiFive Unleashed board, As per the hifive-unleashed-getting-started-guide-v1p1.pdf document **Board Setup**  there was a point regarding JTAG as follows:-->
The USB connector has
**two serial interfaces**. The first contains the linux console running at 115200Hz . The **second
provides JTAG suitable for use with openocd**.

How can i setup the board to use JTAG with openOCD in detailed.

Regards,
Ghalib khan.

timsifive, 2020-12-31:
SiFive's expectation is that you would use Freedom Studio to debug HiFive Unleashed. This package includes everything you need for development.

If you just want to connect with OpenOCD, here's a configuration file for it that works with HiFive Unleashed: https://github.com/riscv/riscv-tests/blob/master/debug/targets/SiFive/HiFiveUnleashed.cfg

ghalib210, 2021-01-21:
hello Tim,

        As you mention, i have installed Freedom Studio to debug HiFive Unleashed, to debug application i have a created an hello world example and flashed it, once it flashed right my board FSBL got corrupted and not booting, any how i have resolved boot issue but again i stuck with Debugging and flashing application.

Do you how to over come this issue?

Thanks,
Ghalib khan.

TommyMurphyTM1234, 2021-01-21:
> but again i stuck with Debugging and flashing application.
> 
> Do you how to over come this issue?

You'll really need to post more detailed information about what you're trying to do and what's going wrong if you want assistance. There's almost nothing to go on there...

timsifive, 2021-01-21:
For issues specifically with Freedom Studio and HiFive Unleashed, I recommend posting at https://forums.sifive.com/ or contacting SiFive customer support (https://sifive.atlassian.net/servicedesk/customer/portals).

en-sc, 2024-01-22:
@ghalib210, is the issue still relevant? I would like to close it.

closed 2024-01-26 by en-sc