riscv-collab/riscv-openocd #543: #536 breaks CustomRegisterTest

timsifive, 2020-10-02:
Change #536 breaks CustomRegisterTest from https://github.com/riscv/riscv-tests/tree/master/debug

@sobuch can you take a look at this? Fixing the test might be an option, depending on exactly why it fails.

timsifive, 2020-10-02:
`./gdbserver.py ./targets/RISC-V/spike32-2-hwthread.py Registers` also fails now. I assume it's due to the same change.

sobuch, 2020-10-05:
Hi, thanks for letting me know about this issue. I looked into it and the most significant difference I see (in OpenOCD logs) is probably due to the switch of the expose-xyz commands to be per-hart.

With previous version both harts had the same exposed custom registers, but now only one of the harts gets any. OpenOCD message `gdb requested a non-existing register`, that I am getting with the new version makes me think this is causing the fail. On the other hand, I am also getting the following gdb message 
```
A problem internal to GDB has been detected,
further debugging may prove unreliable.

This is a bug, please report it.  For instructions, see:
<http://www.gnu.org/software/gdb/bugs/>.
```

I am not familiar enough with the debug tests to say for sure whether the issue is in the tests. Would you know what to change in the tests, so that both harts get the exposed registers? I think that if we can confirm thats causing the fail, the problem must be in tests or gdb and not in openocd

timsifive, 2020-10-05:
That makes sense. These tests rely on the custom registers being set up in the .cfg files, which are in https://github.com/riscv/riscv-tests/tree/master/debug/targets/RISC-V

If you edit them to set up the custom registers on each hart that should solve it. This might be tricky on https://github.com/riscv/riscv-tests/blob/master/debug/targets/RISC-V/spike-rtos.cfg, but `-rtos riscv` will be removed at the end of the year, so any way you can get that test to not fail is fine. (Some hack to return not applicable might be the simplest solution for that.)

sobuch, 2020-10-06:
With this change it doesnt fail for me anymore, is it a valid fix for this issue? https://github.com/sobuch/riscv-tests/commit/a54f04931dc260c5025fa7f8469482e5e90fba63

timsifive, 2020-10-06:
That looks good to me. Thank you! Can you turn that into a pull request?

JanMatCodasip, 2020-10-07:
> With this change it doesnt fail for me anymore, is it a valid fix for this issue? sobuch/riscv-tests@a54f049

Thank you Sam for looking into this. The fix appears all right. I like the use of TCL's foreach construct here.

sobuch, 2020-10-07:
Thank you @timsifive @JanMatCodasip. I opened https://github.com/riscv/riscv-tests/pull/297, think we can close this now and move any further discussion over there

closed 2020-10-07 by timsifive