riscv-collab/riscv-openocd #545: Questions about OpenOCD's handling of Trigger when timing before

jiangshuaili, 2020-10-14:
Hi, all

       When I was looking at the OpenOCD code, I had a question about the CPU's processing after the watchpoint was triggered:

       I saw the process riscv_openocd_resume in OpenOCD, can I understand the process of when a watchpoint is  triggered is as follows:
       Step 1. OpenOcd discover the a watchpoint is triggered.
       Step 2. OpenOcd report the information to GDB (reason and address).
       Step 3. GDB handles this event.
       Step 4. The GDB user operation ends and the release the program continues.
       Step 5. Since the trigger timing of the watchpoint is before the instruction execution, it is found in riscv_openocd_resume that if the current stop_reason is watchpoint, delete all watchpoints first, do one singlestep, insert all watchpoints, and then continue to run.

       My question is: since the timing of a watchpoint is triggered before the store instruction, that is, the store instruction has not been executed completely. So at the time of Step 3, how does GDB find that the value on the address it observes has changed?

       The process I understand is as follows:
       Step 1. OpenOcd discovered a watchpoint is triggered.
       Step 2. Openocd records that the current stop reason is watchpoint and records the address of the watchpoint.
       Step 3. Openocd do one singlestep to complete the current instruction，completes the store execution.
       Step 4. Feed back the information recorded in Step 2 to GDB (At this point, the store instruction has been executed, so when GDB checks the monitored address, it can also find that the value on the address has changed.).
       Step 5. When the user releases the program again, directly make the CPU quit debugging mode and continue to run.

      Do I understand that right? Or am I missing something?

timsifive, 2020-10-21:
Your first set of steps describe accurately what happens.

As you point out, this means that when GDB reads the memory that the trigger was set on, it will see the old value. To deal with this, OpenOCD could be changed to implement your second set of steps. You only really want that behavior when GDB needs to see the final value, so it would be better for GDB to implement this behavior.

Even better would be if GDB and OpenOCD were modified so GDB can actually pass the value to trigger on down, and then the hardware can do the comparison which will be orders of magnitude faster.

The reality is that GDB is not very aware of low-level hardware debugging and the features real hardware might have. As a result its hardware breakpoint features are fairly minimal.

jiangshuaili, 2020-10-28:
Hi tim

If you know about the GDB process for processing watchpoint(like watch *0x1000), you can omit this section:
1. Read the value at the address of the current observation while setting the watchpoint
2. Insert the watchpoint and release the program to running
3. When the debugging agent finds that the program has stopped , and GDB finds that the reason reported by the debugging agent is the observation point,  GDB will check whether the value at 0x1000 has changed? If it has not changed, then insert the watchpoint and release the program  to running again

As you mentioned, GDB is more efficient at debugging if the operation of checking whether the value on the observed address has changed and hand it over by the hardware, which is not possible with the current GDB now.

But for the timing when load/store triggers the watchpoint, there is a corresponding interface control on GDB:
GDB's GDBARCH interface can be configured whether the timing of the watchpoint is triggered before or after the load, store instruction. The function is called "set_gdbarch_have_nonsteppable_watchpoint".
This interface can control whether the value on the currently observed address is directly checked when GDB finds that it is triggered by the watchpoint, or whether the value on the observed address is checked after performing a single step first.The default is to check directly.

For RISCV, the Trigger types used on GDB are all before the instruction is triggered (timing suggested in Spec), so one of the following should be selected:
1. We can configure RISCV GDB to require a single step after its watchpoint is triggered
2. On the debugging agent, such as OpenOCD, after modifying the trigger of the watchpoint, record the reason first, perform a single step again, and then report to the case where the GDB debugging target stops

From my point of view, since the Debug Spec suggest that the trigger time of the watchpoint is before, we can modify GDB to take a step after the trigger of the observation point. Meanwhile, the corresponding debugging agent also needs to pay attention to this point, like Openocd, QEMU, GDBServer, and other debugging agents.

jiangshuaili, 2020-11-02:
Sorry, I didn't take a close look at the GDB code. I have find the answer.

Upon further inspection, I found that the GDB of RISCV has been used set_gdbarch_have_nonsteppable_watchpoint (GDbarch, 1).
GDB has already performed a step for the watchpoint, so I guess the step for watchpoint when resume in Openocd is just for compatibility with the GDB version that hasn't added "set_gdbarch_have_nonsteppable_watchpoint (GDbarch, 1)." before 2018.10.23.

So openocd's handling is fine.

closed 2020-11-02 by jiangshuaili