riscv-collab/riscv-openocd #546: Question about：Read/Write Double-Precision Float Point Registers  when RV32I using memory

jiangshuaili, 2020-10-14:
Hi， all

    In an RV32I CPU, if double-precision floating-point is configured, the floating-point register will be 63bits and GPR 32bits.

    In standard floating point instructions, we can only interact with GPR in the lower 32 bits of double precision floating point.So I see that OpenOCD's code implementation USES FSD and FLD instructions to read and write floating point registers via memory interaction.

    This seems unreasonable, because when it comes to memory, it can cause problems with reading and writing memory:
    1. Read and write memory will be affected by Cache, MMU, MPU, BUS, etc
    2. If the current SoC does not have RAM, the memory region used by the FSD/FLD will not be available
    3. Reading and writing to floating-point registers should be done by the debugging module alone, and no other resources should be requested from the system
 
    Based on the above description, in my opinion, this is a defect of floating point instruction for debugging, and it should be required by debugging module for instruction design:
    An instruction to design a high 32 - bit floating - point register to interact with GPR？

Rolfy360, 2020-10-14:
This is a pending issue
https://github.com/riscv/riscv-isa-manual/pull/301

timsifive, 2020-10-19:
It'll be great if such instructions are added. When they are, we can talk about OpenOCD support. As it is, this doesn't seem like an OpenOCD issue.

closed 2020-10-19 by timsifive