riscv-collab/riscv-openocd #548: Emulating MPRV

pdonahue-ventana, 2020-10-14:
The [RISC-V debug spec](https://github.com/riscv/riscv-debug-spec/blob/master/core_debug.tex#L57) says "In general, the debugger is expected to be able to simulate all the effects of MPRV."

Assuming MPP is S or U mode, MPRV has several effects:
1. translate VA->PA per satp
1. apply endianness using SBE/UBE in mstatus
1. apply virtual memory permissions (page table read/write/execute permissions)
1. apply PMP permissions (which, unless PMP is locked, are ignored in debug mode)
1. other things like mstatus.SUM

I see that `riscv_address_translate()` does the first item but I don't see any of the other things being done.

timsifive, 2020-10-14:
OpenOCD is implemented lazily, meaning we implement features as they are required. It would be better if it implemented everything the spec described, but nobody has volunteered the time to do that.

I will note that what you quote is in a clarification. Not sure how much weight we ought to give those as far as the spec is concerned. Certainly when I wrote that I only had translation in mind, and not everything else you listed. I'm pretty sure endianness was always little back then, for instance.

pdonahue-ventana, 2020-10-14:
OK, laziness is fine.  I'm trying to understand the OpenOCD code and I have some questions.
- Does the code assume that mprven is writable?  It appears to me that the answer is yes because it writes it and never checks the result.
- Can it tolerate mprven being hard-wired to 0 (as the debug spec recommends/assumes for implementations other than RV32 with 33 or 34 bits of PA)?  I would assume so since that's the behavior before mprven existed but I can't convince myself by reading the code that the answer is yes.
- Can it tolerate mprven being hard-wired to 1 (as the debug spec allows)?

Of course, the answer to any of these may be that it cannot handle certain things but eventually should (due to the lazy argument).  I'm just trying to understand the current state of affairs.

timsifive, 2020-10-21:
This is a bit messy. I think we can handle any amount of mprven support. What it looks like you missed when reading the code that whether to use mprven is a user option, and we rely on them not to configure use of mprven if the hardware doesn't support it. Per #522, there is a different option that makes OpenOCD manually attempt to translate the address. The configuration needs to be cleaned up. Potentially OpenOCD could autodetect whether mprven is implemented, but it doesn't do that currently.

pdonahue-ventana, 2020-10-22:
Thanks for the explanation.  That seems reasonable given the way the architecture defines mprven.

closed 2020-10-22 by pdonahue-ventana