riscv-collab/riscv-openocd #552: Debug Module did not become active. dmcontrol=0x0

kingstone1927, 2020-11-02:
I implemented SweRV_EH1 on a Zedboard fpga. I am trying to use openOCD and Jtag Arm-USB-Tiny_H to download code to the board as instructed in the README. I have checked the physical connection and make sure it be correctly connected to the Pmod on the Zedboard. Does anyone know what could potentially cause this error and ways to fix it? I would really appreciate it.

I run:
```openocd -f swerv_openocd.cfg```
here is the error I got:
```
Open On-Chip Debugger 0.10.0+dev-01404-g1ba1b8784 (2020-10-26-11:36)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
Info : clock speed 10 kHz
Info : JTAG tap: riscv.cpu tap/device found: 0x00000001 (mfg: 0x000 (), part: 0x0000, ver: 0x0)
Error: Debug Module did not become active. dmcontrol=0x0
Info : starting gdb server for riscv.cpu on 3333
Info : Listening on port 3333 for gdb connections
Error: Target not examined yet
```

Here is what it shows when I run with -d3 flag
```
/riscv-home/Cores-SweRV_fpga/software/bsp$ openocd -d3 -f swerv_openocd.cfg
Open On-Chip Debugger 0.10.0+dev-01404-g1ba1b8784 (2020-10-26-11:36)
Licensed under GNU GPL v2
For bug reports, read
http://openocd.org/doc/doxygen/bugs.html
User : 13 1 options.c:63 configuration_output_handler(): debug_level: 3
User : 14 1 options.c:63 configuration_output_handler():
Debug: 15 2 options.c:244 add_default_dirs(): bindir=/usr/local/bin
Debug: 16 2 options.c:245 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 17 2 options.c:246 add_default_dirs(): exepath=/usr/local/bin
Debug: 18 2 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 19 2 configuration.c:42 add_script_search_dir(): adding /home/riscv/.config/openocd
Debug: 20 2 configuration.c:42 add_script_search_dir(): adding /home/riscv/.openocd
Debug: 21 2 configuration.c:42 add_script_search_dir(): adding /usr/local/bin/../share/openocd/site
Debug: 22 2 configuration.c:42 add_script_search_dir(): adding /usr/local/bin/../share/openocd/scripts
Debug: 23 2 configuration.c:97 find_file(): found swerv_openocd.cfg
Debug: 24 2 command.c:146 script_debug(): command - adapter speed 100
Debug: 26 2 core.c:1822 jtag_config_khz(): handle jtag khz
Debug: 27 2 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 28 2 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 29 2 command.c:146 script_debug(): command - adapter driver ftdi
Debug: 31 2 command.c:352 register_command_handler(): registering 'ftdi_device_desc'...
Debug: 32 2 command.c:352 register_command_handler(): registering 'ftdi_serial'...
Debug: 33 2 command.c:352 register_command_handler(): registering 'ftdi_channel'...
Debug: 34 2 command.c:352 register_command_handler(): registering 'ftdi_layout_init'...
Debug: 35 2 command.c:352 register_command_handler(): registering 'ftdi_layout_signal'...
Debug: 36 2 command.c:352 register_command_handler(): registering 'ftdi_set_signal'...
Debug: 37 2 command.c:352 register_command_handler(): registering 'ftdi_get_signal'...
Debug: 38 2 command.c:352 register_command_handler(): registering 'ftdi_vid_pid'...
Debug: 39 2 command.c:352 register_command_handler(): registering 'ftdi_tdo_sample_edge'...
Debug: 40 2 command.c:352 register_command_handler(): registering 'ftdi_oscan1_mode'...
Debug: 41 2 command.c:146 script_debug(): command - ftdi_device_desc Olimex OpenOCD JTAG ARM-USB-TINY-H
Debug: 43 2 command.c:146 script_debug(): command - ftdi_vid_pid 0x15ba 0x002a
Debug: 45 2 command.c:146 script_debug(): command - ftdi_layout_init 0x0808 0x0a1b
Debug: 47 2 command.c:146 script_debug(): command - ftdi_layout_signal nSRST -oe 0x0200
Debug: 49 2 command.c:146 script_debug(): command - ftdi_layout_signal LED -data 0x0800
Debug: 51 2 command.c:146 script_debug(): command - transport select jtag
Debug: 52 2 command.c:352 register_command_handler(): registering 'jtag_flush_queue_sleep'...
Debug: 53 2 command.c:352 register_command_handler(): registering 'jtag_rclk'...
Debug: 54 2 command.c:352 register_command_handler(): registering 'jtag_ntrst_delay'...
Debug: 55 2 command.c:352 register_command_handler(): registering 'jtag_ntrst_assert_width'...
Debug: 56 2 command.c:352 register_command_handler(): registering 'scan_chain'...
Debug: 57 2 command.c:352 register_command_handler(): registering 'runtest'...
Debug: 58 2 command.c:352 register_command_handler(): registering 'irscan'...
Debug: 59 2 command.c:352 register_command_handler(): registering 'verify_ircapture'...
Debug: 60 2 command.c:352 register_command_handler(): registering 'verify_jtag'...
Debug: 61 2 command.c:352 register_command_handler(): registering 'tms_sequence'...
Debug: 62 2 command.c:352 register_command_handler(): registering 'wait_srst_deassert'...
Debug: 63 2 command.c:352 register_command_handler(): registering 'jtag'...
Debug: 64 2 command.c:352 register_command_handler(): registering 'jtag'...
[... 205 lines cut ...]
Debug: 275 33 riscv-013.c:404 scan(): 41b w 07ffffc1 @10 -> + 00000000 @00; 0i
Debug: 276 33 riscv-013.c:414 scan(): hasel hartselhi=1023 hartsello=1023 dmactive ->
Debug: 277 34 riscv-013.c:404 scan(): 41b - 00000000 @10 -> + 00000000 @00; 0i
Debug: 278 34 riscv-013.c:404 scan(): 41b r 00000000 @10 -> + 00000000 @00; 0i
Debug: 279 35 riscv-013.c:404 scan(): 41b - 00000000 @10 -> + 00000000 @00; 0i
Error: 280 35 riscv-013.c:1615 examine(): Debug Module did not become active. dmcontrol=0x0
Debug: 281 35 target.c:1650 target_call_event_callbacks(): target event 20 (examine-fail) for core riscv.cpu
Debug: 282 35 openocd.c:160 handle_init_command(): target examination failed
Debug: 283 35 command.c:146 script_debug(): command - flash init
Debug: 285 35 tcl.c:1325 handle_flash_init_command(): Initializing flash devices...
Debug: 286 35 command.c:146 script_debug(): command - nand init
Debug: 288 35 tcl.c:498 handle_nand_init_command(): Initializing NAND devices...
Debug: 289 35 command.c:146 script_debug(): command - pld init
Debug: 291 35 pld.c:206 handle_pld_init_command(): Initializing PLDs...
Info : 292 35 gdb_server.c:3609 gdb_target_start(): starting gdb server for riscv.cpu on 3333
Info : 293 35 server.c:311 add_service(): Listening on port 3333 for gdb connections
Debug: 294 35 command.c:146 script_debug(): command - halt
Debug: 296 35 target.c:3109 handle_halt_command(): -
Error: 297 35 target.c:581 target_halt(): Target not examined yet
Debug: 298 35 command.c:626 run_command(): Command 'halt' failed with error code -4
User : 299 36 command.c:692 command_run_line():
Debug: 300 36 riscv.c:486 riscv_deinit_target(): riscv_deinit_target()
Debug: 301 36 riscv-013.c:1533 deinit_target(): riscv_deinit_target()
Debug: 302 36 target.c:2004 target_free_all_working_areas_restore(): freeing all working areas
```

JanMatCodasip, 2020-11-03:
Hi kingstone1927,

This appears not to be an issue of OpenOCD. Current version of OpenOCD is known to work fine with SweRV. It is more likely a problem with SweRV integration into your design. Please continue the discussion in the SweRV repo, where you have better chance of getting advice for your problem: https://github.com/chipsalliance/Cores-SweRV/issues/

Few notes that may help you:

It appears that access to RISC-V debug module registers is not working for you.

The log shows it is possible to read DTMCS JTAG register but not DMI register:

```
Debug: 262 29 riscv-013.c:458 dtmcontrol_scan(): DTMCS: 0x0 -> 0x71
Debug: 263 29 riscv-013.c:1575 examine(): dtmcontrol=0x71
Debug: 264 29 riscv-013.c:1576 examine(): dmireset=0
Debug: 265 29 riscv-013.c:1577 examine(): idle=0
Debug: 266 29 riscv-013.c:1578 examine(): dmistat=0
Debug: 267 29 riscv-013.c:1579 examine(): abits=7
Debug: 268 29 riscv-013.c:1580 examine(): version=1
Debug: 269 29 riscv-013.c:264 get_dm(): [0] Allocating new DM
Debug: 270 30 riscv-013.c:404 scan(): 41b w 00000000 @10 -> + 00000000 @00; 0i
Debug: 271 30 riscv-013.c:404 scan(): 41b - 00000000 @10 -> + 00000000 @00; 0i
Debug: 272 31 riscv-013.c:404 scan(): 41b w 00000001 @10 -> + 00000000 @00; 0i
Debug: 273 31 riscv-013.c:414 scan(): dmactive ->
Debug: 274 32 riscv-013.c:404 scan(): 41b - 00000000 @10 -> + 00000000 @00; 0i
Debug: 275 33 riscv-013.c:404 scan(): 41b w 07ffffc1 @10 -> + 00000000 @00; 0i
Debug: 276 33 riscv-013.c:414 scan(): hasel hartselhi=1023 hartsello=1023 dmactive ->
Debug: 277 34 riscv-013.c:404 scan(): 41b - 00000000 @10 -> + 00000000 @00; 0i
Debug: 278 34 riscv-013.c:404 scan(): 41b r 00000000 @10 -> + 00000000 @00; 0i
Debug: 279 35 riscv-013.c:404 scan(): 41b - 00000000 @10 -> + 00000000 @00; 0i
```

Please focus on your hardware design  - especially make sure whether the SweRV processor core and the debug module are not held in reset and clocked properly. Also note that there is a separate reset signal for the debug module (`dbg_rst_l`) that needs to be also treated properly.

BTW, it seems that you use command `riscv set_mem_access progbuf`. This is incorrect as SweRV does not implement the debug program buffer. Please use SweRV 1.8 and configure your memory access this way: `riscv set_mem_access progbuf`.

Hope this helped. Again, this appears to be an issue with the hardware, not with OpenOCD. Please try asking at the SweRV repo, as noted above.

Regards,
Jan

kingstone1927, 2020-11-04:
Thank you Jan! This is a great response. I really appreciate you. I will check my hardware design and may try the new SweRV core version

wizkad, 2023-12-01:
@kingstone1927 how did you fix your issue? i'm having a similar issue where it isn't able to read the control register to examine the target. Could you provide your openocd configuration file?

TommyMurphyTM1234, 2023-12-01:
> @kingstone1927 how did you fix your issue? i'm having a similar issue where it isn't able to read the control register to examine the target. Could you provide your openocd configuration file?

* https://github.com/riscv/riscv-openocd/issues/966#issuecomment-1835795496

en-sc, 2024-01-22:
@kingstone1927, @wizkad, is the issue still relevant? I would like to close it.

closed 2024-01-26 by en-sc