riscv-collab/riscv-openocd #553: Can not debug Linux kernel with gdb + openocd on Genesys2 Riscv FPGA chip when MMU is enabled

hilihui, 2020-11-06:
I use gdb with openocd to debug linux kernel on Genesys2 Riscv FPGA chip. First,  I set a breakpoint on kernel entry(start_kernel) with physical address(the physical address is 0x802006be), the breakpoint can work. But from here, the following debug will fail,
the error is:
[image]

timsifive, 2020-11-18:
I've seen "Reply contains invalid hex digit" before. I think it happens when OpenOCD prints out an error that gdb interprets as a packet.

If you run openocd with `-d` and log the output to a file (it will be very large) then it should be possible to extract the error message from there. (There is a also a way to have gdb log the protocol data but I can never remember it.)

timsifive, 2021-01-15:
Idle.

closed 2021-01-15 by timsifive