riscv-collab/riscv-openocd #559: src/target/riscv/riscv-013.c:read_memory() failure corrupts x8/s0/fp register

TommyMurphyTM1234, 2020-12-09:
During debugging I found that the frame pointer (x8/s0/fp), which is also used by the debugger for certain operations, was getting corrupted. I added some logging to the above file and see this:

```
Info : read_memory:3560 Begin         x8/s0/fp = 0x8013000
Warn : Failed to read memory via program buffer.
Warn : Failed to read memory via abstract access.
Error: Target mpfs.hart0_e51: Failed to read memory (addr=0x500000000000000)
Error:   progbuf=failed, sysbus=skipped (unsupported size), abstract=failed
Info : read_memory:3621 End (failure) x8/s0/fp = 0x500000000000000

```

So it seems that a failed memory read leaves the frame pointer corrupted with obvious concomitant negative impact on further execution of the program being debugged.

I guess that at least one code path fails to restore the x8/s0/fp register used by openocd to do some work. There are several places where it saves and restores the original value but in this case I guess that the restore never happens.

One obvious solution would be to save it at the start of read_memory() and restore it at the end (success and failure cases?) but maybe this is not the appropriate place?

@timsifive - any suggestions?

Thanks a lot
Cheers
Tommy

JanMatCodasip, 2020-12-09:
@TommyMurphyTM1234 - 

Hi Tommy,
thank you for reporting this!

The only memory access method that uses the `s0` register is the access via program buffer. The other two methods (system bus & abstract access) do not modify registers. 

So there must be a code path in `read_memory_progbuf()`, `read_memory_progbuf_inner()` or `read_memory_progbuf_one()` that on failure leaves the `s0` register clobbered (maybe also `s1` and `mstatus`). So the bug is very likely in one of those functions and should be fixed directly there

Would you please have a moment to add more debug prints to these functions to find out at which point the memory access fails in your case?

Note that there may be a similar bug in `write_memory_progbuf()` et. al.

Thanks.

TommyMurphyTM1234, 2020-12-09:
Hi @JanMatCodasip - thanks for the pointer to the appropriate code. I will add logging to those functions to see what code path(s) fail to restore s0 properly and will post back when I have more info. I'll also look at the write case too. Do you have any idea if the same or a similar issue might affect riscv-011.c too?

Cheers
Tommy

JanMatCodasip, 2020-12-09:
Thank you for taking the time to look at it.

Unfortunately with `riscv-011.c` I do not have any experience. So far I have been working with RISC-V Debug v0.13 implementations only.

TommyMurphyTM1234, 2020-12-09:
Ok thanks @JanMatCodasip  - I'll have a look at riscv-011.c after I figure riscv-013.c out first. :-)

TommyMurphyTM1234, 2020-12-09:
Here is what I found...

My main problem is that it is not at all clear which error returns are ultimately fatal and which are caught by the caller and handled without causing openocd to fail totally. Obviously there is no point in restoring registers in the former case but it is critical in the latter.

I have only gone through the riscv-013.c memory read functions and not the riscv-013.c memory write functions or anything in riscv-011.c. 

I'm happy to try making changes here but I wonder if the code needs more extensive refactoring/rearrangement in order to handle non fatal return situations and working register restoration better?

(Would be so much nicer and simpler if this was done in C++ :-)

`read_memory_progbuf()`
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3405

Uses: mstatus, s0, s1 and s2

Saves them here: 
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3434

Restores them here: 
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3509

But can return with failure without restoring them here:
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3440
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3442
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3442
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3467
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3478
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3480

`read_memory_progbuf_one()`
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3333

Uses: mstatus and s0

Saves them here:
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3337

Restores them here:
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3391

But can return with failure without restoring them here:
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3344
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3366
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3372
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3374
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3378
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3383
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3387
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3392

`read_memory_progbuf_inner()`
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3097

Uses: s0, s1 and s2

It does no saving/restoring of these registers so is it assumed that these are already saved/restored at a higher level?

TommyMurphyTM1234, 2020-12-09:
`write_memory_progbuf()`
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3757

Uses: mstatus, s0 and s1

Saves them here:
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3774

Restores them here:
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3933

But can return with failure without restoring them here:
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3784
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3786
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3809
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3934
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3936
https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3941

Note that write_memory_progbuf() attempts to employ a single point of exit so that both failure and success returns restore the registers but it there are still cases where the restoration does not happen. read_memory_progbuf() is not structured the same way at the moment.

TommyMurphyTM1234, 2020-12-09:
And looking at other parts of the riscv-013.c code I can see cases where other functions (unrelated to memory progbuf read) can return before all registers used have been correctly restored to their original values. Again maybe in many, most or all cases such a failure is ultimately fatal for openocd anyway but it's not clear from the code if some may be non fatal returns handled by the caller. 

So I'm a bit reluctant to try to change this code without a fuller understanding of it and would appreciate input from others if possible.

It might help if there were something like ERROR_FAIL_FATAL and ERROR_FAIL_NONFATAL return values but the fact that there's only ERROR_FAIL is probably a more general openocd issue/limitation?

TommyMurphyTM1234, 2020-12-09:
I also had a quick look at riscv-011.c but it doesn't seem to use registers such as mstatus, s0-2 in the same way so I presume that it's not impacted by the same issue as riscv-013.c whereby failed (progbuf based?) memory reads (and writes?) can corrupt some or all of these registers?

timsifive, 2020-12-09:
@TommyMurphyTM1234, instead of adding some selective print statements, can you collect a failure using `openocd -l`? That will surely include anything you would have thought to add print statements for, and should allow us to narrow down as close as we can where the register gets corrupted.

It seems possible there is a code path where registers aren't restore correctly, although I'm a bit surprised that nobody has run into it so far. fp is critical to most code, so it would be obvious to almost anybody running into this case.

riscv-011 is fairly different from riscv-013. If there are bugs I will try to fix them, but in general I see this as legacy code for the dwindling amount of hardware that supports that version of the spec. (0.11 was not ratified. 0.13 was.)

timsifive, 2020-12-09:
PS
One big difference between riscv-011 and riscv-013 is that the former always saves all GPRs on halting, and restores them when resuming. The latter attempts to leave all the GPRs with their original values whenever it's idle. So the relevant code paths are pretty different. (I prefer the riscv-011 model, and might some day change riscv-013 to work that way also.)

TommyMurphyTM1234, 2020-12-09:
> PS
> One big difference between riscv-011 and riscv-013 is that the former always saves all GPRs on halting, and restores them when resuming. The latter attempts to leave all the GPRs with their [...]

Thanks Tim.
So the issue of saving and restoring GPRs is not present in 0.11?
What about CSRs - e.g. mstatus - if that's even used at all in the 0.11 openocd support?

BTW - Having looked at the riscv-013.c code I'm not sure if my original issue should be generalized to something like "review all failure returns from all riscv-013.c functions to ensure that they restore any registers that were previously overwritten" sicne this issue potentially exists in many functions and not just the memory read/write ones. However, as I said, without understanding all of the riscv-013.c code (and maybe also caller code) it seems difficult to distinguish a "fatal" ERROR_FAILfrom a "non fatal" ERROR_FAIL - i.e. the former ultimately causing openocd to stop and the latter being handled by the caller and not causing openocd to stop.

timsifive, 2020-12-09:
> So the issue of saving and restoring GPRs is not present in 0.11?

It does happen, but differently than in 0.13. I would guess the code is more reliable, but I'm not aware of any bugs in either. Is your target using 0.11?

I agree that it's tricky to audit the 0.13 code. The best way to figure out the problem you're seeing is to get a full log that shows the problem. That should at a minimum narrow down the code where it could have occurred.

TommyMurphyTM1234, 2020-12-09:
> @TommyMurphyTM1234, instead of adding some selective print statements, can you collect a failure using `openocd -l`? That will surely include anything you would have thought to add print statements for, and should allow us to narrow down as close as we can where the register gets corrupted.

I can do that but it will only cover the specific failure case that I am experiencing as explained in the original post - i.e. the debugger trying to read from 0x0500_0000_0000_0000 because that's what an (as yet uninistialized) pointer on the stack is pointing to and the Variables view is active or I mouse/hover over the varialble and the debugger tries to read some info about 
the variable (note that I am working in the context of Eclipse/CDT/xPack plugins here).

the reason I tend to avoid the default openocd logging is that it's messy, too verbose, impacts performance and sometimes functionality and takes a kitchen sink approach rather than facilitating selective logging so I don't find it useful in most cases.

> It seems possible there is a code path where registers aren't restore correctly, although I'm a bit surprised that nobody has run into it so far. fp is critical to most code, so it would be obvious to almost anybody running into this case.

Some of what I see may be due to the configurable nature of our target (Microchip PolarFire SoC FPGA + hard U540 RISC-C coreplex). Others using more "fixed function" MCUs may not run into similar problems?

But on looking at the code I see many potential cases of failure returns without restoring registers - some of which may be fatal anyway but some of which are definitely non fatal so leave registers of the program being debugged corrupted. That's why I was suggesting maybe a review of all such cases and perhaps a refactor of some code to, for example. use a single point of exit rather than multiple points of exit so that register restoration can be done once in all cases.

> riscv-011 is fairly different from riscv-013. If there are bugs I will try to fix them, but in general I see this as legacy code for the dwindling amount of hardware that supports that version of the spec. (0.11 was not ratified. 0.13 was.)

OK - I undestand. Thanks.

TommyMurphyTM1234, 2020-12-09:
> Is your target using 0.11?

Actually our RV32 FPGA soft cores that use 0.11 are probably deprecated now so our focus is mostly on 0.13 too. 

> I agree that it's tricky to audit the 0.13 code. The best way to figure out the problem you're seeing is to get a full log that shows the problem. That should at a minimum narrow down the code where it could have occurred.

I can collect more logging but the original post seems to contain all the relevant info for that specific case - i.e. the attempted memory read fails and, as a result, the x8/s0/fp register (and maybe others) does not get restored so is corrupted when the program being debugged resumes.

But again - that's just one scenario that I happend to come across and I am cognizant of the possibility that there are more.

timsifive, 2020-12-09:
> the reason I tend to avoid the default openocd logging is that it's messy, too verbose, impacts performance and sometimes functionality and takes a kitchen sink approach rather than facilitating selective logging so I don't find it useful in most cases.

I can sympathize with all of that. The reason I generally ask for the full log is that it saves a bunch of back and forth. The log files are messy, but I'm reasonably experienced at digging through them to find the interesting stuff.

TommyMurphyTM1234, 2020-12-09:
Hi @timsifive - you mentioned openocd -l earlier but that is an unsupported option. I presume you meant -d?
Attached is a -d log in case that helps at all.
Note that although I discovered the issue originally using Eclipse/CDT/GNU MCU Eclipse xPack plugins in this case I reproduced it using just openocd and gdb command line so the specific details may be slightly different but the net effect is the same.
In Eclipse the issue arose because mouse hovering over the pointer variable on the stack caused the attempted memory read of the "bad address" or the Variables view did likewise. 
If you have any questions about the log or the test case let me know.
In the meantime I will look again at riscv-013.c to enumerate all potential failure returns without register restoration just to see if (m)any of these might need attention.

Thanks
Cheers
Tommy

[openocd.zip](https://github.com/riscv/riscv-openocd/files/5669341/openocd.zip)

JanMatCodasip, 2020-12-10:
Hi @TommyMurphyTM1234 -

Thank you for spending the time on this and for collecting the log.

From your log, it appears the bug you hit is in `read_progbuf_one()`.
The execution of program buffer at [riscv-013.c:3382](https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3382) fails (because of the out-of-range address):

```
Debug: 104456 98351 riscv-013.c:785 execute_abstract_command(): command=0x371008; access register, size=64, postexec=1, transfer=1, write=1, regno=0x1008
Debug: 104457 98352 riscv-013.c:404 scan(): 41b w 00371008 @17 -> + 00000000 @00; 3i
Debug: 104458 98353 riscv-013.c:404 scan(): 41b r 00000000 @16 -> + 00000000 @00; 3i
Debug: 104459 98354 riscv-013.c:404 scan(): 41b - 00000000 @16 -> + 10000302 @16; 3i
Debug: 104460 98354 riscv-013.c:414 scan():  ->  progbufsize=16 cmderr=3 datacount=2
Debug: 104461 98355 riscv-013.c:808 execute_abstract_command(): command 0x371008 failed; abstractcs=0x10000302
```

As a result, the value of S0, saved at [riscv-013.c:3343](https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3343), will not get restored.

As you pointed out in earlier comments, there are many more suspicious places that may be prone to the same issue (registers not restored after a failed operation) and they should be reviewed. 

My recommendation for laying out the code would be:
- The saved registers should be restored in that same function (which is perhaps already be true for the current code).
- There should be a single point of exit in such functions so that likelihood of these mistakes is minimized.

There was also question about which errors are fatal and which are recoverable. IMO, majority of the errors should be treated as recoverable - only a certain operation failed, but that should not mess with the processor state or the subsequent operations. There will be few exceptions - for example failures during the initial probing (examination) of the target.

TommyMurphyTM1234, 2020-12-10:
Thanks @JanMatCodasip - I agree with your comments/suggestions.
I will look at a fix for the specific issue that I have hit and try to scope a more general review of the code in relation to failure returns and ensuring that registers are restored properly.

TommyMurphyTM1234, 2020-12-10:
This is a bit of a strawman but I tried to restructure `read_memory_progbuf_one()` (https://github.com/riscv/riscv-openocd/blob/riscv/src/target/riscv/riscv-013.c#L3333) to more safely deal with register save/restore.

Notes/caveats that I can think of:

1. Many people may choke at the use of `goto `but it's already used in this file to implement a single point of return and this may be considered a legitimate use of the keyword. The jump from a block scope (e.g. the `switch` statement here) grates with me though.
2. The code may be more verbose than absolutely necessary but I just wanted to illustrate one way of tackling this general issue with register save/restore in riscv-013.c
3. I tend towards verbose/descriptive variable names which lends itself to point #2 :-)
4. I'm not really sure about the logic involved in only attempting mstatus restore if the previous restore of s0 has succeeded. I suspect that if registers cannot be read/written then there is something serious wrong and it's probably going to be a catastrophic failure anyway so maybe this is moot?
5. I haven't actually compiled this code yet! :-)

Comments/feedback welcome.

```
/* Only need to save/restore one GPR to read a single word, and the progbuf
 * program doesn't need to increment. */
static int read_memory_progbuf_one(struct target *target, target_addr_t address,
		uint32_t size, uint8_t *buffer)
{
	/* result to return - assume failure */
	int result = ERROR_FAIL;
	
	/* Saved registers */
	uint64_t mstatus_saved = 0;
	uint64_t s0_saved = 0;
	uint64_t mstatus_local = 0;
	
	/* Were registers successfully saved? */
	bool mstatus_was_saved = false;
	bool s0_was_saved = false;

	/* Change privilege & save mstatus */
	if (modify_privilege(target, &mstatus_local, &mstatus_saved) == ERROR_OK)
		mstatus_was_saved = true;
	else
		goto cleanup_and_return;

	/* Save s0 */
	if (register_read(target, &s0_saved, GDB_REGNO_S0) == ERROR_OK)
		s0_was_saved = true;
	else
		goto cleanup_and_return;

	/* Write the program (load, increment) */
	struct riscv_program program;
	riscv_program_init(&program, target);
	if (riscv_enable_virtual && has_sufficient_progbuf(target, 5) && get_field(mstatus, MSTATUS_MPRV))
		riscv_program_csrrsi(&program, GDB_REGNO_ZERO, CSR_DCSR_MPRVEN, GDB_REGNO_DCSR);
	switch (size) {
		case 1:
			riscv_program_lbr(&program, GDB_REGNO_S0, GDB_REGNO_S0, 0);
			break;
		case 2:
			riscv_program_lhr(&program, GDB_REGNO_S0, GDB_REGNO_S0, 0);
			break;
		case 4:
			riscv_program_lwr(&program, GDB_REGNO_S0, GDB_REGNO_S0, 0);
			break;
		case 8:
			riscv_program_ldr(&program, GDB_REGNO_S0, GDB_REGNO_S0, 0);
			break;
		default:
			LOG_ERROR("Unsupported size: %d", size);
			goto cleanup_and_return;
	}
	if (riscv_enable_virtual && has_sufficient_progbuf(target, 5) && get_field(mstatus, MSTATUS_MPRV))
		riscv_program_csrrci(&program, GDB_REGNO_ZERO,  CSR_DCSR_MPRVEN, GDB_REGNO_DCSR);

	if (riscv_program_ebreak(&program) != ERROR_OK)
		goto cleanup_and_return;
	if (riscv_program_write(&program) != ERROR_OK)
		goto cleanup_and_return;

	/* Write address to S0, and execute buffer. */
	if (write_abstract_arg(target, 0, address, riscv_xlen(target)) != ERROR_OK)
		goto cleanup_and_return;

	uint32_t command = access_register_command(target, GDB_REGNO_S0,
			riscv_xlen(target), AC_ACCESS_REGISTER_WRITE |
			AC_ACCESS_REGISTER_TRANSFER | AC_ACCESS_REGISTER_POSTEXEC);
	if (execute_abstract_command(target, command) != ERROR_OK)
		goto cleanup_and_return;

	uint64_t value;
	if (register_read(target, &value, GDB_REGNO_S0) != ERROR_OK)
		goto cleanup_and_return;
	buf_set_u64(buffer, 0, 8 * size, value);
	log_memory_access(address, value, size, true);
	
	result = ERROR_OK;

cleanup_and_return:

	/* Restore s0 if necessary */
	if (s0_was_saved)
		if (riscv_set_register(target, GDB_REGNO_S0, s0) != ERROR_OK)
			result = ERROR_FAIL;

	/* Restore mstatus if necessary */
	if (result == ERROR_OK) /* Only proceed if restore of s0 succeeded?! */
		if (mstatus_was_saved)
			if (mstatus_local != mstatus_saved)
				if (register_write_direct(target, GDB_REGNO_MSTATUS, mstatus_saved))
					result = ERROR_OK;

	return result;
}

```

timsifive, 2020-12-10:
Thanks for narrowing this down. I'm fixing at least this one function. I have a fix in theory, but I want to write a test for this failure to confirm that it really fixes the problem.

timsifive, 2020-12-10:
I had trouble reproducing this because spike has a bug where it ignores changes to fp when they are combined with a progbuf execution. So it was hiding this bug in OpenOCD. :-)

Anyway, there's a PR linked that I think fixes Tommy's issue. @TommyMurphyTM1234 , are you able to test that?

TommyMurphyTM1234, 2020-12-10:
Thanks a lot @timsifive - I tried that (only on Windows so far) and the $fp corruption no longer happens in the circumstances mentioned in the original post. That's great! :-) I see that you are using the Linux kernel style goto and multiple labels for the single point of exit approach. If I can be of any help in applying a similar coding style to other functions in riscv-013.c please let me know.

timsifive, 2020-12-11:
> If I can be of any help in applying a similar coding style to other functions in riscv-013.c please let me know.

In general, if you submit a pull request that improves the error handling in the current code, I'll happily approve it.

Longer term I would like to have -013 simply read all GPRs at halt, and restore any that need it on resume. That will simplify code like this, and also speed things up because we won't be saving/restoring GPRs all the time.

TommyMurphyTM1234, 2020-12-11:
> This may even be optimized a bit further - keeping a track of which registers have been touched while in the debug mode, so that only those are restored prior to resuming.

I think that's what @timsifive meant as he refers to saving all but restoring only some (the ones that were used presumably)?

JanMatCodasip, 2020-12-11:
@TommyMurphyTM1234  - You're right, I misread Tim's comment. Thanks for pointing it out.

closed 2020-12-14 by timsifive

TommyMurphyTM1234, 2020-12-15:
Hi @timsifive - thanks a lot for addressing this issue so promptly - much appreciated. 👍 
Your earlier suggestion that all (?) integer regs get saved and (where necessary) restored in one place (halt and resume respectively?) sounds a lot cleaner than the current way where the saves/restores are scattered in different places leading to potential error fail return paths that don't properly restore registers.
Am I naive in thinking that that is a relatively straightforward modification to the code and cleans it up a lot by removing those scattered saves/restores?

JanMatCodasip, 2020-12-15:
> ... is a relatively straightforward modification to the code and cleans it up a lot by removing those scattered saves/restores?

Hi @TommyMurphyTM1234 - I haven't looked into what all it would take to make this modification, but I suspect there are more things to consider than it is immediately apparent. For example:
* If debugger (GDB) asks for a register value, the original value needs to be provided (not the modified one).
* Then there is `force` option for OpenOCD's `reg` command that is supposed to read the value of register from the target (bypass cache) - what value should it provide if the register was modified but not yet restored?
* There may be more.

That said, I do not wish to discourage anyone from thinking about how to improve the code. 

However, 
* from short-term perspective 
* for this particular register restore issue
* and to avoid ripple effects

... I recommend to check & harden the current code. That means checking other functions that modify registers, whether they need similar treatment as Tim did in #560. I'm short on time now but can at least help by consulting or review.

TommyMurphyTM1234, 2020-12-15:
> Hi @TommyMurphyTM1234 - I haven't looked into what all it would take to make this modification, but I suspect there are more things to consider than it is immediately apparent. For example:
>
> * If [...]

Thanks - of course. It's not simply a case of backing them up on halt and then restoring them on resume.
Or if it is then some operations (read/write reg) will have to work off the "backed up" copy.

> That said, I do not wish to discourage anyone from thinking about how to improve the code.
> 
> However,
> 
> * from short-term perspective
> * for this particular register restore issue
> * and to avoid ripple effects
> 
> ... I recommend a conservative approach: to check & harden the current code. That means checking other functions that modify registers, whether they need similar treatment as Tim did in #560. I'm short on time now but can at least help by consulting or review.

I agree. I'm loath to attempt any sweeping changes myself. 
But I might at least resume identifying those cases where there is at least theoretical potential for an error/failure return to bypass register restoration.

Thanks again for the comments.

timsifive, 2020-12-15:
> Am I naive in thinking that that is a relatively straightforward modification to the code and cleans it up a lot by removing those scattered saves/restores?

It is not very complex, but it is a significant amount of work. (As usual, most of the work is in testing and fixing any corner cases you didn't think of.) If you have a week to spend on this, it's probably doable. But if you just have an hour here or there, then Jan's suggestions make more sense.

jimaandro, 2025-04-29:
Hi 
I have the same issue on OpenPiton platform. I want to force GDB to use sysbus, but it returns
`Error:   progbuf=disabled, sysbus=skipped (unsupported size), abstract=disabled`

TommyMurphyTM1234, 2025-04-29:
> Hi I have the same issue on OpenPiton platform. I want to force GDB to use sysbus, but it returns `Error: progbuf=disabled, sysbus=skipped (unsupported size), abstract=disabled`

How exactly are you concluding that your issue is the same as the original one that I reported here and which was fixed over four years ago?

jimaandro, 2025-04-30:
Apologies, I misunderstood the original issue — I assumed the main problem was the `sysbus=skipped (unsupported size)` error, which I'm also encountering.
Should I open a new issue?

I am using OpenOCD 0.12.0 . Here is the output log 

[output.zip](https://github.com/user-attachments/files/19973459/output.zip)

TommyMurphyTM1234, 2025-04-30:
> I am using OpenOCD 0.12.0 . Here is the output log

```
Open On-Chip Debugger 0.12.0
```
Where did you get this buld of OpenOCD from?
This doesn't look like the latest OpenOCD built from this repo.
You should build OpenOCD from this repo to ensure that you have the latest RISC-V support and see if the problem persists.
If it does then this may be a question for the OpenPiton project.

en-sc, 2025-04-30:
@jimaandro, from the logs it seems you are trying to issue a 128-bit wide access.
```
Debug: 350 38 command.c:155 script_debug(): command - read_memory 0x1fff1010000 16 1 phys
```
However, the target you are using only supports 64-bit wide System Bus accesses.
```
Debug: 133 24 riscv-013.c:397 scan(): 41b r 00000000 @38 -> + 00000000 @12; 3i
Debug: 134 25 riscv-013.c:397 scan(): 41b - 00000000 @38 -> + 20060808 @38; 3i
Debug: 135 25 riscv-013.c:407 scan():  ->  sbversion=1 sbaccess=3 sbasize=64 sbaccess64
```
And since you have disabled all access methods except for `sysbus`
```
Debug: 41 4 command.c:155 script_debug(): command - riscv set_mem_access sysbus
```
You get the error message:
```
Error:   progbuf=disabled, sysbus=skipped (unsupported size), abstract=disabled
```

Seems like everything works as expected. Am I missing something?

TommyMurphyTM1234, 2025-04-30:
> Isn't it 16bits here?

Isn't it 16 BYTES x 8 bits = 128 bits?

> @jimaandro, from the logs it seems you are trying to issue a 128-bit wide access.

jimaandro, 2025-04-30:
> > Isn't it 16bits here?
> 
> Isn't it 16 BYTES x 8 bits = 128 bits?
> 
> > [@jimaandro](https://github.com/jimaandro), from the logs it seems you are trying to issue a 128-bit wide access.
> 

I think it's bits, based on the manual [https://openocd.org/doc/html/General-Commands.html](https://openocd.org/doc/html/General-Commands.html)

I tried 64bits and it worked. So I understand that it supports only the exact size of Sysbus (64bits)

> What does your script contain in terms of read memory commands?

I only have this `read_memory` command

en-sc, 2025-04-30:
@jimaandro, I'm mistaken. Issuing `read_memory 0x1fff1010000 16 1 phys` does request a read of a single 16-bit element. However, the issue is still the same -- neither 16 nor 128 is 64 -- the only supported size.

TommyMurphyTM1234, 2025-04-30:
> I think it's bits, based on the manual https://openocd.org/doc/html/General-Commands.html

Sorry - you are correct.
But - as per @en-sc's previous comment - maybe your target implementation simply doesn't support 16 bit accesses this way?

en-sc, 2025-04-30:
> So I understand that it supports only the exact size of Sysbus (64bits)

The target reports the supported sizes for System Bus access in `sbcs` register of the DM.
The specification allows for a target to support any subset of the sizes, indicated by `sbaccess128`, `sbaccess64`, `sbaccess32`, `
sbaccess16` and `sbaccess8` bits.

As indicated in the log:
```
Debug: 133 24 riscv-013.c:397 scan(): 41b r 00000000 @38 -> + 00000000 @12; 3i
Debug: 134 25 riscv-013.c:397 scan(): 41b - 00000000 @38 -> + 20060808 @38; 3i
Debug: 135 25 riscv-013.c:407 scan():  ->  sbversion=1 sbaccess=3 sbasize=64 sbaccess64
```
On your particular target only 64-bit wide accesses are supported (only `sbaccess64` is set).

jimaandro, 2025-04-30:
Yes you are right. Thank you very much both of you!
I though I could use smaller bit accesses if the maximum bit size is 64 bits

PR #560 (2020-12-14): Fix error handling in read_memory_progbuf_one().
Be sure to restore mstatus/s0 even if there is a failure during the
operation.

Fixes #559.

Change-Id: Ib86ca2c7455bad4a668f34703566060a782116db
commit f72e5bb0 (2020-12-14): Fix error handling in read_memory_progbuf_one(). (#560)

* Fix error handling in read_memory_progbuf_one().

Be sure to restore mstatus/s0 even if there is a failure during the
operation.

Fixes #559.

Change-Id: Ib86ca2c7455bad4a668f34703566060a782116db

* Style fix suggested in review.

Change-Id: I444112a9dffea483b7d0e5f96ef7bbdaf58d249f