riscv-collab/riscv-openocd #564: x0  actually is NOT cacheable, because it will always read a "ZERO", not the value writen into it

gxflying, 2021-01-11:
Hi， sir, I  used the "riscv_set_register/riscv_get_register" to test on gpr "x0" , and get the wrong  result !
because,  the openocd treats the "x0" as a cacheable register and returns the value I wrote by "riscv_set_register" which is not zero
so we can see the log below  read the x0 : 12345678 (cached) , it's a little bit weird, isn't it ?

[image]

I  think this is because openocd treat  "x0" as a cacheable register below
https://github.com/riscv/riscv-openocd/blob/d9c73e231e5144e05fe35496975c64817ac54555/src/target/riscv/riscv.c#L3675
and "riscv_get_register->riscv_get_register_on_hart" will return the cached value before the value can be read from the hardware when the register is cacheable
https://github.com/riscv/riscv-openocd/blob/d9c73e231e5144e05fe35496975c64817ac54555/src/target/riscv/riscv.c#L3767
so we get a  non-zero value from "x0"
[image]

gxflying, 2021-01-11:
the pictrue shows below:
"Debug riscv.c : 3770 riscv_get_register_on_hart(): {0} zero: 12345678 (cached)"

timsifive, 2021-01-11:
Thanks for catching that. The solution is probably for the code to ignore writes to x0, since they cannot have an effect on RISC-V.

PR #568 (2021-01-16): Don't write to zero.
It doesn't have any effect on real hardware, and by caching the value we
pretended it did.

Fixes #564

Change-Id: I9f4e2cc8abddee61435bbd8d992cbff971a0c28d
commit fcdecd88 (2021-01-16): Don't write to zero.

It doesn't have any effect on real hardware, and by caching the value we
pretended it did.

Fixes #564

Change-Id: I9f4e2cc8abddee61435bbd8d992cbff971a0c28d
closed 2021-01-16 by timsifive