riscv-collab/riscv-openocd #565: riscv sba v1 read memory error

S-Nomii, 2021-01-12:
Hi, after I set the prefer_sba on, and error occurs when gdb try to read piece of memory during adding breakpoint command. Then I tracked the waveform of the jtag signal on board and discovered that openocd tried to read sbdata1 and sbdata0 but only scanning out the request without scanning in the respond. 

According to my understanding it is supposed to scan sbdata1 and sbdata0 like what dmi_read function does, and use some logic like the one inside of the ensure_success body to scan in the respond. I cant find any algorithm perform that in the read_memory_bus_v1 function. Can anyone explain how it works for me.

timsifive, 2021-01-14:
I'm not exactly sure what your question is. If you run openocd with `-d` and log it, you can see exactly what it scans without tracking waveforms. Here's the memory read when setting a breakpoint using SBA, with some comments I inserted:
```
Debug: 7446 3144 riscv-013.c:2821 mem_should_skip_progbuf(): Skipping mem read via progbuf - insufficient progbuf size.
# Set sbreadonaddr
Debug: 7447 3144 riscv-013.c:402 scan(): 40b w 00130000 @38 -> + 00000000 @04; 3i
Debug: 7448 3145 riscv-013.c:412 scan():  sbreadonaddr sbaccess=1 sbautoincrement ->
# Check that DM is not busy.
Debug: 7449 3145 riscv-013.c:402 scan(): 40b - 00000000 @38 -> + 00130000 @38; 3i
Debug: 7450 3145 riscv-013.c:412 scan():  ->  sbreadonaddr sbaccess=1 sbautoincrement
# Write high 32 bits of address (because this target has sbasize=64).
Debug: 7451 3145 riscv-013.c:402 scan(): 40b w 00000000 @3a -> + 00000000 @38; 3i
# Write low 32 bits of address, which triggers a memory read because sbreadonaddr is set
Debug: 7452 3145 riscv-013.c:402 scan(): 40b w 100101bc @39 -> + 00000000 @3a; 3i
# Make sure DM is not busy.
Debug: 7453 3145 riscv-013.c:402 scan(): 40b - 00000000 @39 -> + 100101bc @39; 3i
# Scan in the DM command to read the value
Debug: 7454 3145 riscv-013.c:402 scan(): 40b r 00000000 @3c -> + 00000000 @39; 3i
# Get the actual value back
Debug: 7455 3145 riscv-013.c:402 scan(): 40b - 00000000 @3c -> + 0000a86d @3c; 3i
Debug: 7456 3145 riscv-013.c:2472 log_memory_access(): M[0x100101bc] reads 0xa86d
# Check that sbbusy isn't set
Debug: 7457 3145 riscv-013.c:402 scan(): 40b r 00000000 @38 -> + 00000000 @3c; 3i
Debug: 7458 3145 riscv-013.c:402 scan(): 40b - 00000000 @38 -> + 2013080f @38; 3i
Debug: 7459 3145 riscv-013.c:412 scan():  ->  sbversion=1 sbreadonaddr sbaccess=1 sbautoincrement sbasize=64 sbaccess64 sbaccess32 sbaccess16 sbaccess8
Debug: 7460 3145 riscv-013.c:2812 log_mem_access_result(): Succeeded to read memory via system bus.
```

Is there something in there that doesn't work the way you expect it to?

S-Nomii, 2021-01-15:
Thanks for showing me how to review debug comments, but i am not able to translate the meaning of each part of the comments yet, for example what does 40b mean and what does 41b mean, and i post my debug comments can u explain what exactly went wrong. I am using gdb 9.1 by the way.

Debug: 2367 16759 gdb_server.c:3382 gdb_input_inner(): received packet: 'm800000c0,40'
Debug: 2368 16759 gdb_server.c:1462 gdb_read_memory_packet(): addr: 0x00000000800000c0, len: 0x00000040
Debug: 2369 16759 target.c:2278 target_read_buffer(): reading buffer of 64 byte at 0x800000c0
Debug: 2370 16759 riscv.c:3572 riscv_set_current_hartid(): setting hartid to 0, was 0
Debug: 2371 16759 riscv-013.c:4060 riscv013_get_register(): [0] reading register priv on hart 0
Debug: 2372 16759 riscv.c:3572 riscv_set_current_hartid(): setting hartid to 0, was 0
Debug: 2373 16759 riscv-013.c:792 execute_abstract_command(): command=0x3207b0; access register, size=64, postexec=0, transfer=1, write=0, regno=0x7b0
Debug: 2374 16759 riscv-013.c:404 scan(): 41b w 003207b0 @17 -> + 00000000 @11; 3i
Debug: 2375 16760 riscv-013.c:404 scan(): 41b r 00000000 @16 -> + 003207b0 @17; 3i
Debug: 2376 16760 riscv-013.c:404 scan(): 41b - 00000000 @16 -> + 08000002 @16; 3i
Debug: 2377 16760 riscv-013.c:414 scan():  ->  progbufsize=8 datacount=2
Debug: 2378 16760 riscv-013.c:404 scan(): 41b r 00000000 @05 -> + 00000000 @16; 3i
Debug: 2379 16760 riscv-013.c:404 scan(): 41b - 00000000 @05 -> + 00000000 @05; 3i
Debug: 2380 16761 riscv-013.c:404 scan(): 41b r 00000000 @04 -> + 00000000 @05; 3i
Debug: 2381 16761 riscv-013.c:404 scan(): 41b - 00000000 @04 -> + 4000b0c3 @04; 3i
Debug: 2382 16761 riscv-013.c:1500 register_read_direct(): {0} dcsr = 0x4000b0c3
Debug: 2383 16761 riscv.c:3753 riscv_get_register_on_hart(): {0} priv: 3
Debug: 2384 16761 riscv.c:3737 riscv_get_register_on_hart(): {0} mstatus: 8000000a00006000 (cached)
Debug: 2385 16761 riscv.c:1562 riscv_mmu(): SATP/MMU ignored in Machine mode (mstatus=0x8000000a00006000).
size is : 8 
Debug: 2386 16761 riscv-013.c:404 scan(): 41b w 00178000 @38 -> + 00000000 @04; 3i
Debug: 2387 16761 riscv-013.c:414 scan():  sbreadonaddr sbaccess=3 sbautoincrement sbreadondata -> 
Debug: 2388 16761 riscv-013.c:404 scan(): 41b - 00000000 @38 -> + 00178000 @38; 3i
Debug: 2389 16761 riscv-013.c:414 scan():  ->  sbreadonaddr sbaccess=3 sbautoincrement sbreadondata
Debug: 2390 16762 riscv-013.c:404 scan(): 41b w 00000000 @3a -> + 00000000 @38; 3i
Debug: 2391 16762 riscv-013.c:404 scan(): 41b w 800000c0 @39 -> + 00000000 @3a; 3i
Debug: 2392 16762 riscv-013.c:404 scan(): 41b - 00000000 @39 -> + 800000c0 @39; 3i 
Debug: 2393 16762 riscv-013.c:404 scan(): 41b r 00000000 @3d -> + 00000000 @39; 3i
Debug: 2394 16762 riscv-013.c:404 scan(): 41b r 00000000 @3c -> + 00000000 @3d; 3i
Debug: 2395 16762 riscv-013.c:2495 log_memory_access(): M[0x800000c4] reads 0x00000000
Debug: 2396 16763 riscv-013.c:404 scan(): 41b r 00000000 @3d -> + 00000000 @3c; 3i
Debug: 2397 16763 riscv-013.c:2495 log_memory_access(): M[0x800000c0] reads 0x00000000
Debug: 2398 16763 riscv-013.c:404 scan(): 41b r 00000000 @3c -> + 00000000 @3d; 3i
Debug: 2399 16763 riscv-013.c:2495 log_memory_access(): M[0x800000cc] reads 0x00000000
Debug: 2400 16763 riscv-013.c:404 scan(): 41b r 00000000 @3d -> + 00000000 @3c; 3i
Debug: 2401 16763 riscv-013.c:2495 log_memory_access(): M[0x800000c8] reads 0x00000000
Debug: 2402 16763 riscv-013.c:404 scan(): 41b r 00000000 @3c -> + 00000000 @3d; 3i
Debug: 2403 16763 riscv-013.c:2495 log_memory_access(): M[0x800000d4] reads 0x00000000
Debug: 2404 16763 riscv-013.c:404 scan(): 41b r 00000000 @3d -> + 00000000 @3c; 3i
Debug: 2405 16763 riscv-013.c:2495 log_memory_access(): M[0x800000d0] reads 0x00000000
Debug: 2406 16764 riscv-013.c:404 scan(): 41b r 00000000 @3c -> + 00000000 @3d; 3i
Debug: 2407 16764 riscv-013.c:2495 log_memory_access(): M[0x800000dc] reads 0x00000000
Debug: 2408 16764 riscv-013.c:404 scan(): 41b r 00000000 @3d -> + 00000000 @3c; 3i
Debug: 2409 16764 riscv-013.c:2495 log_memory_access(): M[0x800000d8] reads 0x00000000
Debug: 2410 16764 riscv-013.c:404 scan(): 41b r 00000000 @3c -> + 00000000 @3d; 3i
Debug: 2411 16764 riscv-013.c:2495 log_memory_access(): M[0x800000e4] reads 0x00000000

timsifive, 2021-01-19:
> `Debug: 2374 16759 riscv-013.c:404 scan(): 41b w 003207b0 @17 -> + 00000000 @11; 3i`

41b means 41 bits were scanned. In this case 2 bits for the operation (w=write), 32 bits for the data (0x003207b0), and 7 bits for the DMI address (0x17).

Looking at the memory read portion of the log:

OpenOCD sets sbreadonaddr sbaccess=3 sbautoincrement sbreadondata:
```
Debug: 2386 16761 riscv-013.c:404 scan(): 41b w 00178000 @38 -> + 00000000 @04; 3i
Debug: 2387 16761 riscv-013.c:414 scan(): sbreadonaddr sbaccess=3 sbautoincrement sbreadondata ->
Debug: 2388 16761 riscv-013.c:404 scan(): 41b - 00000000 @38 -> + 00178000 @38; 3i
Debug: 2389 16761 riscv-013.c:414 scan(): -> sbreadonaddr sbaccess=3 sbautoincrement sbreadondata
```

OpenOCD writes the address, which should trigger the first read:
```
Debug: 2390 16762 riscv-013.c:404 scan(): 41b w 00000000 @3a -> + 00000000 @38; 3i
Debug: 2391 16762 riscv-013.c:404 scan(): 41b w 800000c0 @39 -> + 00000000 @3a; 3i
Debug: 2392 16762 riscv-013.c:404 scan(): 41b - 00000000 @39 -> + 800000c0 @39; 3i
```

OpenOCD starts reading values:
```
Debug: 2393 16762 riscv-013.c:404 scan(): 41b r 00000000 @3D -> + 00000000 @39; 3i
Debug: 2394 16762 riscv-013.c:404 scan(): 41b r 00000000 @3c -> + 00000000 @3D; 3i
Debug: 2395 16762 riscv-013.c:2495 log_memory_access(): M[0x800000c4] reads 0x00000000
Debug: 2396 16763 riscv-013.c:404 scan(): 41b r 00000000 @3D -> + 00000000 @3c; 3i
Debug: 2397 16763 riscv-013.c:2495 log_memory_access(): M[0x800000c0] reads 0x00000000
Debug: 2398 16763 riscv-013.c:404 scan(): 41b r 00000000 @3c -> + 00000000 @3D; 3i
...
```

As you can see by the '+' status, the DMI always returns success.

Further down you should see where OpenOCD scans sbcs again to check that sbbusyerror and sberror are both 0. If not, there's a complicated retry that happens.

S-Nomii, 2021-01-21:
> 41b means 41 bits were scanned. In this case 2 bits for the operation (w=write), 32 bits for the data (0x003207b0), and 7 bits for the DMI address (0x17).
>
> Looking at the memory read portion of the [...]

Thanks, that is very help.

en-sc, 2024-01-22:
@S-Nomii, as far as I understand the issue is resolved. Are you OK with me closing it?

closed 2024-01-26 by en-sc