riscv-collab/riscv-openocd #571: Can't compile

TomasPech, 2021-01-26:
Hello I'm new in  openOCD and I'm facing to compile problem ...
I did copy by:
` git clone --branch riscv https://github.com/riscv/riscv-openocd  `
but when I did next steps:
` ./bootstrap && ./configure  && make -j all ` then I got error about USB :

[image]

I also tried ` ./bootstrap && ./configure --disable-werror && make -j 11 ` but it doesn't help:
[image]

Then I looked into Travis output: [Travis Job - Line 880](https://travis-ci.org/github/riscv/riscv-openocd/jobs/756145377#L880) where I saw that compilation without USB and without any issue.

Can somebody tell me what I can try for sucesfull compilation with USB?

` NAME="Linux Mint"
VERSION="20.1 (Ulyssa)"
ID=linuxmint
ID_LIKE=ubuntu
PRETTY_NAME="Linux Mint 20.1"
VERSION_ID="20.1"
VERSION_CODENAME=ulyssa
UBUNTU_CODENAME=focal `

TommyMurphyTM1234, 2021-01-26:
It'd be better if you posted logs rather than images. They are difficult to read on certain devices.

In my opinion, if you actually need to build openocd yourself same cannot use binary builds available already from various places (e.g. here: https://xpack.github.io/openocd/#releases), then the easiest and most reliable way is using @ilg-ul's xPack project docker based build scripts.

https://xpack.github.io/openocd/
https://xpack.github.io/openocd/#build-details

To build from the RISC-V openocd repo you'll need to edit this file to use this repo and the relevant branch and commit:

https://github.com/xpack-dev-tools/openocd-xpack/blob/xpack/scripts/defs-source.sh#L45

JanMatCodasip, 2021-01-26:
@TomasPech - Hi Tomas,

The suggestion from @TommyMurphyTM1234 is definitely a way to go.

If you'd like to stick to the "traditional" build (configure - make - make install):

Based on your logs, it seems that you're missing libusb and that is the reason for the failed build. Please install it using your package manager including the development headers:

```
# Debian
sudo apt-get install libusb-1.0

# CentOS:
sudo yum install libusb-devel

# Your distribution:
# .....
```
Let us know if that helped.

TomasPech, 2021-01-27:
Thank you for your answers but it didn't help much.

@TommyMurphyTM1234:
 xPacks was first what I installed but sadly it looks like that xPack version have some problem with flash driver for GD32VF103 ` Error: flash driver 'gd32vf103' not found `. Current xPack version was relesed on 13.10.2020 and this [commit](https://github.com/riscv/riscv-openocd/commit/e8379cda326e358a529e302d25cdfc5fde6a1219) was added 16.10.2020 so that is why I would like to try compile this "dev" version. 

@JanMatCodasip:
Thank you for hint but libusb was installed from older project and then problem is still persisting.

FYI: 
During compilation I can see that is included: ` -isystem /usr/include/libusb-1.0 ` - folder really contains header file: **libusb.h**
```
Version of GCC: gcc version 9.3.0 (Ubuntu 9.3.0-17ubuntu1~20.04)

tomas@tomas-desktop: ~/RISC-V/OpenOCD/riscv-openocd$ apt list --installed | grep libusb*

WARNING: apt does not have a stable CLI interface. Use with caution in scripts.

libhidapi-libusb0/focal,now 0.9.0+dfsg-1 amd64
libusageenvironment3/focal,now 2020.01.19-1build1 amd64
libusb-0.1-4/focal,now 2:0.1.12-32 amd64
libusb-1.0-0-dev/focal,now 2:1.0.23-2build1 amd64
libusb-1.0-0/focal,now 2:1.0.23-2build1 amd64
libusb-1.0-0/focal,now 2:1.0.23-2build1 i386
libusb-dev/focal,now 2:0.1.12-32 amd64
libusbmuxd4/now 1.1.0~git20171206.c724e70f-0.1 amd64
libusbmuxd6/focal,now 2.0.1-2 amd64
libusrsctp1/focal,now 0.9.3.0+20190901-1 amd64
```

Thank you for any hint ..

TommyMurphyTM1234, 2021-01-27:
> Thank you for your answers but it didn't help much.
>
> @TommyMurphyTM1234:
> xPacks was first what I installed but sadly it looks like that xPack version have some problem with flash driver for [...]

That's why I also explained how to use @ilg-ul's stuff to compile openocd from whatever repo/branch/commit that contains what you need!

JanMatCodasip, 2021-01-29:
>  Thank you for hint but libusb was installed from older project and then problem is still persisting.

@TomasPech - The missing definition of `struct usb_device` et al. indicates that maybe you are missing the "kernel headers" package. Install it by this command (or similar):

```
sudo apt install linux-headers-$(uname -r)
```

I'd be glad if you leave a note here whether this finally resolved your build issues.

JanMatCodasip, 2021-01-29:
@TomasPech  - One more hint:

These are the arguments that I typically use with `./configure`:
- to enable support of the most commonly used JTAG adapters
- to set a custom installation directory

```
./configure --enable-ftdi --enable-jtag_vpi --enable-remote-bitbang --prefix=`pwd`/install
```

TommyMurphyTM1234, 2021-01-29:
> @TomasPech - Hi Tomas,
> 
> If you'd like to stick to the "traditional" build (configure - make - make install):
> 

Just to be clear @ilg-ul's xPack Project docker based scripts also do a regular bootstrap/configure/make/install but do so within a controlled environment that gets or builds all depedencies, builds for maximum cross platform/distro compatbility and yields predictable results every time. And, as I mentioned, it can easily be tweaked to build from any openocd repo/branch/commit. It's some what analogous to using CT-NG to build a GCC toolchain but with even more predictability. Given how common questions/problems relating to building openocd are (especially when (cross) compiling for Windows) are it is, in my experience, the best approach available right now to simplifying the openocd build process.

JanMatCodasip, 2021-01-29:
@TomasPech - Tomas, you are welcome to use either of the build options presented here:

- a) native build (boostrap - configure - make - make install)
- b) xPack Project docker based scripts, as pointed out by @TommyMurphyTM1234 

Even if you decide to go with b) in the end, I'd be happy if you could try installing the linux kernel headers and trying the native build a). Just to make sure there are not any build issues with OpenOCD that we do not know of. Thanks.

TomasPech, 2021-01-30:
Thank all of you for answers and I apologize for my long delay but end of this week was too hectic for OpenOCD task. 
Last night I spend with compiling OpenOCD with option **B)** - and was successful but I would like to continue with native compiling because is easier more "common" without additional tools. 

@JanMatCodasip: Kernel headers may be good point where I can start because I'm using pretty new kernel (5.10.6-051006-generic) and there is maybe some change in header? Tomorrow I will try compile with other kernel as 5.4 or so.

TomasPech, 2021-01-31:
Today I did compilation with different kernel version (5.4.0-64) and with same result - looks like ` usb.h ` isn't in headers. Problem is definitely in my workstation because also in VirtualBox (Linux Mint - LiveCD) compiled without any error.  I tried lot of procedures to fix it but from unknown reason I still can't find out where problem exactly is.

This issue can be closed because problem isn't on OpenOCD side.
Thank all of you !

JanMatCodasip, 2021-02-01:
Tomas, thank you for spending the additional time to look into the compilation issues and report back here. Very appreciated.

I am closing this issue now.

If you run into anything further regarding riscv-openocd, feel free to reach back.

closed 2021-02-01 by JanMatCodasip