riscv-collab/riscv-openocd #583: thread-specific breakpoints for a dual-core CPU

duchao713, 2021-02-22:
Hi,

I'm debuging a dual-core riscv CPU using riscv-openocd. The configuration is as below

`set _CHIPNAME riscvDual
set _CPUTAPID 0x00000001
jtag newtap $_CHIPNAME cpu -irlen 5 -expected-id $_CPUTAPID

set _TARGETNAME_0 $_CHIPNAME.cpu0
set _TARGETNAME_1 $_CHIPNAME.cpu1
target create $_TARGETNAME_0 riscv -chain-position $_CHIPNAME.cpu -coreid 0 -rtos hwthread
target create $_TARGETNAME_1 riscv -chain-position $_CHIPNAME.cpu -coreid 1

target smp $_TARGETNAME_0 $_TARGETNAME_1`

Now I indend to set thread-specific breakpoints and try to hit the breakpoint, what is the correct procedure ? does scheduler-lock need to be set ?

I checked the codes in _gdb_handle_vcont_packet_, seems the 'c packet' does not distinguish the thread-id. So I could use step command only ?

Thanks.

TommyMurphyTM1234, 2021-02-22:
The script looks correct to me apart from the leading the training backticks.
I presume that you were trying to quote the code or something?

You can use gdb's thread command to select a specific thread/hart before issuing a command.
But bear in mind that resuming (or stepping?) one thread may resume (or step) the other threads depending on the debug implementation.
Check the debug spec for details.

Hope this helps.

timsifive, 2021-02-22:
I suspect that https://sourceware.org/gdb/current/onlinedocs/gdb/Thread_002dSpecific-Breakpoints.html will work, although it will be slow because the target will hart when either hart hits the breakpoint, and then gdb will "quickly" resume all harts if it's not the "correct" thread.

duchao713, 2021-02-25:
Thanks Tommy and Tim.
Now it works.

closed 2021-02-25 by duchao713