riscv-collab/riscv-openocd #584: In debug mode, only one thread display stack info other threads dispaly 0

paul5730, 2021-03-11:
As the title suggests,target chip belong to riscv , when the program runs to a breakpoint, only one thread displays the Stack Info.The other threads are 0 . as shown in the figure below
[image]

JanMatCodasip, 2021-03-11:
Hi @paul5730, 

from that very little information provided, I guess this is a single-hart RISC-V target running a FreeRTOS-based application. Am I correct?

FreeRTOS support has been commited to this riscv-openocd very recently (5 days ago in 50a5971be27ff11061776afc8014bbff22870b7c by @timsifive). Does your openocd build already include that change?

Jan

paul5730, 2021-03-11:
Hi JanMatCodasip 
JanMatCodasip : from that very little information provided, I guess this is a single-hart RISC-V target running a FreeRTOS-based application. Am I correct?
Paul5730: Yes, this is a single-hart RISC-V target , is it support for FreeRTOS thread awareness ?
JanMatCodasip : FreeRTOS support has been commited to this riscv-openocd very recently (5 days ago in 50a5971 by @timsifive). Does your openocd build already include that change?
Paul5730: Yes, I build already include that change.

JanMatCodasip, 2021-03-11:
I may not be able to help here due to limited time/knowledge, but if you'd like someone to take a look at this, please share a full verbose log from your OpenOCD: use argument `-d3`. Thanks.

timsifive, 2021-03-11:
I am actively working on the FreeRTOS support. It's definitely not complete yet. In fact I'm looking right now at a case where OpenOCD reports bad register values for the suspended threads.

timsifive, 2021-03-17:
Did you build your FreeRTOS binary using mainline FreeRTOS by any chance? It stores registers on the stack differently from how it works in the metal fork. :-(

If so, can you see if https://github.com/riscv/riscv-openocd/pull/586 works any better for you?

en-sc, 2024-01-22:
@paul5730, could you please check if the issue is still present? I would like to close it, but I do not have enough information to reproduce it and verify that it's fixed.

closed 2024-01-26 by en-sc