riscv-collab/riscv-openocd #587: Loading programs with program buffer expects autoexec support (optional feature according to debug-spec 0.13)

sujay-pandit, 2021-03-27:
Simple fix in "write_memory_progbuf" in riscv-013.c

JanMatCodasip, 2021-03-27:
Hi Sujay,

Thank you for filing this ticket.

I can confirm this is a valid issue. 

All bits in "abstractauto" DM registers are optional per the RISC-V Debug specification, and targets need not support it. However, the current OpenOCD code for both memory read and write via program buffer relies on the autoexecdata[0] bit - without checking if it is really supported.

Side note: Most targets appear to have autoexecdata[0] bit implemented, which is also recommended for performance reasons. That perhaps masked this issue in OpenOCD.

sujay-pandit, 2021-05-04:
Thanks Jan, yes I agree. We have added autoexec support now.

TommyMurphyTM1234, 2023-03-26:
I take it that this issue is still extant and awaiting a fix?

* https://github.com/riscv/riscv-openocd/blob/a01bd76e5c10eba8bc3481eb7552fb8feab79f91/src/target/riscv/riscv-013.c#L3990

nadime15, 2025-10-29:
Is this still an issue? I usually work with the upstream version of OpenOCD, so I am not sure whether it also affects the downstream version.

> Though IMHO it's a low priority one, since most hardware does support autoexec AFAIK.

This might be true, but at the end of the day it is still an optional feature and should not be assumed to exist. It literally took me a full day to figure out why my GDB sessions were spitting out random data. Most people will probably support it, but OpenOCD should still check whether abstractauto is writable before using it. I am sure most people will test features incrementally rather than implementing autoexec right away.

TommyMurphyTM1234, 2025-10-30:
> OpenOCD should still check whether abstractauto is writable before using it.

Patches and PRs are always welcome. Feel free to submit a PR to address this issue.